espressif/arduino-esp32 #6797: 'LOG_TAG' was not declared in this scope

joseitor, 2022-05-25:
### Board

ESP32_wrover-B

### Device Description

ESP32_Wrover_B

### Hardware Configuration

Any relevant

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Now, when I compile appears the error "'LOG_TAG' was not declared in this scope", when in previous version 2.0.2 all works perfectly fine. What changed has been made in that topic since previous version? I can't find any related in the release notes and not a solution for this neither.

### Sketch

```cpp
ESP_LOGI(LOG_TAG, "Press X:%i / Y:%i", touchX, touchY);
```

### Debug Message

```plain
'LOG_TAG' was not declared in this scope
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-05-25:
1. Why are you using IDF logging?
2. You can `static const char * LOG_TAG = "whatever you want";` in your sketch / source files. This is required by ESP-IDF as well

SuGlider, 2022-05-25:
@joseitor 
ESP32 Arduino has some simpler API to logging:
`log_i("Press X:%i / Y:%i", touchX, touchY);`

There is also `log_d()`, `log_v()`,  `log_w()` and `log_e()` for different logging levels.

SuGlider, 2022-06-01:
Closing it, given it is not Arduino directly related.