espressif/arduino-esp32 #6803: FLASH 8 M impossible to use

moreauftheobald, 2022-05-25:
### Board

ESP32-S3-DevKitC-1 N8R8

### Device Description

ESP32-S3-DevKitC-1 N8R8 stand Alone

### Hardware Configuration

not any connection

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Linux UBUNTU

### Flash frequency

QIO 80Mhz, DIO 80Mhz, OPI 80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

ESP32 reboot continusmy when partition activated envolve memory space UP to 4Mb

Tested With all FLASH mode QIO, DIO OPI a 80 and 120Mz with bootloader upload each time.

when select Huge APP or Default 4MB (both) or other partion Sheme involving 4MB flash everything is working, When using 8M flash or personnal partition file ESP32 S3 reboot with no special error message.

### Sketch

```cpp
int i = 0;

void setup() {
  Serial.begin(9600);

}

void loop() {
  // put your main code here, to run repeatedly:
  i++;
  Serial.println("Hello "+String(i,10));
  delay(1000);
}
```

### Debug Message

```plain
entry 0x403b61c0

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403bb0ce
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x39c
load:0x403b6000,len:0x9a4
load:0x403ba000,len:0x2868
entry 0x403b61c0
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403bb0ce
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x39c
load:0x403b6000,len:0x9a4
load:0x403ba000,len:0x2868
```

### Other Steps to Reproduce

Personnal Partition file (not working ) :

# Name | Type | SubType | Offset         | Size            | Flags
nvs        | data | nvs           | 0x9000      | 0x5000       |  
app0      | app  | ota_0       | 0x10000    | 0x600000   |  
spiffs      | data | spiffs        | 0x610000 | 0xF0000      |  

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

meltdown03, 2022-05-26:
I get a similar error when using **any settings** and tying to flash **any program** on my ESP32-S3-DevKitC-1 (E2N8), no PSRAM, 8MB:
``` 
19:42:54.967 -> ESP-ROM:esp32s3-20210327
19:42:54.967 -> Build:Mar 27 2021
19:42:54.967 -> rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
19:42:54.967 -> Saved PC:0x400454d5
19:42:54.967 -> SPIWP:0xee
19:42:54.967 -> mode:QIO, clock div:2
19:42:54.967 -> load:0x3fcd0108,len:0x43c
19:42:54.967 -> ets_loader.c 78 
```

meltdown03, 2022-05-26:
If I use PlatformIO and the same Arduino version (master branch), I have no problems.

Jason2866, 2022-05-26:
@moreauftheobald Esptool.py v3.3 installed and using? Without you run in this issue.

moreauftheobald, 2022-05-26:
My version of esptool is V3.3-dev

PS /xxx/xxx/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools> python esptool.py version
esptool.py v3.3-dev
3.3-dev
PS /home/moreauf/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools> 

PS /xxx/xxx/.arduino15/packages/esp32/hardware/esp32/2.0.3/tools> python esptool.py -p /dev/ttyACM0 flash_id
esptool.py v3.3-dev
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e3:4e:b0
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4017
Detected flash size: 8MB
Hard resetting via RTS pin...

moreauftheobald, 2022-05-26:
> @moreauftheobald Esptool.py v3.3 installed and using? Without you run in this issue. I never tried to do a partition scheme without the entry
> 
> ```
> otadata, data, ota, 0xe000, 0x2000,
> ```
> 
> Afaik it is needed.

I think to, but Huge_app partition have this partitions but dont work to...

meltdown03, 2022-05-27:
I had to use esptool.py v4 to fix the issue with mine. Strange as it only happens with the Arduino IDE even though PlatformIO uses esptool.py v3.3, but it works.

moreauftheobald, 2022-05-28:
> I had to use esptool.py v4 to fix the issue with mine. Strange as it only happens with the Arduino IDE even though PlatformIO uses esptool.py v3.3, but it works.

could you just explain me how to setup ESPTOOL V4 on arduino IDE ? I've tried to do it, but did not success.

prosd, 2022-06-05:
Same issue here! 

Using partition layout with total of less then 4Mb works. If I use (any combination) of partitions which results in a total of > 4Mb the board keeps rebooting (SW_RESET)

Working partition layout :
`
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x300000,
ffat,   data, fat,  0x310000,0xF0000,
#ffat,   data, fat,  0x310000,0x300000,
`
Changing size of ffat partition from 0xF0000 to 0x300000 causes issue

Tested on 2 different boards with 4 and 16 Mb of psram (Lily Go V7 and Lolin D32 Pro)

Thanks!

Jason2866, 2022-06-05:
My AI Thinker ESP-S3-12K Board (8 MB Flash / 8 MB PSRAM both QIO) works fine with this boards definition and Platformio
```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "memory_type": "qspi_qspi"
    },
    "core": "esp32",
    "extra_flags": "-DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 -DESP32_8M -DESP32S3",
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dio",
    "mcu": "esp32s3",
    "variant": "esp32s3",
    "partitions": "partitions/esp32_partition_app2944k_fs2M.csv"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "espidf",
    "arduino"
  ],
  "name": "Espressif Generic ESP32-S3 8M Flash, Tasmota 2944k Code/OTA, 2112k FS",
  "upload": {
    "flash_size": "8MB",
    "maximum_ram_size": 327680,
    "maximum_size": 8388608,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/",
  "vendor": "Espressif"
}
```
and this partition scheme
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x2F0000,
app1,     app,  ota_1,   0x300000, 0x2F0000,
spiffs,   data, spiffs,  0x5F0000,0x210000,
```

Jason2866, 2022-06-06:
@BlueAndi What should Ivan Kravets do here? The boards.json `flash_size` has to be align with the partition scheme. It is not possible to use with a 4MB definition a parttition scheme which has definitions >4MB. If you have such a case override the `flash_size` in your platformio.ini.

BlueAndi, 2022-06-06:
@Jason2866 Yes, you right. Using the overwrite mechanism can fix it.
```
board_upload.flash_size = "8MB"
board_upload.maximum_size = 8388608
```
It was more like FYI, because in older setups it was working without explicit overwrite it. A custom 8MB partition table was enough. Anyway my notification to Ivan was not accurate enough, thats true. Sry ...

dexter323i, 2022-06-20:
Same for me. I have ESP32 wrover 16MB flash version. It works with deafult partition, but goes into infinite restart loop with large_spiffs_16MB.csv partition scheme. Addig this line into platforio.ini solved it for me too:
board_upload.flash_size=16MB

VojtechBartoska, 2022-09-21:
Hello @moreauftheobald, Can you please help us with testing under latest version of Arduino core 2.0.5? Thanks in advance!

moreauftheobald, 2022-09-21:
I'll try to test it.... i'm a kind busy, but i will try to find time to d it.

moreauftheobald, 2022-09-23:
Just tryed it, il works perfectly event with custom partition scheme.

me-no-dev, 2022-09-23:
Thanks for confirming @moreauftheobald

closed 2022-09-23 by me-no-dev