espressif/arduino-esp32 #6806: ESP32 - Crash during File Append operation v 1.0.6 and in 2.0.3

mihirgraspberry, 2022-05-26:
### Board

ESP32 Wroom 32D

### Device Description

ESP32 WROOM 32D

### Hardware Configuration

ESP32 WROOM 32D

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

When Appendfile function is called, the code crashes some times. It does not happen every time but only once in a while. Like say out of 100 times may be around 2 times. 
Please find the crash log. 
We are actually reading some data from another ESP module and then writing into the LittleFS. While calling appendfile function 
this crash happens/. 

### Sketch

```cpp
void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\r\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("- failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("- message appended");
    } else {
        Serial.println("- append failed");
    }
    file.close();
}
```

### Debug Message

```plain
??
??:0
??
??:0
_fseeko_r
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fseeko.c:221
_fseek_r
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fseek.c:114
_fopen_r
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fopen.c:163
fopen
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fopen.c:181
ZN11VFSFileImplC2EP7VFSImplPKcS3
C:\Users\MST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src/vfs_api.h:31
ZN9__gnu_cxx13new_allocatorI11VFSFileImplE9constructIS1_JRP7VFSImplPKcRS8_EEEvPT_DpOT0
C:\Users\MST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src/vfs_api.h:31
ZNSt10shared_ptrI11VFSFileImplEC4ISaIS0_EJRP7VFSImplPKcRS8_EEESt19_Sp_make_shared_tagRKT_DpOT0
C:\Users\MST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src/vfs_api.h:31
ZN2fs2FS4openEPKcS2
C:\Users\MST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src/FS.cpp:258
_ZN2fs2FS4openERK6StringPKc
C:\Users\MST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\FS\src/FS.cpp:258
Z10appendFileRN2fs2FSE6StringS2
C:\Users\MST01\Desktop\gwcode_with_getcommands/gwcode_with_getcommands.ino:1636
_Z15getfilefromnode6String
C:\Users\MST01\Desktop\gwcode_with_getcommands/gwcode_with_getcommands.ino:1636
_Z6divide6String
C:\Users\MST01\Desktop\gwcode_with_getcommands/gwcode_with_getcommands.ino:1636
_Z19GetFileListfromNodev
C:\Users\MST01\Desktop\gwcode_with_getcommands/gwcode_with_getcommands.ino:1636
_Z18LocalNetworkReaderv
C:\Users\MST01\Desktop\gwcode_with_getcommands/gwcode_with_getcommands.ino:1636
_Z4loopv
C:\Users\MST01\Desktop\gwcode_with_getcommands/gwcode_with_getcommands.ino:1636
_Z8loopTaskPv
C:\Users\MST01\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/main.cpp:23
vPortTaskWrapper
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-05-26:
Hello, do you face this also on Arduino Core version 2.0.3?

VojtechBartoska, 2022-05-26:
@me-no-dev can be related to https://github.com/espressif/arduino-esp32/commit/adb88d7bed128048de783d11964a80143bdd44c0?

me-no-dev, 2022-05-26:
> @me-no-dev can be related to [adb88d7](https://github.com/espressif/arduino-esp32/commit/adb88d7bed128048de783d11964a80143bdd44c0)?

No. Issue is reported with Arduino version, so he is not using github, where that change actually is.

@P-R-O-C-H-Y PTAL

@mihirgraspberry any serial output? I see that there are print statements, also the exeption should be visible in the console. Please use the ExceptionDecoder to decode the backtrace. And also please make sure your are on 2.0.3

mihirgraspberry, 2022-05-26:
ok. on 2.0.3 it is not visible now. I have upgraded it. I have kept it for observation.

VojtechBartoska, 2022-05-27:
ok, I'm closing as solved. If needed, please reopen the issue. Thanks for your contibution.