espressif/arduino-esp32 #6807: ESP32 C3 reboot when OTA

WangChun420, 2022-05-26:
### Board

LILYGO® T-OI PLUS RISC-V ESP32-C3

### Device Description

ESP32C3_DEV

### Hardware Configuration

no

### Version

v2.0.3

### IDE Name

Sloeber

### Operating System

Window10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

when  i use OTA anyway ble ,basic ota or webserver ,the "abort() was called at PC 0x40380fe9 on core 0" accours, i trace the pc at "/esp32-arduino-lib-builder/esp-idf/components/spi_flash/esp_flash_api.c:541 (discriminator 5)", is ther some error?

### Sketch

```cpp
#include "Update.h"
#include "Arduino.h"
#include "esp_spi_flash.h"
#include "esp_ota_ops.h"
#include "esp_image_format.h"

static const char * _err2str(uint8_t _error){
    if(_error == UPDATE_ERROR_OK){
        return ("No Error");
    } else if(_error == UPDATE_ERROR_WRITE){
        return ("Flash Write Failed");
    } else if(_error == UPDATE_ERROR_ERASE){
        return ("Flash Erase Failed");
    } else if(_error == UPDATE_ERROR_READ){
        return ("Flash Read Failed");
    } else if(_error == UPDATE_ERROR_SPACE){
        return ("Not Enough Space");
    } else if(_error == UPDATE_ERROR_SIZE){
        return ("Bad Size Given");
    } else if(_error == UPDATE_ERROR_STREAM){
        return ("Stream Read Timeout");
    } else if(_error == UPDATE_ERROR_MD5){
        return ("MD5 Check Failed");
    } else if(_error == UPDATE_ERROR_MAGIC_BYTE){
        return ("Wrong Magic Byte");
    } else if(_error == UPDATE_ERROR_ACTIVATE){
        return ("Could Not Activate The Firmware");
[... 439 lines cut ...]
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403818f0
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: add61d73313dc1c42721c564614c39a1730db11ed9a0f70b12aa2d493e175109
Expected: f24fa095b9eee178e67fb399bb138b85dabcd0850fd9b144fdce2ba6e429fb35
Attempting to boot anyway...
entry 0x403ce000

.......
Connected to Xiaomi_DA77
IP address: 192.168.31.156
mDNS responder started
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-05-26:
Seems that you have pasted the source of the UpdateClass instead of a minimal sketch example of how we can reproduce this.

Jason2866, 2022-05-26:
You mentioned BLE. We had WDTs in Project Tasmota too when BLE is not deactivated before doing a OTA upgrade.

WangChun420, 2022-05-27:
here is the basic OTA sketch, now i found OTA is just fit default 4M Partition Scheme ,others also have the same problem
`#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "Xiaomi_DA77";
const char* password = "ylzk123456";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 3232
   ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });

  ArduinoOTA.begin();

  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}
`

me-no-dev, 2022-05-27:
OTA get's the next partition from what is installed on the device. It does not have anything hardcoded.

edieruby, 2022-06-27:
log:
SHA-256 comparison failed:
Calculated: add61d73313dc1c42721c564614c39a1730db11ed9a0f70b12aa2d493e175109
Expected: f24fa095b9eee178e67fb399bb138b85dabcd0850fd9b144fdce2ba6e429fb35
Attempting to boot anyway...

I have this log too.
It seems the  Secure Boot V2 ?
Do this matter to the program?

WangChun420, 2022-07-10:
> @WangChun420 Do you still need help?

Thanks,i'v alredy solved!

closed 2022-07-10 by WangChun420

PovoduDigital, 2022-10-03:
@WangChun420 How did you solve this issue? Can you explain please. Or is anyone know how to solve this? I really need to solve this immediately.

EnesCinr, 2022-10-04:
@WangChun420 How did you solve this issue? Can you explain please. Or is anyone know how to solve this? I really need to solve this immediately.