espressif/arduino-esp32 #6818: reboot loop if starting ethernet on ESP32-PICO-V3

hgatesCPV, 2022-05-27:
### Board

ESP32-PICO-KIT (modified), also on a custom board

### Device Description

I have two ESP32-PICO-KIT boards. They originally both had D4 chips on them. On one of them I replaced the chip with a V3 chip.

### Hardware Configuration

nothing, just USB cable

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

default

### PSRAM enabled

no

### Upload speed

115200

### Description

I can load and run simple test code, like LED or I/O pin blink. But when I try to start ethernet, program hangs and resets. This is only with V3 chip. With D4 chip I don't have the issue. This has come about because I designed a custom board for V3 and ordered 2k pieces of the chip. So I would really like to get V3 working!

### Sketch

```cpp
#include <Arduino.h>
#include <SPI.h>
#include <Ethernet.h>       // Ethernet library v2 is required

// Enter a MAC address and IP address for your controller below.
byte mac[] = {
  0x98, 0x76, 0xB6, 0x11, 0x85, 0x82
};
IPAddress ip(10,0,6,75); // The IP address will be dependent on your local network:

void setup() {
  Serial.begin(115200);     // Open serial communications and wait for port to open
  Serial.println("in setup\r\n");
  Ethernet.init(33);        // SS pin
  Ethernet.begin(mac, ip);  // start the Ethernet connection
  delay(1000);              // give the Ethernet shield a second to initialize
}

void loop() {
  Serial.println(Ethernet.localIP());
  delay(1000); 
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1    
load:0x3fff0030,len:184  
load:0x40078000,len:12624
load:0x40080400,len:2916 
entry 0x400805c4
in setup

ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:184
load:0x40078000,len:12624
load:0x40080400,len:2916
entry 0x400805c4
in setup
```

### Other Steps to Reproduce

On same board type (ESP32-PICO-KIT) with original D4 chip I get this output:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2    
load:0x3fff0030,len:184  
load:0x40078000,len:12596
load:0x40080400,len:2916 
entry 0x400805c4
in setup

0.0.0.0
0.0.0.0
0.0.0.0
0.0.0.0
0.0.0.0
0.0.0.0

here is my platformio.ini

[env:pico32]
platform = https://github.com/Jason2866/platform-espressif32.git
framework = arduino
board = pico32
monitor_speed = 115200
build_flags = -DLOG_LEVEL=6

lib_deps =
    arduino-libraries/Ethernet@^2.0.0

I found some other similar things from other people about rebooting loops on V3 and D4 chips. Some people had power supply issues - I've tried adding capacitors and running off a lab power supply. Other people had luck using different sets of libraries and packages - I've tried all those and no change. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

hgatesCPV, 2022-05-27:
Also tried basic code with Wifi instead of Ethernet, same thing. When I get to ETH.begin() in setup() board hangs and then reboots from rst:0x08, perpetually.

lbernstone, 2022-05-27:
Why can you not use the esp32 ethernet library?  This repo does not support the library you are including. https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet/examples

hgatesCPV, 2022-05-27:
I don’t think that library works for W5500 Ethernet chip, right? 

My application uses W5500. I chose that because it seemed pretty common with lots of projects based on it. It works fine with ESP32 non-pico

hgatesCPV, 2022-05-27:
But I also tried it (on the esp32-pics-kit board) with arduino built in library and Wi-Fi. Same result: seemed to work for D4 chip, reboot loop for V3 chip

lbernstone, 2022-05-28:
When you say running off a lab supply, do you mean you are supplying 3v3 from the lab supply directly to a 3v3 bus, or running 5v to a LDO/regulator that then converts to 3v3?  If you have these reboots with wifi as well, then the Ethernet part can be ignored for now until you get Wifi working.

hgatesCPV, 2022-05-28:
3.3v from lab supply for custom board. For esp32-pico-kit board power is coming from usb and through the ldo on that board.

lbernstone, 2022-05-28:
Do you have a way to bypass the ldo and deliver 3v3 directly from the lab supply to the esp32 Vcc to test?  These sorts of issues typically come from an inadequate current.

hgatesCPV, 2022-05-28:
Sure I can do that. That is how I already have the custom board set up (though also tried it with switching regulator 24->3.3). On Monday I will try 3.3 direct to esp32-pico-kit board.

hgatesCPV, 2022-05-30:
Attaching 3.3V from lab supply to pins on board does not change anything. Reboot loop for V3, not for D4.

To summarize where I am at now: 

- Using two ESP32-PICO-KIT boards from Espressif, nothing else connected to board except USB cable
- one with the D4 chip it came with, one with a V3 chip I swapped in
- D4 chip seems to work normally, as far as I have tested it
- V3 chip can load and execute code, use serial terminal, and do simple things like toggle a pin. But when enabling wifi the chip hangs and reboots from watchdog. 

I have tried direct 3.3V power supply, various other libraries and repositories, lower chip clock speed. So far nothing has an impact.

Here are the two boards I am using
[image]

Here is the basic sketch I am trying to use
```C++
#include <Arduino.h>
/*
    This sketch shows the Ethernet event usage
*/
#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case ARDUINO_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  Serial.println("in setup\r\n");
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  Serial.println(ETH.localIP());
  Serial.println("tick\r\n");
  delay(10000);
}
```
platformio.ini
```
[env:pico32]
;platform = espressif32
platform = https://github.com/Jason2866/platform-espressif32.git
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
board = pico32
framework = arduino
monitor_speed = 115200
build_flags = -DLOG_LEVEL=6
board_build.f_cpu = 40000000L
```

output from D4 chip:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13000
load:0x40080400,len:3036 
entry 0x400805e4
in setup

E (1058) esp.emac: emac_esp32_init(349): reset timeout
E (1060) esp_eth: esp_eth_driver_install(214): init mac failed 
[  1067][E][ETH.cpp:323] begin(): esp_eth_driver_install failed
0.0.0.0
tick

0.0.0.0
tick   

0.0.0.0
tick
```

output from V3 chip
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13000
load:0x40080400,len:3036 
entry 0x400805e4
in setup

ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13000
load:0x40080400,len:3036
entry 0x400805e4
in setup

ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13000
load:0x40080400,len:3036
entry 0x400805e4
in setup

ets Jul 29 2019 12:21:46
```

mrengineer7777, 2023-04-13:
Unfortunately this repo does not yet support SPI ethernet modules like the W5500.  You'll have to study `Ethernet.h` to see which pins it uses.

Note W5500 IS supported at IDF level.

closed 2023-04-13 by mrengineer7777