espressif/arduino-esp32 #6819: sha256 hash hardware acceleration?

torntrousers, 2022-05-29:
### Board

M5Stack EduKit

### Device Description

.

### Hardware Configuration

.

### Version

v2.0.3

### IDE Name

platformio

### Operating System

Windows

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I happened to be doing some testing of different sha256 impls and found that in my test mbedtls sha256 with esp32/arduino v2.0.3 was slower than v2.0.2, taking about 80 microseconds compared to 59 microseconds on v2.0.2, so over a 35% slower.  

I remember seeing [this issue](https://github.com/espressif/arduino-esp32/issues/976#issuecomment-1072222200) that I think is saying from 2.0.3 mbedtls will use the ESP32 hardware acceleration for things like mbedtls sha256 hashes. 

Its not a big deal I'm just curious - shouldn't v2.0.3 be faster at this if its using hardware acceleration now?

I do have a testcase to show this if interested but would have to do some work to pullout just this sha256 test into a standalone project.

### Sketch

```cpp
.
```

### Debug Message

```plain
.
```

### Other Steps to Reproduce

.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-05-30:
@me-no-dev PTAL, thanks!

me-no-dev, 2022-05-30:
@torntrousers actually it's slower in hardware, because the clock speed of the hardware is 80MHz and the CPU at 240MHz can calculate a bit faster. If the CPU was also at 80MHz, then hardware would have been twice faster or so...

torntrousers, 2022-05-30:
I see, interesting. thanks.

closed 2022-05-30 by torntrousers