espressif/arduino-esp32 #6820: Fire-Beetle ESP32; No COM Port in Arduino IDE

Kgray44, 2022-05-29:
### Board

Fire Beetle ESP32-E

### Device Description

The ESP32 is wired to a motor driver, ultrasonic sensor, and a current sensor.

### Hardware Configuration

D2, D3, D5, D6, D7, D10, D11, D13 are wired to the motor driver.
D12 is wired to a hall sensor.
A0 and A1 are wired to the current sensor to read current and voltage.
A2, A3, and A4 are wired to three MOSFETs that control a RGB LED strip.
Power is supplied via a L7805CV-DG with a 10 µF cap on the input and output to the VIN pin.

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

macOS 12.4

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am trying to program the ESP32, but it doesnt show up as a port under Tools / Ports.  I checked my System Information under USB, and it shows up as "USB Serial".  I have downloaded the CH340 driver, and the CP210x driver multiple times.  I have also restarted my Mac, tried three different wires (of which one has already worked for this before), tried both USB ports, re-downloaded the IDE, etc.  Nothing is working.

### Sketch

```cpp
#define BLYNK_TEMPLATE_ID "*********"
#define BLYNK_DEVICE_NAME "DrivewayESP"
#define BLYNK_AUTH_TOKEN "************************"
#define BLYNK_PRINT Serial

//#include <ESP8266WiFi.h>
//#include <BlynkSimpleEsp8266.h>
//#include <ESP8266mDNS.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <BlynkSimpleEsp32.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#define led           D9  //D9
#define motorAspeed   12  //D13
#define motorBspeed   16  //D11
#define motorAdir     D6  //D6
#define motorBdir     D5  //D5
#define hall          4   //D12
#define ultra1        19
#define ultra2        22
#define voltIn        A0
[... 294 lines cut ...]
```

### Debug Message

```plain
Sketch uses 757710 bytes (57%) of program storage space. Maximum is 1310720 bytes.
Global variables use 41588 bytes (12%) of dynamic memory, leaving 286092 bytes for local variables. Maximum is 327680 bytes.
/Users/user/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool --chip esp32 --port /dev/cu.usbserial-1410 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/partitions/boot_app0.bin 0x1000 /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 /var/folders/2y/mn13ny595dzftc5swsy9_7940000gn/T/arduino_build_181944/Hall_Driveway_Gates_mar30a.ino.bin 0x8000 /var/folders/2y/mn13ny595dzftc5swsy9_7940000gn/T/arduino_build_181944/Hall_Driveway_Gates_mar30a.ino.partitions.bin 
esptool.py v3.0-dev
Traceback (most recent call last):
  File "esptool.py", line 3969, in <module>
Serial port /dev/cu.usbserial-1410
  File "esptool.py", line 3962, in _main
  File "esptool.py", line 3551, in main
  File "esptool.py", line 271, in __init__
  File "serial/__init__.py", line 88, in serial_for_url
  File "serial/serialposix.py", line 268, in open
serial.serialutil.SerialException: [Errno 2] could not open port /dev/cu.usbserial-1410: [Errno 2] No such file or directory: '/dev/cu.usbserial-1410'
Failed to execute script esptool
An error occurred while uploading the sketch
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-05-31:
@Kgray44 - This is not an Arduino issue.

But from the schematics, I can see that this board uses CH340K chip for UART-USB communication. 
https://dfimg.dfrobot.com/nobody/wiki/fd28d987619c16281bdc4f40990e5a1c.PDF

The driver can be downloaded from DFRobot site:
https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654#target_10

Quick search about CH340K and MacOS resulted in:
There is a reference to issues with the CH340K and MacOS here: https://github.com/LilyGO/LILYGO-T-OI/issues/3#issuecomment-648293224
Possible solution? https://github.com/LilyGO/LILYGO-T-OI/issues/3#issuecomment-948289389

Kgray44, 2022-05-31:
> @Kgray44 - This is not an Arduino issue.
> 
> But from the schematics, I can see that this board uses CH340K chip for UART-USB communication. https://dfimg.dfrobot.com/nobody/wiki/fd28d987619c16281bdc4f40990e5a1c.PDF
> 
> The driver can be downloaded from DFRobot site: https://wiki.dfrobot.com/FireBeetle_Board_ESP32_E_SKU_DFR0654#target_10
> 
> Quick search about CH340K and MacOS resulted in: There is a reference to issues with the CH340K and MacOS here: [LilyGO/LILYGO-T-OI#3 (comment)](https://github.com/LilyGO/LILYGO-T-OI/issues/3#issuecomment-648293224) Possible solution? [LilyGO/LILYGO-T-OI#3 (comment)](https://github.com/LilyGO/LILYGO-T-OI/issues/3#issuecomment-948289389)

Thanks @SuGlider !  Sorry, I did not know where to ask this question 🤷.  I have spent at least 5 hours looking and testing to try to figure out this issues from 50 different ways.  I'll take a look at your links and see where I get.

Kgray44, 2022-05-31:
Ah ha!  Wow, of course the download in the Fire-beetle Wiki worked 🤦.  It works perfectly now.  Thanks again @SuGlider 😉!

closed 2022-05-31 by SuGlider