espressif/arduino-esp32 #6827: ESP32代码在1.0.6版本上能正常运行，但升级到2.0.3版本就出现看门狗复位重启

aithinker2222, 2022-05-31:
### Board

ESP32 Dev module

### Device Description

ESP32nodemcu开发板

### Hardware Configuration

外接了一个lora模块  芯片使用SX1278

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

代码在1.0.6库版本上运行是正常的 但是在2.0.3版本运行就出现了看门狗复位  这个属于必现问题

### Sketch

```cpp
/*
  LoRa Simple Gateway/Node Exemple

  This code uses InvertIQ function to create a simple Gateway/Node logic.

  Gateway - Sends messages with enableInvertIQ()
          - Receives messages with disableInvertIQ()

  Node    - Sends messages with disableInvertIQ()
          - Receives messages with enableInvertIQ()

  With this arrangement a Gateway never receive messages from another Gateway
  and a Node never receive message from another Node.
  Only Gateway to Node and vice versa.

  This code receives messages and sends a message every second.

  InvertIQ function basically invert the LoRa I and Q signals.

  See the Semtech datasheet, http://www.semtech.com/images/datasheet/sx1276.pdf
  for more on InvertIQ register 0x33.

  created 05 August 2018
  by Luiz H. Cassettari
*/

#include <SPI.h>              // include libraries
#include <LoRa.h>

const long frequency = 470E6;  // LoRa Frequency

const int csPin = 22;          // LoRa radio chip select
const int resetPin = 17;        // LoRa radio reset
const int irqPin = 26;          // change for your board; must be a hardware interrupt pin

void setup() {
  Serial.begin(9600);                   // initialize serial
  while (!Serial);

  LoRa.setPins(csPin, resetPin, irqPin);

  if (!LoRa.begin(frequency)) {
    Serial.println("LoRa init failed. Check your connections.");
    while (true);                       // if failed, do nothing
  }

  Serial.println("LoRa init succeeded.");
  Serial.println();
  Serial.println("LoRa Simple Gateway");
  Serial.println("Only receive messages from nodes");
  Serial.println("Tx: invertIQ enable");
  Serial.println("Rx: invertIQ disable");
  Serial.println();

  LoRa.onReceive(onReceive);
  LoRa.onTxDone(onTxDone);
  LoRa_rxMode();
}

void loop() {
  if (runEvery(5000)) { // repeat every 5000 millis

    String message = "HeLoRa World! ";
    message += "I'm a Gateway! ";
    message += millis();
    LoRa_sendMessage(message); // send a message

  }
}

void LoRa_rxMode(){
  LoRa.disableInvertIQ();               // normal mode
  LoRa.receive();                       // set receive mode
}

void LoRa_txMode(){
  LoRa.idle();                          // set standby mode
  LoRa.enableInvertIQ();                // active invert I and Q signals
}

void LoRa_sendMessage(String message) {
  LoRa_txMode();                        // set tx mode
  LoRa.beginPacket();                   // start packet
  LoRa.print(message);                  // add payload
  LoRa.endPacket(true);                 // finish packet and send it
}

void onReceive(int packetSize) {
  String message = "";

  while (LoRa.available()) {
    message += (char)LoRa.read();
  }

  Serial.print("Gateway Receive: ");
  Serial.println(message);
}

void onTxDone() {
  Serial.println("TxDone");
  LoRa_rxMode();
}

boolean runEvery(unsigned long interval)
{
  static unsigned long previousMillis = 0;
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    return true;
  }
  return false;
}   这个是网关部分

/*
  LoRa Simple Gateway/Node Exemple

  This code uses InvertIQ function to create a simple Gateway/Node logic.

  Gateway - Sends messages with enableInvertIQ()
          - Receives messages with disableInvertIQ()

  Node    - Sends messages with disableInvertIQ()
          - Receives messages with enableInvertIQ()

  With this arrangement a Gateway never receive messages from another Gateway
  and a Node never receive message from another Node.
  Only Gateway to Node and vice versa.

  This code receives messages and sends a message every second.

  InvertIQ function basically invert the LoRa I and Q signals.

  See the Semtech datasheet, http://www.semtech.com/images/datasheet/sx1276.pdf
  for more on InvertIQ register 0x33.

  created 05 August 2018
  by Luiz H. Cassettari
*/

#include <SPI.h>              // include libraries
#include <LoRa.h>

const long frequency = 470E6;  // LoRa Frequency

const int csPin = 22;          // LoRa radio chip select
const int resetPin = 17;        // LoRa radio reset
const int irqPin = 26;          // change for your board; must be a hardware interrupt pin

void setup() {
  Serial.begin(9600);                   // initialize serial
  while (!Serial);

  LoRa.setPins(csPin, resetPin, irqPin);

  if (!LoRa.begin(frequency)) {
    Serial.println("LoRa init failed. Check your connections.");
    while (true);                       // if failed, do nothing
  }

  Serial.println("LoRa init succeeded.");
  Serial.println();
  Serial.println("LoRa Simple Node");
  Serial.println("Only receive messages from gateways");
  Serial.println("Tx: invertIQ disable");
  Serial.println("Rx: invertIQ enable");
  Serial.println();

  LoRa.onReceive(onReceive);
  LoRa.onTxDone(onTxDone);
  LoRa_rxMode();
}

void loop() {
  if (runEvery(1000)) { // repeat every 1000 millis

    String message = "HeLoRa World! ";
    message += "I'm a Node! ";
    message += millis();
    LoRa_sendMessage(message); // send a message
  }
}

void LoRa_rxMode(){
  LoRa.enableInvertIQ();                // active invert I and Q signals
  LoRa.receive();                       // set receive mode
}

void LoRa_txMode(){
  LoRa.idle();                          // set standby mode
  LoRa.disableInvertIQ();               // normal mode
}

void LoRa_sendMessage(String message) {
  LoRa_txMode();                        // set tx mode
  LoRa.beginPacket();                   // start packet
  LoRa.print(message);                  // add payload
  LoRa.endPacket(true);                 // finish packet and send it
}

void onReceive(int packetSize) {
  String message = "";

  while (LoRa.available()) {
    message += (char)LoRa.read();
  }

  Serial.print("Node Receive: ");
  Serial.println(message);
}

void onTxDone() {
  Serial.println("TxDone");
  LoRa_rxMode();
}

boolean runEvery(unsigned long interval)
{
  static unsigned long previousMillis = 0;
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    return true;
  }
  return false;
}  这个是节点部分
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
10:18:00.518 -> 
10:18:00.518 -> Core  1 register dump:
10:18:00.553 -> PC      : 0x4008a390  PS      : 0x00060b35  A0      : 0x800895e2  A1      : 0x3ffbecac  
10:18:00.622 -> A2      : 0x3ffb8a00  A3      : 0x3ffb8890  A4      : 0x00000004  A5      : 0x00060b23  
10:18:00.725 -> A6      : 0x00060b23  A7      : 0x00000001  A8      : 0x3ffb8890  A9      : 0x00000018  
10:18:00.828 -> A10     : 0x3ffb8890  A11     : 0x00000018  A12     : 0x3ffc184c  A13     : 0x00060b23  
10:18:00.930 -> A14     : 0x007beeb8  A15     : 0x003fffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
10:18:01.033 -> EXCVADDR: 0x00000000  LBEG    : 0x400860d1  LEND    : 0x400860e1  LCOUNT  : 0xfffffffb  
10:18:01.137 -> Core  1 was running in ISR context:
10:18:01.171 -> EPC1    : 0x400da647  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
10:18:01.241 -> 
10:18:01.276 -> 
10:18:01.276 -> Backtrace:0x4008a38d:0x3ffbecac |<-CORRUPTED
10:18:01.276 -> 
10:18:01.276 -> 
10:18:01.276 -> Core  0 register dump:
10:18:01.310 -> PC      : 0x4008a507  PS      : 0x00060035  A0      : 0x8008920b  A1      : 0x3ffbe80c  
10:18:01.412 -> A2      : 0x3ffbeeb8  A3  ⸮
```

### Other Steps to Reproduce

使用1.0.6版本没有出现该问题

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-05-31:
Please explain your issue into English.

SuGlider, 2022-05-31:
@aithinker2222 - There are 2 sketches, one for the Gateway and another for the Node. Which one has the WDT issue?

aithinker2222, 2022-05-31:
> @aithinker2222- 有 2 个可能有一个 DT 用于，另一个网关 。
> 
> 我认为 WDT 可能与图书馆代码的这部分相关： [https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L361-L396 ](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L361-L396) [https://github.com/sandeepmistry/ arduino-LoRa/blob/master/src/LoRa.cpp#L692-L752](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L692-L752)
> 
> 它与执行 SPI 事务可能相关的 GPIO ISR 有关。 `attachInterrupt(digitalPinToInterrupt(_dio0), LoRaClass::onDio0Rise, RISING);`+ SPI 事务执行。
> 
> 我这里没有用于测试的 LoRa 模块。验证和调试它是必需的。

How can I fix this since running on version 1.0.6 there is no problem

SuGlider, 2022-05-31:
As I said, it is just my guess that the issue is related to GPIO interrupt code running SPI transactions.

I see that the LoRa Module needs SPI. 
ESP32 SPI default pins are 23(MOSI), 19(MISO), 18(CLK) and 5 (SS).  In the code is seems that it uses SS = 22 
`const int csPin = 22;          // LoRa radio chip select`

Maybe changing the code to use pin 5 as the SS may solve it... please try it and let me know.
Also please check if the SPI pins are those default ones.

SuGlider, 2022-06-01:
I'd need this LoRa module to be able to debug it... maybe it'd be possible to simulate its GPIO ISR with SPI function - not sure.

Sorry for not being able to help.

aithinker2222, 2022-06-02:
> I'd need this LoRa module to be able to debug it... maybe it'd be possible to simulate its GPIO ISR with SPI function - not sure.
> 
> Sorry for not being able to help.

Are you working in Shanghai? I can send you several Lora modules if necessary

me-no-dev, 2022-06-02:
I think that the problem is that in the pin interrupt routine there is SPI code. SPI is not made to be used in ISR routines on ESP32, it uses FreeRTOS semaphores and is most probably why you get ISR WDT. You need to spin a task, which will wait for the interrupt to trigger and then do the SPI stuff inside. You can use FreeRTOS Queue or Semaphore to signal the task that ISR has been received.

SuGlider, 2022-06-02:
> I think that the problem is that in the pin interrupt routine there is SPI code. SPI is not made to be used in ISR routines on ESP32, it uses FreeRTOS semaphores and is most probably why you get ISR [...]

It is very possible to be the issue with WDT, as you say.

@aithinker2222 says this same LoRa Library code runs fine with 1.0.6.
Do you know if something has changed that would generate the WDT with 2.0.0+?

me-no-dev, 2022-06-03:
SPI has had locks for a while, so I am not sure why it even ran on 1.0.6. Probably some change in FreeRTOS

VojtechBartoska, 2022-06-14:
@aithinker2222 Can I consider this as solved?

SuGlider, 2022-06-15:
@VojtechBartoska - I'll try to reproduce the issue using a pair of ESP32 and emulate the LoRa behavior. 
Another option is to get a pair of modules from @aithinker2222 

>Are you working in Shanghai? I can send you several Lora modules if necessary

No... but a colleague from Shanghai may help in sending it to my place, if necessary.

aithinker2222, 2022-06-16:
There is no need to solve the problem at present. I have added a watchdog reset program by myself. At present, the code works well

closed 2022-06-16 by aithinker2222

JamshidJD, 2022-06-29:
> There is no need to solve the problem at present. I have added a watchdog reset program by myself. At present, the code works well

Can you please guide on how and what you added to control this problem.
I'm having a similar problem when using BLE and FileWrite.
Also the basic OTA version Fails. 
So there seems to be something changed with the core.

aithinker2222, 2022-06-30:
> 

https://blog.csdn.net/jianfengbeyond/article/details/123979350

aithinker2222, 2022-06-30:
> Can you please guide on how and what you added to control this problem. I'm having a similar problem when using BLE and FileWrite. Also the basic OTA version Fails. So there seems to be something [...]

Check out the link I sent you

JamshidJD, 2022-06-30:
> 

I tried your snippet with the very basic code and still faced Resets
#include "soc/rtc_wdt.h"

void setup(void) {
    rtc_wdt_protect_off();
  rtc_wdt_enable();
  rtc_wdt_feed();
   rtc_wdt_set_time(RTC_WDT_STAGE0, 7000);
  /*
Alternatively I TRIED BUT THIS ALSO FAILED TO STOP RESETS
 rtc_wdt_protect_off();
  rtc_wdt_disable();
  disableCore0WDT();
  disableLoopWDT();

*/
}

void loop(void) {
  while(1)
  {}
}

////////////////////////////////////////////////////////////////
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
Jul 29 2019 12:21:46