espressif/arduino-esp32 #6832: The HiFreq_ADC Example Code cannot run.

Globefishp, 2022-06-03:
### Board

WEMOS LOLIN32 LITE

### Device Description

just on breadboard.

### Hardware Configuration

a Potentiometer is connected to my broad and used as Analogue input.

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 10 1809

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

The example code <HiFreq_ADC.ino> can compile and upload to board, but Serial output says it is continuously rebooting. 

Here is my serial output, the backtrace tool cannot decode that(probably my usage is wrong, i just paste that to the tool.)
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
Output frequency: 78125

assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace:0x4008306d:0x3ffb2dc00x40087ebd:0x3ffb2de0 0x4008cd1d:0x3ffb2e00 0x400886b2:0x3ffb2f30 0x4008465a:0x3ffb2f70 0x40084715:0x3ffb2fa0 0x400d6f1e:0x3ffb2fc0 0x400d2e42:0x3ffb2ff0 0x400d131f:0x3ffb3020 

ELF file SHA256: 0000000000000000
```

Beside, when compile the code, there is a warning says:
```
C:\Users\Globefish\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\ESP32\examples\I2S\HiFreq_ADC\HiFreq_ADC.ino: In function 'void i2sInit()':
C:\Users\Globefish\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\ESP32\examples\I2S\HiFreq_ADC\HiFreq_ADC.ino:23:29: warning: 'I2S_COMM_FORMAT_I2S_MSB' is deprecated [-Wdeprecated-declarations]
     .communication_format = I2S_COMM_FORMAT_I2S_MSB,
                             ^~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\Globefish\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from C:\Users\Globefish\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\ESP32\examples\I2S\HiFreq_ADC\HiFreq_ADC.ino:6:
C:\Users\Globefish\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:88:5: note: declared here
     I2S_COMM_FORMAT_I2S_MSB   __attribute__((deprecated)) = 0x01, /*!< I2S format MSB, (I2S_COMM_FORMAT_I2S |I2S_COMM_FORMAT_I2S_MSB) correspond to `I2S_COMM_FORMAT_STAND_I2S`*/
     ^~~~~~~~~~~~~~~~~~~~~~~

```
Seems `I2S_COMM_FORMAT_I2S_MSB` need to be replaced by other attribute?

I'm new to ESP32 and Arduino so I would like to learn ADC with DMA from example code but unfortunately it cannot work.
Any one can help me figure out a solution?

### Sketch

```cpp
/*
 * This is an example to read analog data at high frequency using the I2S peripheral
 * Run a wire between pins 27 & 32
 * The readings from the device will be 12bit (0-4096) 
 */
#include <driver/i2s.h>

#define I2S_SAMPLE_RATE 78125
#define ADC_INPUT ADC1_CHANNEL_4 //pin 32
#define OUTPUT_PIN 27
#define OUTPUT_VALUE 3800
#define READ_DELAY 9000 //microseconds

uint16_t adc_reading;

void i2sInit()
{
   i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
    .sample_rate =  I2S_SAMPLE_RATE,              // The format of the signal using ADC_BUILT_IN
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // is fixed at 12bit, stereo, MSB
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 4,
    .dma_buf_len = 8,
    .use_apll = false,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0
   };
   i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
   i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);
   i2s_adc_enable(I2S_NUM_0);
}

void reader(void *pvParameters) {
  uint32_t read_counter = 0;
  uint64_t read_sum = 0;
// The 4 high bits are the channel, and the data is inverted
  uint16_t offset = (int)ADC_INPUT * 0x1000 + 0xFFF;
  size_t bytes_read;
  while(1){
    uint16_t buffer[2] = {0};
    i2s_read(I2S_NUM_0, &buffer, sizeof(buffer), &bytes_read, 15);
    //Serial.printf("%d  %d\n", offset - buffer[0], offset - buffer[1]);
    if (bytes_read == sizeof(buffer)) {
      read_sum += offset - buffer[0];
      read_sum += offset - buffer[1];
      read_counter++;
    } else {
      Serial.println("buffer empty");
    }
    if (read_counter == I2S_SAMPLE_RATE) {
      adc_reading = read_sum / I2S_SAMPLE_RATE / 2;
      //Serial.printf("avg: %d millis: ", adc_reading);
      //Serial.println(millis());
      read_counter = 0;
      read_sum = 0;
      i2s_adc_disable(I2S_NUM_0);
      delay(READ_DELAY);
      i2s_adc_enable(I2S_NUM_0);
    }
  }
}

void setup() {
  Serial.begin(115200);
  // Put a signal out on pin 
  uint32_t freq = ledcSetup(0, I2S_SAMPLE_RATE, 10);
  Serial.printf("Output frequency: %d\n", freq);
  ledcWrite(0, OUTPUT_VALUE/4);
  ledcAttachPin(OUTPUT_PIN, 0);
  // Initialize the I2S peripheral
  i2sInit();
  // Create a task that will read the data
  xTaskCreatePinnedToCore(reader, "ADC_reader", 2048, NULL, 1, NULL, 1);
}

void loop() {
  delay(1020);
  Serial.printf("ADC reading: %d\n", adc_reading);
  delay(READ_DELAY);
}
```

### Debug Message

```plain
Decoding stack results
0x4008306d: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x40087ebd: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 128
0x4008cd1d: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 85
0x400886b2: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 820
0x4008465a: lock_release_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 201
0x40084715: _lock_release at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 207
0x400d6f1e: adc1_lock_release at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/adc_common.c line 423
0x400d2e42: i2s_adc_disable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c line 1364
0x400d131f: reader(void*) at C:\Users\Globefish\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\ESP32\examples\I2S\HiFreq_ADC/HiFreq_ADC.ino line 59
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-06-03:
i2s_adc_disable is causing the crash.  If you remark that line and the subsequent enable, it will work.
Something has definitely changed in how that disable is handled.  It seems like perhaps it is trying to access a queue or semaphore that doesn't exist.  Maybe it is a scoping issue- if you put an i2s_adc_enable in before the disable it doesn't crash (but it also doesn't work :smile:)

P-R-O-C-H-Y, 2022-06-03:
@PilnyTomas Can you take a look?

lbernstone, 2022-06-15:
Is the I2S plotter example using continuous ADC?  If so, drop this example in favor of using the API.

PilnyTomas, 2022-06-20:
Hi @Globefishp,
regarding the deprecated settings - you can take a look in [this file](https://github.com/espressif/esp-idf/blob/e8d172eb87e606c52c6a995d118c1eb0f4810abc/components/hal/include/hal/i2s_types.h#L79-L93) to see which constants are supported, but this is least of the problems.
There seems to be a bug in the lock while disabling ADC.
I will take look and try to find out why is this happened and how to fix it.

VojtechBartoska, 2022-07-11:
Closing as solved, PR was merged.

closed 2022-07-11 by VojtechBartoska