espressif/arduino-esp32 #6835: Inconsistency in setTimeout implementation

cziter15, 2022-06-03:
### Board

Doesn't matter, affect all ESP32 boards

### Device Description

-

### Hardware Configuration

-

### Version

v2.0.3

### IDE Name

Visual Micro

### Operating System

Windows 11

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

**This is not really a bug to be honest. It's rather small inconsistency that can cause troubles and might be somehow tweaked in arduino-esp32**

In development of my hobby IoT framework I've encountered inconsistent behavior between Arduino for ESP32 and ESP8266.

The problem is that everywhere in Arduino framework timeout passed to TCP clients is passed in units of milliseconds. Also ESP8266's WiFiClient based requires milliseconds input. However, ESP32s' WiFiClient doesn't follow that rule and requires seconds. This is described in header files, but users porting existing code might not notice that fact.

https://www.arduino.cc/reference/en/libraries/ethernet/client.setconnectiontimeout/
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClient.h

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiClient.h

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-06-03:
Unfortunately, this is seriously breaking behavior for people that already have things set for seconds.  Perhaps this could be implemented in stages, with a note indicating that the behavior will be deprecated at some point- temporarily put in a change so that numbers over 600 are interpreted as milliseconds, and then anything below as seconds, with a warning that the behavior will change.  Then once the change has had time to soak, change over to milliseconds only.

SuGlider, 2022-06-04:
@me-no-dev / @VojtechBartoska - I think we could do as @lbernstone suggested starting for the release of Core 2.1.0.
What do you think about it?

petrkr, 2022-06-06:
Compared how many warnings are during compilation.. I am afraid nobody will notice just one more warning about deprecation.

lbernstone, 2022-06-06:
I'd say 6-8 months for notification.  It will be quite obvious once it changes.  I think most people that actively set it are using timeouts less than 60 seconds, so when every connection fails, I'd expect they read the compile log or go look at the header and see that it has changed to milliseconds...

petrkr, 2022-06-06:
> Warnings should be fixed (unless you mean from other libs or your code

I agree, but lot of ArduinoIDE peoples are not developers but only glue-coders from stack overflow. And they do not understand to that black window output...

And still I do not take there are lot of warnings from other libs and code and this one will really be hidden in mass output... it should be somehow better visible. But in IDE do not know how...

What about use logger? As other debug messages take severity 'warn' and print it out for few releases by logging subsystem. So if someone will be debug why their code works 'strage' this increased log level will warn them

lbernstone, 2022-06-06:
You can't make everybody happy.  Noobs are unlikely to be setting a timeout in WiFiClient.  It is a breaking change, so it is going to break things at some point, and then people will look to see what is failing.  A warning in your compile regarding the piece that is breaking should be pretty clear.

VojtechBartoska, 2022-06-07:
this relates to open Pull Request #6676

VojtechBartoska, 2022-06-07:
I'm adding this to our Roadmap and also under 2.1.0 milestone.

EdwardCooke, 2022-09-29:
When this is implemented will we be able to set a timeout less than 1 second?

EdwardCooke, 2022-09-29:
What about a new method, `setTimeoutMs` and a warning/error on `setTimeout`?

cziter15, 2022-09-29:
There is connect method which should be considered here too.

supcik, 2023-08-14:
Units in computer science are not easy to handle (remember [Mars Orbiter Crash](https://en.wikipedia.org/wiki/Mars_Climate_Orbiter)).

What do you about overriding the method with an argument of type [`std::chrono::duration`](https://en.cppreference.com/w/cpp/chrono/duration)? It does not solve the inconsistency, but personally, I would prefer this instead of several methods such as `setTimeoutMs` and `setTimeoutSec`.

cziter15, 2023-08-14:
> Units in computer science are not easy to handle (remember [Mars Orbiter Crash](https://en.wikipedia.org/wiki/Mars_Climate_Orbiter)).
>
> What do you about overriding the method with an argument of [...]

Might be problematic, as the direction taken is to avoid std library extensions, but someone from the core team should confirm this.

VojtechBartoska, 2023-12-21:
Closing this ticker as solved, mentioned PR have been merged.

closed 2023-12-21 by VojtechBartoska