espressif/arduino-esp32 #6837: ESP32-S2 Touch read values do not change

wacon44, 2022-06-05:
### Board

ESP32-S2 Feather

### Device Description

Using SPI to drive display, temp sensor and SD drive.  Physical pins do not conflict with this function.

### Hardware Configuration

Using SPI to drive display, temp sensor and SD drive.

### Version

v2.0.3

### IDE Name

Arduino IDE 2.0

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

15200

### Description

The TouchRead function returns fixed values that do not change when the respective pin is touched.  The values also seem to be larger than expected.  With the simple sketch the following are typical return values.
Touch T11 = 2416619
Touch T12 = 1744852
Touch T13 = 4189143
Touch T3 = 3667959
Touch T4 = 3440595
The values will change when restarting the sketch. There are no reported errors or other abnormal behavior.  

### Sketch

```cpp
// ESP32 Touch Test

int testCntr;
void setup()
{
  Serial.begin(9600);
  delay(1000); // give me time to bring up serial monitor
  Serial.println("ESP32 Touch Test");
  testCntr = 0;
}

void loop()
{
  Serial.print("Touch T11 = "); Serial.println(touchRead(11));  // get value
  Serial.print("Touch T12 = "); Serial.println(touchRead(12));  // get value
  Serial.print("Touch T13 = "); Serial.println(touchRead(13));  // get value
  Serial.print("Touch T3 = "); Serial.println(touchRead(3));  // get value
  Serial.print("Touch T4 = "); Serial.println(touchRead(4));  // get value
  Serial.print("Counter="); Serial.println(testCntr);
  testCntr++;
  delay(1000);
}
```

### Debug Message

```plain
None.
```

### Other Steps to Reproduce

I have also attempted to use the interrupt method to detect a touch event, with no results.  All other features on the ESP32 seem to work as expected (using different sketch).  I have also tried only activating a single touch pin, with similar results.  I first brought this issue to the attention of Adafruit staff, they recommended I forward to this site.  

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-06-06:
@SuGlider Please help with triage, thanks!

SuGlider, 2022-06-06:
@wacon44 - I used Arduino IDE 1.8.15 with Arduino Core 2.0.3, using a ESP32-S2-DevkitC-1 (ESP32-S2-Solo) board.

T11 goes from 10095-10100 when untouched to 12000..43000 when touched
T12 goes from 10620-10635 when untouched to 20000..68000 when touched
T13 goes from 11185-11196 when untouched to 15000..75000 when touched
T3 goes from 9250-9255 when untouched to 10000..65000 when touched
T4 goes from 7020-7028 when untouched to 20000..110000 when touched

Thus I read very different values compared to your report.

I also run the examples from `Files->Examples->ESP32->Touch->TouchButtonV2`, which is the proper examples for S2 and S3. 
It runs fine with T3, T4, T11, T12, T13.
This example tells you presicely when the Touchpad was touched and when it was released.

SuGlider, 2022-06-06:
Also tested with a ESP32-S2-WROOM and Also tested with a ESP32-S2-WROOM and it works fine.
The values for T3, T4, T11, T12, T13 are similar... 
All of them go from about 7000..12000 when untouched to about over 50000 when touched.
The value read will depend on physical conditions that may have an impact on capacitance, such as if the pad has a wire connect to it and so on.it works fine.
The values for T3, T4, T11, T12, T13 are a bit different now... 
All of them go from about 7000..12000 when untouched to about over 50000 when touched.
The read alue will depend on physical conditions that may have an impact on capacitance, such as if the pad has a wire connect to it and so on.

wacon44, 2022-06-06:
I understand the range of expected values, however the values reported are
magnitudes greater in the static no-touch state.  Rather than the 50,000 I
am seeing values around 2-4 million.  Is there a possibility the device is
not setup properly with the ESP32 v2.0.3 package?

Wayne Connell
K7QHQ
Echo Lake Community Group, Admin

Echo Lake, WA
Where all the women are strong, all the men are good looking, and all the
children are above average.

SuGlider, 2022-06-06:
@wacon44  - Could you try it with the Arduino IDE 1.8,x and make sure it has the ESP32 Core 2.0.3 installed?
It may be possible that something is wrong with the installation on the IDE 2.0 ... not sure.
Thanks!

wacon44, 2022-06-06:
Outstanding!  I installed the Arduino 1.8.19 with the ESP32 Core 2.0.3.
Compiled and ran the touch read example.  It seems to work as expected.  So
it seems the Arduino IDE 2.0... has an issue with the ESP Core.  While the
older version of Arduino is clunky it does work especially with this
feature.  What do you recommend I do to help fix the incompatibility and
allow the use of the Arduino IDE with the ESP Core?

Wayne Connell
K7QHQ
Echo Lake Community Group, Admin

Echo Lake, WA
Where all the women are strong, all the men are good looking, and all the
children are above average.

SuGlider, 2022-06-07:
@wacon44 - Arduino IDE 2.0 is still in Beta. I think it will be fully supported by ESP32 Arduino as soon as its final release goes out.
Not sure how what was the issue with it and 2.0.3.

SuGlider, 2022-06-09:
Closing the issue. Feel free to reopen it.

closed 2022-06-09 by SuGlider

hykilpikonna, 2024-05-11:
I have the same issue, but neither using 1.8.19 nor using arduino core 2.0.3 fixes the issue.