espressif/arduino-esp32 #6843: AUTH_EXPIRE sometimes

collin80, 2022-06-08:
### Board

ESP32 WROOM32

### Device Description

It's a custom battery management system with a custom shield and a custom board but using an ESP32WROOM32 module.

### Hardware Configuration

WROOM32 on a custom board speaking to batteries over TTL serial, connected over wifi to a PC, hosts a website

### Version

v2.0.3

### IDE Name

Arduino IDE 1.8

### Operating System

Ubuntu 22.04

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I've been using the same basic code for a long time. In the past I'd get issues where about every other time it would fail to connect to my AP. But, I've been using 2.0.2 for a while and it seemed not to do that. However, with 2.0.3 I now get AUTH_EXPIRED as the reason for failing to connect. But, it only happens sometimes. It's probably about 40% of the time. It's nearly every other time but not consistently. It seems semi-random. I saw something like this get reported in #6767 but I'm using the standard ESP32 not the newer ESP32-C3. 

### Sketch

```cpp
sketch is huge but any generic wifi connecting sketch seems to trigger this
```

### Debug Message

```plain
20:31:40.721 -> [ 18242][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
20:31:40.721 -> [ 18242][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 2 - AUTH_EXPIRE
20:31:40.721 -> [ 18245][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
20:31:40.754 -> 
20:31:40.754 -> STA Disconnected
```

### Other Steps to Reproduce

Run my program which connects to wifi, unplug USB, plug it back in. There's a good chance the error will happen. Unplug, plug back in, works now in all likelihood. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

collin80, 2022-06-08:
Scratch that... 2.0.2 and 2.0.1 do the exact same thing. I suspect my router is being evil. Closing the issue because it's almost certainly on my end... Nothing to see here.

closed 2022-06-08 by collin80

guestisp, 2023-09-04:
Same here.
And starting from the very latest version (i think) the AUTH_EXPIRE is followed by this:
```
22:10:06.208 > [  9965][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
22:10:06.208 > [  9973][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 201 - NO_AP_FOUND
```