espressif/arduino-esp32 #6847: BluetoothSerial  phone can pair but cannot connect

liangzhichao63, 2022-06-08:
### Board

ESP32 PICO KIT

### Device Description

ESP32 PICO KIT 

### Hardware Configuration

Empty development board

### Version

v1.0.6

### IDE Name

arduino ide

### Operating System

win10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Arduino  bluetoothserial code example, you can pair but cannot connect.

### Sketch

```cpp
#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

### Debug Message

```plain
No Debug Message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-06-09:
have you tried with version 2.0.3?

liangzhichao63, 2022-06-09:
Yes, I have. The connection works when using BLE.But the BluetoothSerial was not working properly.

SuGlider, 2022-06-10:
@liangzhichao63

I have just made it work correctly with a ESP32-DevKit board and Arduino Core 2.0.3.
Basically, I just paired it in Win11 and then open the Device Manager to see the COM port associated to the Bluetooth Serial connection. Finally I open 2 Putty windows, one for the ESP32-UART and another for the Bluetooth COM port.

When I type in one of the Putty windows, it displays in the other one, as expected.

SuGlider, 2022-06-10:
[image]

[image]

liangzhichao63, 2022-06-14:
It may be that the ESP32 PICO D4 chip is not adapted, thank you for helping me verify.

VojtechBartoska, 2022-06-14:
@liangzhichao63 Can I consider this as solved?

VojtechBartoska, 2022-07-26:
Closing as expired, explanation was provided and there were not further questions.

closed 2022-07-26 by VojtechBartoska