espressif/arduino-esp32 #6848: WifiClientBasic not pairing since 2.0.2 with ESP32CAM error 201 NO_AP_FOUND

atsju, 2022-06-08:
### Board

[ESP32-CAM-MB](https://fr.aliexpress.com/item/32981449079.html?spm=a2g0o.order_list.0.0.21ef5e5bLHVetK&gatewayAdapt=glo2fra)

### Device Description

This is the exact board : [ESP32-CAM-MB](https://fr.aliexpress.com/item/32981449079.html?spm=a2g0o.order_list.0.0.21ef5e5bLHVetK&gatewayAdapt=glo2fra) 
No modification or connection done on board.

### Hardware Configuration

No modification, no addiction

### Version

v2.0.3

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

460800

### Description

## Problem :
from version 2.0.2 and up, wifiClientBasic example is not able to connect to my box neither to my phone. Works fine with versions <=2.0.1

See debug message for details of error.

### Sketch

```cpp
/*
 *  This sketch sends a message to a TCP server
 *
 */

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti WiFiMulti;

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network
    WiFiMulti.addAP("SSID", "passpasspass");

    Serial.println();
    Serial.println();
    Serial.print("Waiting for WiFi... ");

    while(WiFiMulti.run() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    delay(500);
}

void loop()
{
//    const uint16_t port = 80;
//    const char * host = "192.168.1.1"; // ip or dns
    const uint16_t port = 1337;
    const char * host = "192.168.1.10"; // ip or dns

    Serial.print("Connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;

    if (!client.connect(host, port)) {
        Serial.println("Connection failed.");
        Serial.println("Waiting 5 seconds before retrying...");
        delay(5000);
        return;
    }

    // This will send a request to the server
    //uncomment this line to send an arbitrary string to the server
    //client.print("Send this data to the server");
    //uncomment this line to send a basic document request to the server
    client.print("GET /index.html HTTP/1.1\n\n");

  int maxloops = 0;

  //wait for the server's reply to become available
  while (!client.available() && maxloops < 1000)
  {
    maxloops++;
    delay(1); //delay 1 msec
  }
  if (client.available() > 0)
  {
    //read back one line from the server
    String line = client.readStringUntil('\r');
    Serial.println(line);
  }
  else
  {
    Serial.println("client.available() timed out ");
  }

    Serial.println("Closing connection.");
    client.stop();

    Serial.println("Waiting 5 seconds before restarting...");
    delay(5000);
}
```

### Debug Message

```plain
## With version 2.0.1 (no modification of log level)

Waiting for WiFi... .
WiFi connected
IP address: 
192.168.1.67
```

## With version 2.0.3 (max log)

SSID and other info eddited for privacy reasons

```
[   842][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   898][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: SFR_AB12

Waiting for WiFi... [   919][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  1014][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[  1016][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 2 - STA_START
[  3643][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 14
[  3643][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[  3648][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  3652][I][WiFiMulti.cpp:119] run(): [WIFI] 14 networks found
[  3658][D][WiFiMulti.cpp:149] run():  --->   0: [11][12:13:14:15:16:17] SFR_AB12 (-48) *
[  3666][D][WiFiMulti.cpp:151] run():        1: [1][12:13:14:15:16:17] xxxxxxxx (-79) *
[  3674][D][WiFiMulti.cpp:151] run():        2: [11][12:13:14:15:16:17] xxxxxxxx (-84) *
[  3682][D][WiFiMulti.cpp:151] run():        3: [1][12:13:14:15:16:17] xxxxxxxx (-88) *
[  3690][D][WiFiMulti.cpp:151] run():        4: [7][12:13:14:15:16:17] xxxxxxxx (-88) *
[  3698][D][WiFiMulti.cpp:151] run():        5: [7][12:13:14:15:16:17] xxxxxxxx (-88) *
[  3707][D][WiFiMulti.cpp:151] run():        6: [6][12:13:14:15:16:17] xxxxxxxx (-89) *
[  3715][D][WiFiMulti.cpp:151] run():        7: [1][12:13:14:15:16:17] xxxxxxxx (-90) *
[  3723][D][WiFiMulti.cpp:151] run():        8: [11][12:13:14:15:16:17] xxxxxxxx (-90) *
[  3731][D][WiFiMulti.cpp:151] run():        9: [1][12:13:14:15:16:17] SFR WiFi FON (-91)  
[  3739][D][WiFiMulti.cpp:151] run():        10: [6][12:13:14:15:16:17] xxxxxxxx (-91) *
[  3748][D][WiFiMulti.cpp:151] run():        11: [11][12:13:14:15:16:17] xxxxxxxx(-91) *
[  3756][D][WiFiMulti.cpp:151] run():        12: [1][12:13:14:15:16:17] xxxxxxxx (-93) *
[  3764][D][WiFiMulti.cpp:151] run():        13: [6][12:13:14:15:16:17] xxxxxxxx (-93) *
[  3773][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 12:13:14:15:16:17 SSID: SFR_AB12 Channel: 11 (-48)
[  3875][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  6514][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: SFR_BC90, BSSID: 00:00:00:00:00:00, Reason: 201
[  6514][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  6521][W][WiFiGeneric.cpp:873] _eventCallback(): Reason: 201 - NO_AP_FOUND
[  6528][D][WiFiGeneric.cpp:889] _eventCallback(): WiFi Reconnect Running
[  6535][E][WiFiMulti.cpp:181] run(): [WIFI] Connecting Failed AP not found.
[  6536][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.[  7052][D][WiFiMulti.cpp:195] run(): [WIFI] delete old wifi config...
[  7052][D][WiFiMulti.cpp:198] run(): [WIFI] start scan
....[  8968][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 130, Status: 0, Results: 16
[  8969][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 10980][V][WiFiGeneric.cpp:317] _arduino_event_cb(): SCAN Done: ID: 131, Status: 0, Results: 16
[ 10981][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 10985][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 10990][I][WiFiMulti.cpp:119] run(): [WIFI] 16 networks found
[ 10995][D][WiFiMulti.cpp:149] run():  --->   0: [11][12:13:14:15:16:17] SFR_AB12 (-48) *
[ 11003][D][WiFiMulti.cpp:151] run():        1: [1][12:13:14:15:16:17] xxxxxxxx (-79) *
[ 11011][D][WiFiMulti.cpp:151] run():        2: [11][12:13:14:15:16:17] xxxxxxxx (-85) *
[ 11020][D][WiFiMulti.cpp:151] run():        3: [7][12:13:14:15:16:17] xxxxxxxx (-87) *
[ 11028][D][WiFiMulti.cpp:151] run():        4: [7][12:13:14:15:16:17] xxxxxxxx (-88) *
[ 11036][D][WiFiMulti.cpp:151] run():        5: [1][12:13:14:15:16:17] xxxxxxxx (-89) *
[ 11045][D][WiFiMulti.cpp:151] run():        6: [6][12:13:14:15:16:17] xxxxxxxx (-89) *
[ 11053][D][WiFiMulti.cpp:151] run():        7: [1][12:13:14:15:16:17] xxxxxxxx (-90) *
[ 11061][D][WiFiMulti.cpp:151] run():        8: [1][12:13:14:15:16:17] SFR WiFi FON (-90)  
[ 11069][D][WiFiMulti.cpp:151] run():        9: [6][12:13:14:15:16:17] xxxxxxxx (-90) *
[ 11077][D][WiFiMulti.cpp:151] run():        10: [11][12:13:14:15:16:17] xxxxxxxx (-90) *
[ 11085][D][WiFiMulti.cpp:151] run():        11: [11][12:13:14:15:16:17] xxxxxxxx (-90) *
[ 11094][D][WiFiMulti.cpp:151] run():        12: [1][12:13:14:15:16:17] xxxxxxxx (-91) *
[ 11101][D][WiFiMulti.cpp:151] run():        13: [1][12:13:14:15:16:17] xxxxxxxx (-92) *
[ 11110][D][WiFiMulti.cpp:151] run():        14: [1][12:13:14:15:16:17] xxxxxxxx (-92) *
[ 11118][D][WiFiMulti.cpp:151] run():        15: [6][12:13:14:15:16:17] xxxxxxxx (-92) *
[ 11126][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 12:13:14:15:16:17 SSID: SFR_AB12 Channel: 11 (-48)
[ 11138][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 11155][E][WiFiMulti.cpp:181] run(): [WIFI] Connecting Failed AP not found.
.[ 11656][D][WiFiMulti.cpp:195] run(): [WIFI] delete old wifi config...
[ 11657][D][WiFiMulti.cpp:198] run(): [WIFI] start scan
...
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Rotzbua, 2022-06-10:
Since `2.0.3` (https://github.com/espressif/arduino-esp32/commit/5f6d093d4a8a371c78acd6e1a0e7a95b7346cbec) the default scan method changed.
I observed this problem with a multi AP setup and same SSID.

Forcing the old behavior to scan all channels reduced the problem:
```cpp
WiFi.setScanMethod(WIFI_ALL_CHANNEL_SCAN); // Force to scan all channels to find best AP.
```

atsju, 2022-06-10:
@Rotzbua  I will try but please note I have problems with 2.0.2 also. So if this change has been introduced in 2.0.3 it's not the cause of my issue.

atsju, 2022-06-11:
`setScanMethod` did not help. However `WiFi.setMinSecurity(WIFI_AUTH_WPA_PSK);` did fix the issue with my box.

Why I said it didn't connect to my phone AP (using WPA2) is still a mistery. Maybe I did type SSID or password wrong. Example works fine by default with phone AP.

closed 2022-06-11 by atsju