espressif/arduino-esp32 #6851: WiFi.setTxPower does not work for WIFI_POWER_MINUS_1dBm

societyofrobots, 2022-06-09:
### Board

ESP32E, custom board, hardware irrelevant

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Win 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

WIFI_POWER_MINUS_1dBm found in WiFiGeneric.h isn't working, however the other power settings work fine.

### Sketch

```cpp
This reports 78 and 5, which is correct.

  Serial.printf("\nWiFi Tx Power Level: %u",WiFi.getTxPower());
  WiFi.setTxPower(WIFI_POWER_2dBm);
  Serial.printf("\nWiFi Tx Power Level changed to: %u",WiFi.getTxPower());
```

However this reports 78, changed to 78, which means it isn't working.

```
  Serial.printf("\nWiFi Tx Power Level: %u",WiFi.getTxPower());
  WiFi.setTxPower(WIFI_POWER_MINUS_1dBm);
  Serial.printf("\nWiFi Tx Power Level changed to: %u",WiFi.getTxPower());
```
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Rob58329, 2024-06-05:
Note that https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp contains

```
wifi_power_t WiFiGenericClass::getTxPower() {
  int8_t power;
  if (!WiFi.STA.started() && !WiFi.AP.started()) {
    log_w("Neither AP or STA has been started");
    return WIFI_POWER_19_5dBm;
  }
  if (esp_wifi_get_max_tx_power(&power)) {
    return WIFI_POWER_19_5dBm;
  }
  return (wifi_power_t)power;
}
```

And https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.h sets

`WIFI_POWER_19_5dBm = 78`

You did not show your full code above, but I suspect that you were trying to set & read the wifi_power before  you’d done a `“WiFi.mode(WIFI_STA)”` or `“WiFi.mode(WIFI_AP)”`, as for me this works as long as I do it after I’ve set WiFi.mode().  (You might also need a `“WiFi.begin()”` if the ESP doesn’t auto-begin this.)

You can tell if the SET command worked using:
`SerialM.printf("setTxPower=%s\n",WiFi.setTxPower((wifi_power_t)70) ? "OK" : "FAIL");`

If you had had the “Core Debug Level” enabled you would probably have also seen the warning message "Neither AP or STA has been started".

zebraonpc, 2024-06-20:
> Note that https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp contains
>
> ```
> wifi_power_t WiFiGenericClass::getTxPower() {
>   int8_t power;
>   if [...]

```
WiFi.setHostname("Admins_Cock");  
WiFi.begin(ssid, password);
WiFi.mode(WIFI_STA);
while(WiFi.status()!=WL_CONNECTED){
Serial.println("Connecting..");
delay(1000);
}
 WiFi.setSleep(WIFI_PS_NONE);
 WiFi.setTxPower(WIFI_POWER_2dBm);
```
So in my case  when i put any WIFI_POWER_dBm, exept WIFI_POWER_MINUS_1dBm which always shows 78, TxPower  shows almost correct result +-2 , but that is not the problem.
The real thing that i mesure the sigal from  my esp32-s and it is always the same no matter what Tx power i set 
i use ESP-12 (esp8266) all works correctly but this one seem like doesnt really change power output

me-no-dev, 2024-06-20:
@zebraonpc you are setting TX power and sleep too late. It should be done after WiFi is started and before it has connected. This issue should be closed, because that is fixed now in master and will be part of 3.0.2
```cpp
WiFi.setHostname("Admins_Cock");  
WiFi.mode(WIFI_STA);
WiFi.setSleep(WIFI_PS_NONE);
WiFi.setTxPower(WIFI_POWER_2dBm);
WiFi.begin(ssid, password);
while(WiFi.status()!=WL_CONNECTED){
Serial.println("Connecting..");
delay(1000);
}
```

closed 2024-06-20 by me-no-dev