espressif/arduino-esp32 #6854: Support of ESP32-H2

VojtechBartoska, 2022-06-10:


MacSass, 2022-06-10:
Just out of curiosity: Is my understanding correct that this ESP32-H2 Soc would not feature any WIFI capability?
As much as I can read, this is not part of IEEE 802.15.4, right?

me-no-dev, 2022-06-10:
correct @MacSass no WiFi :)

Suxsem, 2023-05-13:
We are starting to see ESP32-H2 being sold on aliexpress at a reasonable price, it has the potential to disrupt the IoT segment with zigbee and matter-over-thread support. Arduino IDE support will be the sparkle to have this board becoming massively popular through makers, please think about adding it!

TD-er, 2023-05-13:
IMHO is the C6 way more interesting as it has both 802.15.4 and WiFi.

Suxsem, 2023-05-13:
Yes but usually you don't use two different radio channels for IoT devices. Furthermore, the H2 is already available and fills a big gap in DIY community for matter devices

VojtechBartoska, 2023-06-23:
@TD-er We will add support for both in Arduino ESP32 Core, C6 & H2.

TD-er, 2023-06-23:
Great!
That means I could add support for both at the same time in my project.
Saves a lot of time.

P-R-O-C-H-Y, 2023-07-03:
## Table with Support overview

_We will keep this table up to date._

Date of last update: 3.7.2023

| **ESP32-H2 Support Overview**   |      |                |             |              |
|---------------------------------|:------------:|:--------------:|:-----------:|:------------:|
|          **Peripheral**         |   Assignee  |   **Testing**  | **Updates** | **Comments** |
|               ADC               | @P-R-O-C-H-Y |             |             | WIP on ESP-IDF side |
|          Ethernet  SPI          |              |                |             |  In progress  |
|               GPIO              | @P-R-O-C-H-Y |             |             |              |
|       I2C (Slave & Master)      |            |                |             |              |
|               I2S               |              |                |             |  In progress |
|             LEDC & PWM             |  @P-R-O-C-H-Y  | Fully OK |             |              |
|               RMT               |   @SuGlider  |            |             |              |
|               SPI               | @P-R-O-C-H-Y  |             |             |               |
|               TWAI              |              |                |             | run examples |
|           Temp. Sensor          | @P-R-O-C-H-Y  |           |             |              |
|               UART              |   @SuGlider  |              |             |              |
|         USB Serial/JTAG         |  @SuGlider |             |             |              |
|            SigmaDelta           | @P-R-O-C-H-Y |             |             |              |
| Timers (System/General-purpose) | @P-R-O-C-H-Y |                |             |              |
|           **General**           |              |                |             |              |
|              Wi-Fi              |               |              |             | not available |
|            Bluetooth            |  @SuGlider |            |             |              |
|           Bluetooth LE          |   @SuGlider  |             |             |              |
|             802.15.4  protocol  |              |                |             | support missing |
|              Thread             |              |                |             |  support missing |
|              Zigbee             |              |                |             |  support missing |
|              Power management |              |                |             |  under investigation |

229311JK, 2024-01-03:
I'd like to help w/this...

VojtechBartoska, 2024-01-03:
@229311JK with what exactly? Support is available in latest Development 3.0.0-apha 3 release, you can test and use already ESP32-H2, we are working on the stable release right now.

OgLocGreen, 2024-01-09:
When i was trying to use the BLE examples for the ESP32-H2 it uses the older [ESP32 BLE Arduino](https://www.arduino.cc/reference/en/libraries/esp32-ble-arduino/) and not the newer [Arduino BLE](https://www.arduino.cc/reference/en/libraries/arduinoble/).
Is this on purpose or is there a plan to switch to the Arduino BLE lib?

229311JK, 2024-01-26:
> @229311JK with what exactly? Support is available in latest Development 3.0.0-apha 3 release, you can test and use already ESP32-H2, we are working on the stable release right now.

working w/the ESP Thread/border Zigbee router V1.2...  Will be in touch...

jlabrada71, 2024-05-06:
Hi, I'm trying to use the ESP32-H2 in Arduino IDE,
When I install the board lib 2.0.16 (the latest stable) using the stable link
https://espressif.github.io/arduino-esp32/package_esp32_index.json

The ESP32-H2 doesn't get listed in the boards list. What could I be missing?
What would be a work around for this?

229311JK, 2024-05-09:
Been distracted w/the Esp32 programming board 4-the 32-C6 module.  It's nice;  Within the US, we have some excellent vendors to get hard to acquire Espresif HW.

xvolte, 2024-05-12:
Hello all
Is it available now?
Is there a way to test it?
I have a esp32-h2-mini-1 and would love to test it!

jlabrada71, 2024-05-12:
> Hello all Is it available now? Is there a way to test it? I have a esp32-h2-mini-1 and would love to test it!
Follow the install steps in this page and you will be able to tested. I received mine last week an it worked like a charm.

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

229311JK, 2024-05-13:
> Hello all Is it available now? Is there a way to test it? I have a esp32-h2-mini-1 and would love to test it!

Here you go:  https://github.com/espressif/arduino-esp32/releases   Now the hard part, scroll down & read.

229311JK, 2024-05-14:
> Hello all Is it available now? Is there a way to test it? I have a esp32-h2-mini-1 and would love to test it!

Did you get this squared away yet?  Enjoy your H2...

xvolte, 2024-05-14:
> > Hello all Is it available now? Is there a way to test it? I have a esp32-h2-mini-1 and would love to test it!
> 
> Did you get this squared away yet? Enjoy your H2...

I don t know how i did it, but i mixed two github projects… i wrote on the incorrect one…

this is the project :
https://github.com/tbnobody/OpenDTU

Sorry for inconveniance…

VojtechBartoska, 2024-05-28:
Hello all,

version [3.0.0-stable](https://github.com/espressif/arduino-esp32/releases/tag/3.0.0) was released yesterday and it includes support for ESP32-H2.

I am closing this ticket as solved and if you find any new issue related to ESP32-H2 chip, please open a new ticket.

Thanks for your contribution.

closed 2024-05-28 by VojtechBartoska