espressif/arduino-esp32 #6855: Arduino-esp32 SDK Wire library error.

Arya11111, 2022-06-10:
### Related area

I2C Wire library.

### Hardware specification

no, only soft error.

### Is your feature request related to a problem?

no

### Describe the solution you'd like

Please modify  function [uint8_t endTransmission(bool sendStop) ]'s accomplish.

When sendStop = false, I2c master can't send data to i2c slave, this is a wrong way.
In other SDK's Wire library, sendStop = false means I2C master will send data to slave, but not send stop bit.

### Describe alternatives you've considered

//#define I2C_BUFFER_LENGTH 128
When sending more than 128 bytes of data at a time, it will cause data loss.
/**
 *@param endflag 
 *@n    true   send data and send stop condition.
 *@n    false  send data and don't send stop condition.
*/
bool sendData(uint8_t addr, void* pData, uint16_t size, bool endflag){
    if(pData == NULL){
        return false;
    }
    uint8_t *pBuf = (uint8_t *)pData;
    uint16_t remain =  size ;
    uint8_t ret = 0;

    while(remain){
        size = (remain > I2C_BUFFER_LENGTH ? I2C_BUFFER_LENGTH : remain;
        remain -= size;
        Wire.beginTransmission(addr);
        Wire.write(pBuf, size);
        if(remain){
            ret = Wire.endTransmission(/*sendStop =*/false); //in this step, use Arduino-esp32 SDK ， I2C master can’t send data to //I2C slave.This phenomenon is not in line with the design of the Wire library
            pBuf += size;
        }else{
            ret = Wire.endTransmission(endflag);
        }
        if(ret != 0) return false;
        yield();
    }
    return true;
}

### Additional context

/**
 *@param sendStop
 *@n    true   send data and send stop condition.
 *@n    false  send data and don't send stop condition， but in Arduino-esp32 SDK ，sendStop=false，I2C master can't send data to slave,which will cause data lost.
 */
uint8_t endTransmission(bool sendStop);

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-06-10:
this is an ESP32 specific limitation. The I2C peripheral will not send anything out until a STOP is issued. Sending without STOP implies that you will read from that device right after (not write again), and after issuing that read, the peripheral will send first the write portion then the read portion and return the result to you.

marmou-fritzing, 2022-06-10:
I switched today from 1.6.0 to 2.0.3 core.
After I did requested change from "endTransmission()" to "endTransmission(bool)", where bool is not defaultly set, then I heave an issue xQueueGenericSend at endTransmission > __assert_fun > esp_system_abort > panic_abort.
(WROVER-B). Any idea ?

Arya11111, 2022-06-20:
I understand, but I think the processing here can be changed to avoid this bug. The user should use Wire.endTransmission(/*sendStop =*/false); instead of letting the sent data be lost, it should be handled as, when the user sends When Wire.endTransmission(/*sendStop =*/true);, the data will be sent 2 times at a time, which will avoid some bugs.
`Wire.beginTransmission(addr);
 Wire.write(0x00);
 Wire.endTransmission(false);
 Wire.write(0x01);
 Wire.endTransmission(true);
`
I hope that after running this code, I2C can send 0x00 0x01 data instead of losing Wire.endTransmission(false); the previous data, only send 0x01, I hope you can make some adjustments based on your limit to ensure its overall Data integrity, thank you.

closed 2025-01-16 by Parsaabasi