espressif/arduino-esp32 #6858: How to integrate llvm toolset?

ScSofts, 2022-06-12:
### Related area

Board Support

### Hardware specification

ESP32 DevKitC

### Is your feature request related to a problem?

&nbsp;&nbsp;&nbsp;&nbsp;I'm using [PlatformIO](https://platformio.org/) to develop  program on esp32 board, it uses GCC8, it seems that it doesn't support C++20 features like coroutines and compare.
&nbsp;&nbsp;&nbsp;&nbsp;Since there exists a  official project [LLVM-Xtensa](https://github.com/espressif/llvm-project) , why can't we use it to compile my project? 

    

### Describe the solution you'd like

&nbsp;&nbsp;&nbsp;&nbsp;I has tried to replace C:\Users\Root\.platformio\packages\toolchain-xtensa-esp32 with [xtensa-esp32-elf-llvm14_0_0-esp-14.0.0-20220415-win64](https://github.com/espressif/llvm-project/releases/download/esp-14.0.0-20220415/xtensa-esp32-elf-llvm14_0_0-esp-14.0.0-20220415-win64.zip) and edit  `C:\Users\Root\.platformio\packages\framework-arduinoespressif32\platform.txt`.
&nbsp;&nbsp;&nbsp;&nbsp;It seemd not to work.
My configuration:
```ini
96   # Compilers
97   compiler.c.cmd={compiler.prefix}clang
98   compiler.cpp.cmd={compiler.prefix}clang++
99   compiler.S.cmd={compiler.prefix}clang
100  compiler.c.elf.cmd={compiler.prefix}clang++
101  compiler.as.cmd={compiler.prefix}as
102  compiler.ar.cmd={compiler.prefix}ar
103  compiler.size.cmd={compiler.prefix}size
```

My build log:
```log
> Executing task: C:\Users\Root\.platformio\penv\Scripts\platformio.exe run --environment TWatch <

Processing TWatch (platform: espressif32; board: ttgo-t-watch; framework: arduino)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-watch.html
PLATFORM: Espressif 32 (4.4.0) > TTGO T-Watch
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.0 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TFT_eSPI @ 2.4.70
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SPI @ 2.0.0
|-- TTGO TWatch Library @ 1.4.2
|   |-- Wire @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- Ticker @ 2.0.0
|   |-- TFT_eSPI @ 2.4.70
|   |   |-- SPIFFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- LittleFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- SD @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|-- PNGdec @ 1.0.1
Building in release mode
Compiling .pio\build\TWatch\src\main.cpp.o
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
Compiling .pio\build\TWatch\lib67d\FS\FS.cpp.o
Compiling .pio\build\TWatch\lib67d\FS\vfs_api.cpp.o
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
Compiling .pio\build\TWatch\libc56\SPIFFS\SPIFFS.cpp.o
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
Compiling .pio\build\TWatch\liba36\LittleFS\LittleFS.cpp.o
Compiling .pio\build\TWatch\lib7e5\SPI\SPI.cpp.o
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
Compiling .pio\build\TWatch\lib458\TFT_eSPI\TFT_eSPI.cpp.o
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
Compiling .pio\build\TWatch\libe0d\Wire\Wire.cpp.o
Compiling .pio\build\TWatch\libf0a\Ticker\Ticker.cpp.o
clang: clangerror: unknown argument: '-fstrict-volatile-bitfields'
: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
Compiling .pio\build\TWatch\lib823\SD\SD.cpp.o
Compiling .pio\build\TWatch\lib823\SD\sd_diskio.cpp.o
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
Compiling .pio\build\TWatch\lib823\SD\sd_diskio_crc.c.o
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
Compiling .pio\build\TWatch\libfd5\TTGO TWatch Library\BBQ10Keyboard.cpp.o
Compiling .pio\build\TWatch\libfd5\TTGO TWatch Library\MPR121.cpp.o
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
clang: error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
Compiling .pio\build\TWatch\libfd5\TTGO TWatch Library\TTGO.cpp.o
*** [.pio\build\TWatch\src\main.cpp.o] Error 1
*** [.pio\build\TWatch\lib67d\FS\vfs_api.cpp.o] Error 1
*** [.pio\build\TWatch\lib67d\FS\FS.cpp.o] Error 1
Compiling .pio\build\TWatch\libfd5\TTGO TWatch Library\drive\axp\axp20x.cpp.o
*** [.pio\build\TWatch\libc56\SPIFFS\SPIFFS.cpp.o] Error 1
*** [.pio\build\TWatch\liba36\LittleFS\LittleFS.cpp.o] Error 1
*** [.pio\build\TWatch\lib7e5\SPI\SPI.cpp.o] Error 1
*** [.pio\build\TWatch\lib458\TFT_eSPI\TFT_eSPI.cpp.o] Error 1
*** [.pio\build\TWatch\libe0d\Wire\Wire.cpp.o] Error 1
*** [.pio\build\TWatch\libf0a\Ticker\Ticker.cpp.o] Error 1
*** [.pio\build\TWatch\lib823\SD\SD.cpp.o] Error 1
*** [.pio\build\TWatch\lib823\SD\sd_diskio_crc.c.o] Error 1
*** [.pio\build\TWatch\lib823\SD\sd_diskio.cpp.o] Error 1
*** [.pio\build\TWatch\libfd5\TTGO TWatch Library\BBQ10Keyboard.cpp.o] Error 1
*** [.pio\build\TWatch\libfd5\TTGO TWatch Library\MPR121.cpp.o] Error 1
clang: clang: error: unknown argument: '-fstrict-volatile-bitfields'
error: unknown argument: '-fstrict-volatile-bitfields'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: error: unknown argument: '-fno-tree-switch-conversion'
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-freorder-blocks' is not supported [-Wignored-optimization-argument]
*** [.pio\build\TWatch\libfd5\TTGO TWatch Library\drive\axp\axp20x.cpp.o] Error 1
*** [.pio\build\TWatch\libfd5\TTGO TWatch Library\TTGO.cpp.o] Error 1
============================================ [FAILED] Took 4.50 seconds ============================================
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-06-12:
Those cpreprocessor build flags, along with all the rest of the build command are also in platform.txt.  Did you try removing those to see if the build completes?
As to when/why, arduino will always be tied to the default build process from esp-idf.  If they switch over to use clang/llvm at some point, arduino-esp32 will as well, but that upstream source will have to drive the switch.

Jason2866, 2022-06-13:
Platformio does NOT use platform.txt.

ScSofts, 2022-06-16:
Thank you very much

closed 2023-03-24 by ScSofts