espressif/arduino-esp32 #6862: Unlimited restart of esp32c3 under version 2.0.3

WishToo, 2022-06-13:
### Board

ESP32C3 

### Device Description

AirM2M_CORE_ESP32C3

### Hardware Configuration

esp32-c3-devkitm-1

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

macOS 12.2.1

### Flash frequency

40MHz & 80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Infinite restart always occurs when testing any simple example

I have made the following settings, but the problem still exists：
USB CDC On Boot: Enabled
Flash Mode: DIO

### Sketch

```cpp
#include <Arduino.h>
void setup()
{
  Serial.begin(115200);
  Serial.println("test msg");
  // put your setup code here, to run once:
  pinMode(13, OUTPUT);
  pinMode(12, OUTPUT);
  digitalWrite(12, LOW);

}

void loop()
{
  // put your main code here, to run repeatedly:

  digitalWrite(13, digitalRead(13) == HIGH ? LOW : HIGH);
  digitalWrite(12, digitalRead(12) == HIGH ? LOW : HIGH);
  Serial.println("test msg");
  delay(1000);
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381782
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4bd624aff3257f126bd68f521ac53163cd7fe48cae8d3a77a847ef36aa76ad41
Expected: 2dd871cd993fccdf87b5ef3c78d949bd29ad8f4d97ecb130ff4297267fa0ddbe
Attempting to boot anyway...
entry 0x403ce000
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381782
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4bd624aff3257f126bd68f521ac53163cd7fe48cae8d3a77a847ef36aa76ad41
Expected: 2dd871cd993fccdf87b5ef3c78d949bd29ad8f4d97ecb130ff4297267fa0ddbe
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-06-13:
Check the [pinmap](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitc-02.html).  There is no gpio12 & 13 on a C3.

lbernstone, 2022-06-13:
gpio 12-17 are used for the spi flash.  If you want your device to work, I would recommend not developing on-board LEDs on gpio 12 & 13.

WishToo, 2022-06-13:
> > gpio 12-17 are used for the spi flash. If you want your device to work, I would recommend not developing on-board LEDs on gpio 12 & 13.
> 
> @lbernstone it use `DIO` mode, not `QIO`

I have used DIO mode

WishToo, 2022-06-13:
> DIO mode does not use gpio12 and gpio13, so it doesn't matter
> 
> it is a bug
> 
> if you flash esp32c3 with USB-CDC and use `2.0.3` version of `arduino-esp32`, it won't boot. but `2.0.2` is ok.
> 
> i tested it

Yes, Arduino 2.0.2 works normally, but the status value of the IO port cannot be read. It is always low, so I am confused now

lbernstone, 2022-06-14:
see #6572

chenxuuu, 2022-06-15:
> Still using GPIOs which are normaly used for connecting Flash is a bad design idea. You never know what changes future brings...

bad idea: arduino not work, others works well😀

a funny joke

lbernstone, 2022-06-15:
You need to set the pinmode to input if you want to read.

11wanqin, 2022-07-05:
> 

> > Still using GPIOs which are normaly used for connecting Flash is a bad design idea. You never know what changes future brings...
> 
> bad idea: arduino not work, others works well😀
> 
> a funny joke

[image]
[image]

I meet same problem when i use the luatos board;you can try to use platfomio , The Espressif 32 5.00 frameworks can
work well on this demo board, but you should to change the "esp32-c3-devkitm-1.json"'s "flash_mode" to "dio";everything work well after i do this.

me-no-dev, 2022-07-06:
problem is that C3 DIO mode does not work in 2.0.3, but it will work in 2.0.4 that will be released today. I guess PIO guys will also release their update these days and you will be OK. Till then `"flash_mode":"qio"`

VojtechBartoska, 2022-07-12:
@WishToo Can you please retest your code under v2.0.4? Thanks.

WishToo, 2022-07-15:
> @WishToo Can you please retest your code under v2.0.4? Thanks.

Thank you very much. I just tested it. Under version 2.0.4, this development board can be used normally

closed 2022-07-15 by VojtechBartoska

spilz87, 2022-08-23:
Hi

I'm just trying to used these pins too on a ESP32 c3 but I got the restart all the time :(

Which pins can be used ?
Which way ? Input / output ?

I tried DIO and QIO but doesn't solve :(

Thanks in advance for you help

spilz87, 2022-08-24:
ESP32 C3
On Arduino IDE
With 2.0.4 
With DIO

I'm only able to set GPIO12 and GOIO13 in OUTPUT mode. 

Setting other pins GPIO14-17 makes it restart.
Setting in INPUT mode makes it restart 

So no way to use other pins and only in output

Is it normal ???

Thanks for your help

beachmiles, 2022-09-29:
I too was getting infinite reboots, the only fix for me was enabling USB CDC On Boot

thilojaeggi, 2023-02-11:
Using 2.0.6 DIO works while QIO doesn't.

Jason2866, 2023-02-12:
@thilojaeggi was a bug in core 2.0.3 solved later. Verified working in latest core.