espressif/arduino-esp32 #6866: Add Support for ESP Insights in arduino core

osamahnini, 2022-06-14:
### Related area

https://github.com/espressif/esp-insights

### Hardware specification

All esp32 devices

### Is your feature request related to a problem?

no

### Describe the solution you'd like

Add ESP Insights modules to Arduino core. So, you can use this in Arduino development environments.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-06-27:
What is your opinion @shahpiyushv? Thanks for sharing your 2c :)

mrengineer7777, 2023-04-13:
@osamahnini I believe Insights was added a few months ago.

closed 2023-04-13 by mrengineer7777