espressif/arduino-esp32 #6872: When use SPI device (use TFT_SPI) The task is crash after setCpuFrequencyMhz()

LusterWong, 2022-06-15:
### Board

ESP32-WROVER-E

### Device Description

DevKitC V4

### Hardware Configuration

pin_sclk = 18; 
pin_mosi = 23; 
pin_miso = 19; 
pin_dc   = 21; 

### Version

v2.0.3

### IDE Name

Sloeber

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921000

### Description

In projects using LCD, down the frequency to 80Mhz when idle.
Initialize LCD after call setCpuFrequencyMhz(80) and print the current frequency, which freezes each time.

Run the output from serial "in...." will task crash.

### Sketch

```cpp
#include "Arduino.h"
#include <TFT_eSPI.h>

static TFT_eSPI tft;

void setup() 
{
        Serial.begin(115200);
        Serial.println("Test!");
        tft.init();
        tft.startWrite();
}

loop()
{
        Serial.printf("in....\n");
        delay(200);
   
	if(setCpuFrequencyMhz(80) == true)
	{
		delay(200);
		Serial.printf("f = %d\n", getCpuFrequencyMhz());
	}
        else
        {
                Serial.printf("fail\n");
        }
        delay(60 * 1000L);
}
```

### Debug Message

```plain
There is no output.
```

### Other Steps to Reproduce

I tried setCpuFrequencyMhz(80) on another task and found that only the current task was crash, the loop() running normally .

```
#include "Arduino.h"
#include <TFT_eSPI.h>

static TFT_eSPI tft;

void TestTask(void* Parameter)
{
	delay(1000 * 15);
	while(1)
	{
		Serial.printf("in....\n");
                delay(200);
   
	        if(setCpuFrequencyMhz(80) == true)               //<-------------------carsh-------------------
	        {
		       delay(200);
		       Serial.printf("f = %d\n", getCpuFrequencyMhz());
	        }
                else
                {
                        Serial.printf("fail\n");
                }
	}
}

void setup() 
{
        Serial.begin(115200);
        Serial.println("Test!");

        xTaskCreate(TestTask, "TestTask", 4096, NULL, 1, NULL);
        tft.init();
        tft.startWrite();
}

loop()
{
        Serial.printf("run\n");
        delay(5 * 1000L);
}

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-06-15:
can you post the logs from the crash? Changing CPU frequency has not yet been verified with Arduino 2.0.x and the HAL changes it includes. We might drop this feature (or allow it only in the beginning of setup)

LusterWong, 2022-06-15:
> can you post the logs from the crash? Changing CPU frequency has not yet been verified with Arduino 2.0.x and the HAL changes it includes. We might drop this feature (or allow it only in the beginning of setup)

Yes, in fact 2.0.2 also crashed.
I want to print logs, but if there's a crash the logs won't print.

LusterWong, 2022-06-15:
> so how do you know that it crashed? does it reboot?

Not reboot, and the loop() no longer run when call setCpuFrequencyMhz(80).
Another case you can see above the post "**Other Steps to Reproduce**". The **TestTask** call setCpuFrequencyMhz(80) this task will crashed but loop() running normally. Prove that setCpuFrequencyMhz is just stuck to own task.

VojtechBartoska, 2022-11-21:
Issue update:

We're postponing this issue into 3.0.0 release, it needs more investigation from our side.

mrengineer7777, 2023-04-13:
@LusterWong Please retest on current version 2.0.7.  Do you still see a crash?

VojtechBartoska, 2024-02-20:
@P-R-O-C-H-Y Can you please give this a try? Thanks

TD-er, 2024-02-20:
Hmm I think this is what also happened to me last week when I tried to change the clock freq. on M5Stack core2 with SPI Ethernet.
I can retry, but I think this is still an issue.

TD-er, 2024-02-20:
I just tried and could not reproduce it anymore.
So I guess this is fixed (or at least when using SPI Ethernet) 
I used these latest commits from ESP-IDF and Arduino-esp32, cherry-picked by @Jason2866 
https://github.com/Jason2866/esp32-arduino-lib-builder/releases/tag/2036

P-R-O-C-H-Y, 2024-02-21:
Hello @LusterWong, I have retested you issue and I there is actually no crash in the task.
What is happening is, that when you call `tft.startwrite()` which is basically `SPI.beginTransaction()` it will take the semaphore, which means that another tasks have to wait until the semaphore is released. It is released when `tft.startwrite()` or `SPI.endTransaction()` is called. After that you can successfully set the CPU frequency.

This ways its done to keep correct SPI Frequency for whole transaction and also to protect the SPI when multiple tasks are used.

```
#define SPI_PARAM_LOCK()    do {} while (xSemaphoreTake(paramLock, portMAX_DELAY) != pdPASS)
#define SPI_PARAM_UNLOCK()  xSemaphoreGive(paramLock)
```

I am closing this as no action needed there.

closed 2024-02-21 by P-R-O-C-H-Y