espressif/arduino-esp32 #6879: ESP32S3 can not interface SD slot in SDIO modes

idea--list, 2022-06-18:
### Board

Adafruit ESP32S3 Feather - no PSRAM (ID: 5323)

### Device Description

Adafruit ESP32S3 Feather - no PSRAM (ID: 5323)

### Hardware Configuration

uSD card slot wired up like that:
SD Card | ESP32
 CLK      36
 CMD      35
 D0       37
 D1       14
 D2       5
 D3       13
 VSS      GND
 VDD      3.3V

### Version

v2.0.3

### IDE Name

Arduino

### Operating System

Windows 10 x64

### Flash frequency

40, 80

### PSRAM enabled

no

### Upload speed

115200

### Description

ESP32S3 data sheet states, that the S3 can provide SDIO functionality on any pins.
Using ESP-IDF i can confirm this, however using the arduino core for this chip i can not interface any SD cards using SDIO neither in 1 nor in 4 bit modes. Only SPI connection works with ESP32S3 on the arduino core.

Lines 103-108 inside SD_MMC.cpp hard code pin numbers (as ESP32 can provide SDIO only on those pins), however if i override these pin numbers manually to match pin numbers how i wired things up... well it still does not work. Not even after pulling those pins high inside setup().

Interestingly arduino core V2.0.3 is based on ESP-IDF V4.4.1 but it seems there is still some breaking difference in this regard. 

### Sketch

```cpp
#include "FS.h"
#include "SD_MMC.h"

#define SDSLOT_HSPI_SCLK 36
#define SDSLOT_HSPI_MOSI__CMD 35
#define SDSLOT_HSPI_MISO__DAT0 37
#define SDSLOT_DAT1 14
#define SDSLOT_DAT2 5
#define SDSLOT_HSPI_CS_DAT3 13

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
        Serial.println("Dir created");
    } else {
        Serial.println("mkdir failed");
    }
}

void removeDir(fs::FS &fs, const char * path){
    Serial.printf("Removing Dir: %s\n", path);
    if(fs.rmdir(path)){
        Serial.println("Dir removed");
    } else {
        Serial.println("rmdir failed");
    }
}

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\n", path);

    File file = fs.open(path);
    if(!file){
        Serial.println("Failed to open file for reading");
        return;
    }

    Serial.print("Read from file: ");
    while(file.available()){
        Serial.write(file.read());
    }
}

void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("File written");
    } else {
        Serial.println("Write failed");
    }
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
}

void renameFile(fs::FS &fs, const char * path1, const char * path2){
    Serial.printf("Renaming file %s to %s\n", path1, path2);
    if (fs.rename(path1, path2)) {
        Serial.println("File renamed");
    } else {
        Serial.println("Rename failed");
    }
}

void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\n", path);
    if(fs.remove(path)){
        Serial.println("File deleted");
    } else {
        Serial.println("Delete failed");
    }
}

void testFileIO(fs::FS &fs, const char * path){
    File file = fs.open(path);
    static uint8_t buf[512];
    size_t len = 0;
    uint32_t start = millis();
    uint32_t end = start;
    if(file){
        len = file.size();
        size_t flen = len;
        start = millis();
        while(len){
            size_t toRead = len;
            if(toRead > 512){
                toRead = 512;
            }
            file.read(buf, toRead);
            len -= toRead;
        }
        end = millis() - start;
        Serial.printf("%u bytes read for %u ms\n", flen, end);
        file.close();
    } else {
        Serial.println("Failed to open file for reading");
    }

    file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }

    size_t i;
    start = millis();
    for(i=0; i<2048; i++){
        file.write(buf, 512);
    }
    end = millis() - start;
    Serial.printf("%u bytes written for %u ms\n", 2048 * 512, end);
    file.close();
}

void setup(){
    pinMode(SDSLOT_HSPI_SCLK, INPUT_PULLUP);
    pinMode(SDSLOT_HSPI_MOSI__CMD, INPUT_PULLUP);
    pinMode(SDSLOT_HSPI_MISO__DAT0, INPUT_PULLUP);
    pinMode(SDSLOT_DAT1, INPUT_PULLUP);
    pinMode(SDSLOT_DAT2, INPUT_PULLUP);
    pinMode(SDSLOT_HSPI_CS_DAT3, INPUT_PULLUP);

    Serial.begin(115200);
    if(!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD_MMC.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD_MMC card attached");
        return;
    }

    Serial.print("SD_MMC Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD_MMC.cardSize() / (1024 * 1024);
    Serial.printf("SD_MMC Card Size: %lluMB\n", cardSize);

    listDir(SD_MMC, "/", 0);
    createDir(SD_MMC, "/mydir");
    listDir(SD_MMC, "/", 0);
    removeDir(SD_MMC, "/mydir");
    listDir(SD_MMC, "/", 2);
    writeFile(SD_MMC, "/hello.txt", "Hello ");
    appendFile(SD_MMC, "/hello.txt", "World!\n");
    readFile(SD_MMC, "/hello.txt");
    deleteFile(SD_MMC, "/foo.txt");
    renameFile(SD_MMC, "/hello.txt", "/foo.txt");
    readFile(SD_MMC, "/foo.txt");
    testFileIO(SD_MMC, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD_MMC.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD_MMC.usedBytes() / (1024 * 1024));
}

void loop(){

}
```

### Debug Message

```plain
Card Mount Failed

Nothing else is displayed neither on the serial port nor in the console despite verbose debug level.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-06-18:
Remove the pinMode calls, and use `SDMMC.setPins(int clk, int cmd, int d0, int d1, int d2, int d3);`

idea--list, 2022-06-18:
@lbernstone 
Thank you very much for the suggestion!
I began using ESP chips just recently and for sure there are still a bunch of things to learn about the SW frameworks.
But hey, the idea works and it made my day! 😄

closed 2022-06-18 by idea--list

lbernstone, 2022-06-18:
S3 is new, so google does not know much about it.  This was added specifically to support the new chip.