espressif/arduino-esp32 #6882: BluetoothSerial fails with Platform.io and ESP32-S3

fabiencomte, 2022-06-19:
### Board

ESP32-S3-DevKitC-1-N32R8V

### Device Description

ESP32-S3-DevKitC-1-N32R8V
please see
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

### Hardware Configuration

USB to PC

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windos 11

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

BluetoothSerial link fails

c:/users/comte/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32app/libe96/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL8_stop_btv+0x18): undefined reference to `esp_spp_disconnect'
c:/users/comte/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32app/libe96/libBluetoothSerial.a(BluetoothSerial.cpp.o):(.literal._ZL8_stop_btv+0x1c): undefined reference to `esp_spp_deinit'
c:/users/comte/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio/build/esp32app/libe96/libBluetoothSerial.a(BluetoothSerial.cpp.o): in function `_stop_bt()':
C:/Users/comte/.platformio/packages/framework-arduinoespressif32/libraries/BluetoothSerial/src/BluetoothSerial.cpp:716: undefined reference to `esp_spp_disconnect'
c:/users/comte/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:/Users/comte/.platformio/packages/framework-arduinoespressif32/libraries/BluetoothSerial/src/BluetoothSerial.cpp:717: undefined reference to `esp_spp_deinit'

### Sketch

```cpp
#include "Arduino.h"
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;
const char *pin = "1234";

void setup() {
 
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  Serial.print("Hello world\n");

 SerialBT.setPin(pin);
  SerialBT.begin("ESP32test", true); 
}

void loop() {
  String inputFromOtherSide;
  if (SerialBT.available()) {
    inputFromOtherSide = SerialBT.readString();
    SerialBT.println("You had entered: ");
    SerialBT.println(inputFromOtherSide);
  }
}
```

### Debug Message

```plain
No debug, it's on build time
```

### Other Steps to Reproduce

Try to build this sketch

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-06-19:
Bluetooth SPP (aka classic) is only available on esp32.

closed 2022-06-20 by SuGlider

gnpaone, 2022-07-20:
But I'm using Arduino IDE and it's ESP32-S3 other Bluetooth codes working fine except Bluetoothserial.h, ESP32-S3 is ESP32 only right?

Franck78, 2022-12-27:
Hello, @SuGlider 
can you please complete the answer in just some words : what kind of thing/include/protocol can send a simple 'text sms' to this S3-BLE 5.0 ???

SuGlider, 2022-12-27:
SMS is service that is part of SS7 network signaling used in a GSM mobile network.

There is no way to receive SMS directly to any MCU, unless than using a GSM Module that will comunícate with the ESP32xxx using UART or SPI and some AT commands.

An example os such GSM Module is the SIM800L.

Another option is using some Cloud based SMS gateway that uses TCP/IP. Some smartphone may work as "GSM Module" and provide a service through an APP using maybe Bluetooth services for the communication with the MCU.

Example of a HTTP SMS based Gateway Service is the http://www.twilio.com

In that case, the ESP32xxx can use WiFi to connect to the Internet and access Twilio TCP/IP based services from the Cloud.

I hope this information helps you.

Franck78, 2022-12-27:
oups , should have not used 'SMS'. I was literally thinking 'send small message' , from a bluetooth device  to an ESP32S3.
Not the GSM thing.
I hope my response is somewhere in Examples/ESP32 BLE Arduino/*.ino
Just want cheap aliexpress HC-06 to send a message to ESP32S3  ;)

But sure enough , someone will make a device needing a real GSM connectivity (alarm or thing like that).

SuGlider, 2022-12-27:
> I was literally thinking 'send small message' , from a bluetooth device to an ESP32S3.

The HC06 is SPP only, using Bluetooth Classic (v2.0) and not BLE. 
Therefore, it can't talk to the ESP32-S3 or ESP32-C3.
You shall use another BLE module, like the HM-10, or another ESP32-C3/ESP32-S3/ESP32 for the BLE communication.

BLE uses Services/Characteristics to exchange data.

The ESP32 and ESP32-C3/S3 can be programmed as BLE Server/Client to send/receive data using these examples:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_server/BLE_server.ino
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_client/BLE_client.ino

SuGlider, 2022-12-27:
For more information about how to make an ESP32-S3 connect to a HM-10 BLE module, please read this:
https://esp32-server.de/hm-10-esp32/

Franck78, 2022-12-28:
Good news SuGlider
I can receive what enters the serial port of the HC-06 !?!!

I updated the code [BLE_client/BLE_client.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_client/BLE_client.ino) with corrects UUID for serviceUUID & charUUID

Bad news is now I have to found what are thoses UUID, specifics to one HC06, specifics for all HC06 specifics for all HCxx ;)
I was expecting to provide something more NAME & PINCODE
Apparently anyone on earth can connect to my little HC06.

Thank you

```
BLE Advertised Device found: Name: FrogBox, Address: 98:da:60:04:b2:3c, manufacturer data: 000098da6004b23c, serviceUUID: 0000ffe0-0000-1000-8000-00805f9b34fb, rssi: -73
Forming a connection to 98:da:60:04:b2:3c
 - Created client
 - Connected to server
 - Found our service
 - Found our characteristic
We are now connected to the BLE Server.
Setting new characteristic value to "Time since boot: 3"
....
Setting new characteristic value to "Time since boot: 9"
Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 34
data: Sélection validée
FROG  21  !
⸮Y@0	         <=SOME EXTRA GARBAGE to understand
Setting new characteristic value to "Time since boot: 10"
....
Notify callback for characteristic 0000ffe1-0000-1000-8000-00805f9b34fb of data length 34
data: Sélection validée
FROG  22  !
⸮Y@0	
Setting new characteristic value to "Time since boot: 14"

```

SuGlider, 2022-12-29:
Bluetooth v2.0 and BLE have different stack implementation. 
They may seem to have parts of the stack in common, but they work diferently. 
It is not possible to intercommunicate both correctly.

Franck78, 2022-12-29:
then our chineese friend have rebadged some bt4.0 chip. It make sense because their are not stupid enough to keep selling a devive with obsolete capabilities.
Here is a photo of some 'data' advertised by the 'HC-06'
[image]

SuGlider, 2022-12-29:
Does it also works as SPP BT 2.0?
Or just BLE?

SuGlider, 2022-12-29:
Please post a photo of this HC module and it's chip. I'll try to find it in the market for testing as well.

Franck78, 2022-12-29:
It does SPP that certain because that was my goal to have my device send some text. Now was it on BT2.0, 4.0, or 5, honestly, I didn't even think to search for that, as long as the link was established. 

The other side is a Qualcomm Atheros ar9565
Under linux, I don't  know how to find bt version used (ubuntu bluetoothctl).

[image]
[image]
CSR BC417143BGQ
K9296LY
bought here:
https://fr.aliexpress.com/item/1005002168517770.html