espressif/arduino-esp32 #6892: wifi nvs cfg alloc out of memory

ln-12, 2022-06-21:
### Board

esp32dev (Chip is ESP32-D0WDQ6 (revision 1))

### Device Description

I am using this board: https://joy-it.net/de/products/SBC-NodeMCU-ESP32

### Hardware Configuration

The board is only connected to one of these: https://smile.amazon.de/dp/B06XHJTGGC

### Version

v2.0.3

### IDE Name

PlatformIO

### Operating System

macOS 12.4

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I am using my board on a mobile installation which means that I am sometimes out of range of my wifi AP. So I am periodically scanning the wifi networks around and try to connect afterwards. After some hours, I get this error when I start the scan:

```
Scanning WiFi networks...
E (163569179) wifi:wifi nvs cfg alloc out of memory
E (163569180) wifi:init nvs: failed, ret=101
[163563115][E][WiFiGeneric.cpp:601] wifiLowLevelInit(): esp_wifi_init 257
Scan done
```

My code is quite complex, so I just attach the relevant scanning code below. I am using the FreeRTOS framework.

### Sketch

```cpp
uint8_t waitForConnectResult2(unsigned long timeoutLength)
{
    //1 and 3 have STA enabled
    if((WiFiGenericClass::getMode() & WIFI_MODE_STA) == 0) {
        return WL_DISCONNECTED;
    }
    unsigned long start = millis();
    while((!WiFi.status() || WiFi.status() >= WL_DISCONNECTED) && (millis() - start) < timeoutLength) {
        vTaskDelay(250);
    }

    return WiFi.status();
}

void TaskCommunication(void *pvParameters) {   
    for(;;) {
        if (WiFi.status() != WL_CONNECTED) {
            // disconnect and shutdown wifi to start fresh
            WiFi.disconnect(true, true);

            vTaskDelay(100);

            // initialize wifi (but don't connect yet)
            WiFi.begin(config::ssid, config::password, 0, nullptr, false);

            getStringFor(WiFi.status(), statusBuffer);
            globals::sdl.log("Current Wifi status: " + String(statusBuffer), INFO, true);

            Serial.println("Scanning WiFi networks...");

            //int16_t n = WiFi.scanNetworks(false, false, false, 300U, 0, config::ssid);
            int16_t n = WiFi.scanNetworks();

            Serial.println("Scan done");

            if (n == 0) {
                // no networks are found (probably because the bike is driving), so we
                // check later again
                globals::sdl.log("No networks found", INFO, true);
                vTaskDelay(5000);
                continue;
            } else if (n < 0) {
                // some error occurred, so we try again later
                globals::sdl.log("Error while scanning: " + String(n), INFO, true);
                vTaskDelay(5000);
                continue;
            } else {
                // we found at least 1 network
                Serial.print(n);
                Serial.println(" networks found");

                String networks = "";

                for (int i = 0; i < n; ++i) {
                    if(i < 5) {
                        networks += "(" + WiFi.SSID(i) + ", " + WiFi.RSSI(i) + ") ";
                    }

                    Serial.print(i + 1);
                    Serial.print(":\tSSID ");
                    Serial.print(WiFi.SSID(i));

                    String padding = "";
                    int length = WiFi.SSID(i).length();
                    for(int y = 0; y < 32 - length; y++) {
                        padding += " ";
                    }

                    Serial.print(padding);
                    Serial.print("\t\tRSSI ");
                    Serial.print(WiFi.RSSI(i));
                    Serial.print("\t\tChannel ");
                    Serial.print(WiFi.channel(i));
                    Serial.print("\t\tEncryption ");
                    Serial.println(WiFi.encryptionType(i));

                    vTaskDelay(10);
                }

                globals::sdl.log("Networks: " + networks, INFO, true);
            }

            // init wifi
            globals::sdl.log("Connecting to Wifi: " + String(config::ssid), INFO, true);

            // no wifi connection, try to connect
            WiFi.begin(config::ssid, config::password);

            globals::sdl.log("Starting connection timeout...", INFO, true);

            // Wait 10s to connect to wifi
            waitForConnectResult2(10000);

            if (WiFi.status() != WL_CONNECTED) {
                globals::sdl.log("No Wifi connection, abort.", INFO, true);

                WiFi.disconnect(true, true);

                // wait 5s before we try again to save some energy
                vTaskDelay(5000);

                continue;
            }

            globals::sdl.log("Connected to Wifi: " + String(config::ssid), INFO, true);
            globals::sdl.log("IP: " + WiFi.localIP().toString(), INFO, true);

            // sync current time
            configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
            Serial.print("Current time: ");
            printLocalTime();

            // check for firmware updates
            // if (isFirmwareUpdateAvailable())
            // {
            //     performFirmwareUpdate();
            // }

            // setup websocket callbacks
            ws::websocketClient.onMessage(ws::onMessageCallback);
            ws::websocketClient.onEvent(ws::onEventsCallback);

            // TODO only use wss with server certificate pinning
            #ifdef SECURE_WS
                        websocketClient.setCACert(rootCACertificateCityBox);
            #endif
        } else {
            // perform actual connection
            vTaskDelay(5000);
        }
    }
}
```

### Debug Message

```plain
E (163569179) wifi:wifi nvs cfg alloc out of memory
E (163569180) wifi:init nvs: failed, ret=101
[163563115][E][WiFiGeneric.cpp:601] wifiLowLevelInit(): esp_wifi_init 257
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-06-21:
FreeRTOS framework?  This is Arduino Espressif framework. You cant choose a framework.

ln-12, 2022-06-21:
Sorry for the confusion, of course I use the arduino framework:
```
platform = espressif32@4.4.0
board = esp32dev
framework = arduino
```

I just wanted to make clear that I also make use of [FreeRTOS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html) from `#include "freertos/FreeRTOS.h"` although I don't think that it is relevant for my issue.

ln-12, 2022-06-21:
@lbernstone I added your code and the free heap memory is indeed getting less. I'll fix the leak and update this issue after it runs without errors for some time. Thank you very much!

ln-12, 2022-07-04:
@lbernstone Thanks again for your suggestion, but unfortunately the error still remains. I had my test setup running (without the `scanNetworks()` call) over the weekend and got the following result:
```
141667.57 [INFO] ESP free heap: 90624 (35.20%), minimum esp free heap size: 64652 (25.11%)
E (141684079) wifi:wifi nvs cfg alloc out of memory
E (141684079) wifi:init nvs: failed, ret=101
[141667994][E][WiFiGeneric.cpp:601] wifiLowLevelInit(): esp_wifi_init 257
[141667996][E][WiFiSTA.cpp:227] begin(): STA enable failed!
141668.00 [INFO] Current Wifi status: UNKNOWN
141668.01 [INFO] Connecting to Wifi: TestWiFi
E (141684101) wifi:wifi nvs cfg alloc out of memory
E (141684101) wifi:init nvs: failed, ret=101
[141668019][E][WiFiGeneric.cpp:601] wifiLowLevelInit(): esp_wifi_init 257
[141668025][E][WiFiSTA.cpp:227] begin(): STA enable failed!
141668.03 [INFO] Starting connection timeout...
141668.04 [INFO] No Wifi connection, abort.
```

I noticed that `waitForConnectResult2` returns almost immetiately (see the last two lines). The error `UNKNOWN` indicates that the returned code is not in 0-6.

bmedici, 2022-11-24:
Hi, any news about this ? Same kind of problem here:
```
    1.814 | kB  101 /  342| | display.cpp          | +ms       7 | network: wifi
E (3847) wifi:Expected to init 16 rx buffer, actual is 7
[E][WiFiGeneric.cpp:137] wifiLowLevelInit(): esp_wifi_init 257
E (3854) wifi:wifi nvs cfg alloc out of memory
[E][WiFiGeneric.cpp:137] wifiLowLevelInit(): esp_wifi_init 257
```

closed 2023-04-13 by mrengineer7777