espressif/arduino-esp32 #6897: LittleFS -> lfs_file_Close crash

zekageri, 2022-06-22:
### Board

esp-wrover-kit

### Device Description

Custom board with esp32-wrover-E ( 16mb flash 8mb psram )

### Hardware Configuration

ETH_Lan8720 -> Ethernet
DS3231 -> RTC
ST7735 -> Display

Heavy use of Serial2 for Modbus.

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have an admin page where it automatically requests the top level LittleFs file informations. When this happens, the ESP sometimes crash, sometimes not.

### Sketch

```cpp
#include <Arduino.h>
#include "FS.h"
#include <LittleFS.h>
#include <ArduinoJson.h>

int fSystem::getFolderSize(const char* dirPath){
    int totalSize = 0;
    File dir = LittleFS.open(dirPath,FILE_READ);
    if( dir ){
        File entry = dir.openNextFile();
        while(entry){
            if( entry.isDirectory() ){
                vTaskDelay(2);
                totalSize += getFolderSize( entry.path() );
            }else{
                totalSize += entry.size();
            }
            entry = dir.openNextFile();
        }
        entry.close();
        dir.close();
    }
    return totalSize;
}

/*
* List the files in a directory on one level
*/
String fSystem::listDir(const char* dirPath){
    if (!isMounted) {return "";}
    fsLock();
    String JsonResult = "";
    //DynamicJsonDocument doc(5500);
    SpiRamJsonDocument doc(DIR_STRUCT_SIZE);
    doc["type"] = "directoryData";
    JsonArray content = doc.createNestedArray("content");
    File dir = LittleFS.open(dirPath,FILE_READ);
    if( dir ){
        File entry = dir.openNextFile();
        while(entry){
            JsonObject newEntry = content.createNestedObject();
            String entryName    = entry.path();

            time_t epochSec     = entry.getLastWrite();
            newEntry["lastModified"] = epochSec;
            newEntry["isDirectory"]  = entry.isDirectory();
            newEntry["name"]         = entryName;

            if( entry.isDirectory() ){
                newEntry["size"] = getFolderSize( entry.path() ); //0;//
            }else{
                newEntry["size"] = entry.size();
            }
            vTaskDelay(5); // --> this does not help unfortunatelly.
            entry = dir.openNextFile();
        }
        entry.close();
        dir.close();
        serializeJson(doc, JsonResult);
    }
    fsUnlock();
    return JsonResult;
}

void setup(){
    Serial.begin(115200);
    listDir("/");
}

void loop(){
    vTaskDelete(NULL);
}
```

### Debug Message

```plain
assert failed: lfs_file_close lfs.c:5127 (lfs_mlist_isopen(lfs->mlist, (struct lfs_mlist*)file))

Backtrace:0x40083e1d:0x3ffb24f00x4008edc5:0x3ffb2510 0x40094521:0x3ffb2530 0x401489b7:0x3ffb2660 
0x401447e7:0x3ffb2680 0x401198eb:0x3ffb26a0 0x4000bd4b:0x3ffb26c0 0x40001781:0x3ffb26e0 0x401944dd:0x3ffb2700 0x40194513:0x3ffb2720 0x4019452a:0x3ffb2740 0x400f3edd:0x3ffb2760 0x401abc95:0x3ffb2780 0x400f4f72:0x3ffb27a0 0x400f3261:0x3ffb27c0 0x400d2b8e:0x3ffb27e0 0x400d28b2:0x3ffb2800 0x4010aca5:0x3ffb2820
  #0  0x40083e1d:0x3ffb24f00 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402

Decoding stack results
0x40083e1d: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008edc5: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 128
0x40094521: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 85
0x401489b7: lfs_file_close at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp_littlefs/src/littlefs/lfs.c line 5127
```

### Other Steps to Reproduce

The produced JSON string will be send by Async webserver's websocket.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2022-06-22:
Hi @zekageri, I am not able to reproduce your issue.
While it's related to closing file, can you try moving `entry.close();` before `entry = dir.openNextFile();`.

P-R-O-C-H-Y, 2022-06-22:
No problem :) 
For what I searched, there is not a limitation of max opened files.
Maybe there is, but it is not any defined value.

zekageri, 2022-06-22:
Looks like this solved my problem. 
Thank you for your help.

closed 2022-06-22 by zekageri