espressif/arduino-esp32 #6901: RAM CONFUSION AND QUESTION

zekageri, 2022-06-23:
### Board

esp-wrover-kit

### Device Description

Custom board with a lot of peripherials

### Hardware Configuration

Custom hardware with a lot of peripherials

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm using a lot of things in my code.
I have a lot of classes, char arrays, tasks and i use a lot of ArduinoJson documents.
Using peripherials like ETH_LAN_8720, ST_7735 Display and DS3231 and a lot of hardware things.
And ofc since this is a chip supposed to focus on IOT, i use a lot of HTTP(S) requests and (SSL)Websocket connections.
I have to save RAM in every corner of my code.

_But my confusion starting from here._

**My ESP has 4MB external ram.**

Currently it is using 210kb RAM which was allocated from the **INTERNAL** memory and using 618KB from external ram.
I attached a picture for reference which is from the esp's admin page where i can live monitor the resources.

[image]

I'm in a situation where i have to maintain an SSL websocket connection with a NodeJs server trought socketIO.
Meanwhile the esp has to ask the same server for file/firmware upgrades which uses HTTPS, and an other to download if there are new versions.

The problem with that is the TLS implementation allows me only **ONE** SSL connection at a time.
So what i have done to get rid of this problem?

I drop the websocket connection for the duration of the HTTPS requets, and reinitializing it after that, with newly pulled CA from the filesystem. 
**This causes some serious memory fragmentation because everything uses my INTERNAL memory.**

So as you can see, while i have no internal memory left, the chip sits there and looking at my **3.4 MEGABYTE** external memory while crashing on an SSL handshake because the internal memory so fragmented it can not allocate enough space for a new connection.

Why isn't it throws an error or allocates from the external ram for the SSL instead of crashing with an `mbedtls_ssl_read_record` error you ask?
### I dont know, and it seems to me that nobody has any idea why.

So you say i just have to allocate enough RAM for **every unknown sized** packet that comes on a socket?
If you can tell me a safe size for the packets to allocate and you can promise that my software will never crash because there is not enough space allocated for the message, i will gladly do that. But as long as i can't do such a thing, i have to rely on dynamic allocation, which ofc will fragment my heap no matter what i do.

So there are a couple of questions in my mind.

-  Why can't it allocate memory from my external ram if there is not enough space in internal?
Is there a flag where i can tell the RTOS or whatever allocates the memory to always use my external ram?
- Is there any garbage collection feature that i don't know of? They say it would slow the chip down. I would take that risk if it 
would ensure that my heap will not be fragmented.
- Can i lower the barrier which handles the ram allocation process?

### Sketch

```cpp
// Just put a lot of SSL requests into.
// Here is my current problem:

boolean fSystem::downloadFileFromServer(const char* path, const char* url){
    if (!isMounted) {return false;}
    boolean success = false;
    createPathIfNotExists(path);
    hsh_ServerComSys.disconnect();
    HTTPClient http;
    http.setReuse(false);
    http.begin(url);
    int httpCode = http.GET();
    if (httpCode > 0) {
        #if FS_DEBUG_MODE
            Serial.println("[FS] Downloading file from server...");
        #endif
        fsLock();
        File file = LittleFS.open(path, "w");
        if (file) {
            file.print(http.getString());
            file.close();
            success = true;
        }
        fsUnlock();
    }
    http.end();
    hsh_ServerComSys.loopUnlock();
    return success;
}
```

### Debug Message

```plain
Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060830  A0      : 0x8014b50a  A1  
    : 0x3ffb22e0
A2      : 0x00000278  A3      : 0x00000d65  A4      : 0x3fff8fe8  A5  
    : 0x3ffb2330
A6      : 0x00000d65  A7      : 0x00000060  A8      : 0x8014a96c  A9  
    : 0x00000aed
A10     : 0x00000000  A11     : 0x00000aed  A12     : 0x00000278  A13 
    : 0x3ffb2338
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000d  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4008b0a4  LEND    : 0x4008b0ba  LCOUNT  : 0xffffffff  

Backtrace:0xfffffffd:0x3ffb22e00x4014b507:0x3ffb2300 0x4014d83a:0x3ffb2370 0x4014ef3e:0x3ffb23b0 0x4014cfa1:0x3ffb2450 0x4014cfdb:0x3ffb2470 0x401950f9:0x3ffb2490 0x401947f2:0x3ffb2720 0x4019490e:0x3ffb2760 0x401af08b:0x3ffb2780 0x400f513e:0x3ffb27a0 0x400f3371:0x3ffb27c0 0x400d2bcb:0x3ffb27e0 0x400d28c2:0x3ffb2800 0x4010adb5:0x3ffb2820 
  #0  0xfffffffd:0x3ffb22e00 in ?? ??:0

PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x4014b507: mbedtls_ssl_read_record at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_msg.c line 4650
0x4014d83a: mbedtls_ssl_parse_certificate at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c line 2874
0x4014ef3e: mbedtls_ssl_handshake_client_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c line 4455
0x4014cfa1: mbedtls_ssl_handshake_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c line 5836
0x4014cfdb: mbedtls_ssl_handshake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c line 5873
0x401950f9: start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp line 269
0x401947f2: WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*, char const*) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp line 132
0x4019490e: WiFiClientSecure::connect(char const*, unsigned short) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp line 117
0x401af08b: WiFiClientSecure::connect(char const*, unsigned short, int) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp line 122
0x400f513e: WebSocketsClient::loop() at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h line 328
0x400f3371: SocketIOclient::loop() at lib/arduinoWebSockets/src/SocketIOclient.cpp line 184
0x400d2bcb: serverCommSys::loop() at src/utilities/HsHServerComm.cpp line 114
0x400d28c2: loop() at src/main.cpp line 85
0x4010adb5: loopTask(void*) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 50
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Ouss4, 2022-06-26:
> Why can't it allocate memory from my external ram if there is not enough space in internal?
Is there a flag where i can tell the RTOS or whatever allocates the memory to always use my external ram?

Looks like the `sdkconfigs` all have `CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC` enabled.  That should be replaced by `CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC` to be able to allocate from the external memory.

rykovv, 2022-06-27:
Have you considered to reorganize your code creating a network or, specifically, https connections manager? Possibly, you can keep better control of the RAM making efficient resource reuse. You can do something like FIFO (connection requests) + single point of network connection with reusable memory.

zekageri, 2022-06-28:
Thank you for the answers.

As iam responsible for freeing my resources iam not in controll of the internal allocations.

I was told earlier that the new version of arduino idf will prefer the external ram allocations with new and malloc, but this is not the case. It does not matter if my heap is really low or if my flag to force the ram allocation from ext ram is set, my internal ram is suffering because iam using a lot of dynamic structures, classes and HTTPClients.

Yes i considered to use some kind of a https helper class but it would not change the fact that my heap is gone. 

If i throw a wrapper class to the httpclient it would still use my internal ram. Doesnt it?

For example. My CA must be pulled from the file system right before a request. The reason for this is the esp pulls new CA from the server if the used one is about to expire and it can not live in the program space as const. And it is the same with a lot of other things. I have a bunch of user defined classes which can all be manipulated and changed by a user at runtime. It is necessary to me for these things to go into the PS-Ram.

Right now, i dont see how a http wrapper/helper class could help me. :/
I want my 3.5mb ram thats sitting there

zekageri, 2022-06-28:
Looks like the `sdkconfigs` all have `CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC` enabled.  That should be replaced by `CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC` to be able to allocate from the external memory.

Why cant a user controll these flags?

rykovv, 2022-06-29:
> If i throw a wrapper class to the httpclient it would still use my internal ram. Doesnt it?

Most likely. Although, my idea was not about creating an HTTP wrapper class because it would only worsen the fragmentation issue. But rather to have an RTOS task, if you use one, where you can dynamically/statically allocate memory for one connection (client, cert, etc.), thought for one connection at a time, and reuse this memory, having queued requests. That would reduce the heap fragmentation issue. But it will not solve the unreachable external RAM. 

Anyway, it was just a suggestion.

zekageri, 2022-06-29:
Thank you for the suggestion.
There is one http client at a time without a new task too. I cant even create multiple clients because i can only open one TLS connection at a time. I even drop an ssl socketio connection for the duration of a https request for this same reason.

VojtechBartoska, 2022-07-14:
@zekageri Can this be considered as answered and close? Thanks

closed 2022-07-14 by zekageri