espressif/arduino-esp32 #6906: PSRAM severely affects wakeup time from deepsleep

Algonsi, 2022-06-24:
### Board

ESP32-PICO-DevKitM-2 (config as ESP32 Dev Module with 8MB Flash)

### Device Description

2 DevKit modules on breadboard sharing power supply from USB
1 output pin from one device attached to 1 input pin from the other device
1 button connected at one end to the positive and at the other end to 1 input pin of each device (shared button)

### Hardware Configuration

GPIO21 from one device attached as OUTPUT to GPIO21 from the other device as INPUT_PULLDOWN
GPIO33 attached to a button as INPUT_PULLDOWN (for 2 devices)

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Measuring wakeup time from deepsleep:
There are two devices, A and B, and a button conected to both.
Device A goes into Deep Sleep mode and wakes up by pressing the button. In its setup() the first thing it does is activate the output of pin 33 which is connected to device B.
Device B measures the time that elapses from when the button is pressed until it receives the signal from pin 33 of device A, that is, until device A starts executing its setup().
Results:
If device A has PSRAM enabled, wakeup time is 537 ms. If it has PSRAM disabled, wakeup time is 101 ms.

Expected behaviour:
Not increase wakeup time when PSRAM is enabled

### Sketch

```cpp
// SKETCH FOR DEVICE A:

#define OUTPUT_PIN 21
#define BUTTON 33

void setup() {
  pinMode(OUTPUT_PIN, OUTPUT);
  pinMode(BUTTON, INPUT_PULLDOWN);

  digitalWrite(OUTPUT_PIN, 1);
  delay(1000);
  
  esp_deep_sleep_disable_rom_logging();

  esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);

  esp_sleep_enable_gpio_wakeup();

  esp_sleep_enable_ext1_wakeup(0x200000000, ESP_EXT1_WAKEUP_ANY_HIGH);  // PIN 33

  esp_deep_sleep_start();
}

void loop() {}

///////////////////////

// SKETCH FOR DEVICE B:

#define INPUT_PIN 21
#define BUTTON 33

uint32_t buttonTime;
uint32_t wakeupTime;

void setup()
{
  Serial.begin(115200);

  pinMode(INPUT_PIN, INPUT_PULLDOWN);
  pinMode(BUTTON, INPUT_PULLDOWN);

  Serial.println("Waiting...");
}

void loop()
{
  if (digitalRead(BUTTON))
  {
    buttonTime = micros();

    while (!digitalRead(INPUT_PIN))
      delay(1);

    wakeupTime = micros() - buttonTime;

    Serial.print("Wakeup time = ");
    Serial.println(wakeupTime);

    delay(1000);
  }

  delay(1);
}
```

### Debug Message

```plain
No error messages
```

### Other Steps to Reproduce

Removing in the sketch for device A the line

`esp_deep_sleep_disable_rom_logging();`

the wakeup time increases by about 20 ms in any case

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-06-24:
try adding to your sketch:
```cpp
bool testSPIRAM(void){ return true; }
```

We could probably detect if it's deep-sleep wakeup and skip the test automatically

Algonsi, 2022-06-24:
> try adding to your sketch:
> 
> ```c++
> bool testSPIRAM(void){ return true; }
> ```
> 
> We could probably detect if it's deep-sleep wakeup and skip the test automatically

Thank you for your answer.
Sorry but I don't know exactly what to do with

> bool testSPIRAM(void){ return true; }

I would appreciate it if you were more explicit.

me-no-dev, 2022-06-24:
just add that line to your sketch. It will cause the PSRAM test to be skipped and boot faster. I really meant you to just add it to your sketch :)

Algonsi, 2022-06-24:
Thanks. I didn't understand you. I had to literally do what you said :)
Ok, I did and that's a good solution.
Adding testSPIRAM definition to device A sketch, with PSRAM enabled, time is reduced to 117ms (vs 537ms)
Would this be the lowest limit we could get for the wakeup time with PSRAM enabled?

lbernstone, 2022-06-24:
No, it is the lowest startup time you will get from arduino-esp32 with psram enabled.  If you compile your own bootloader to skip some other checks and use only esp-idf, you can likely get the boot time under 20 ms.

Algonsi, 2022-06-24:
I see. It would be very recommendable using ESP-IDF. By now I find it a bit difficult, but it is in my todo list :)
Thak you!