espressif/arduino-esp32 #6914: Starting to get assert failed: xQueueGenericSend Release >= 2.0.0

savejeff, 2022-06-26:
### Board

ESP32 DevModule, also ESP32-S3

### Device Description

Soldered and Breadboard. 

### Hardware Configuration

Dual Core
Usage of SPI, I2C and Serial on Both cores but Hardware Ports clearly separated to one or the other core to avoid race conditions

### Version

v2.0.3

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

240Mhu

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm Getting this error from time to time now

assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

I get this on EndTransaction for Hardware SPI as well as Hardware I2C

[image]

**MY QUESTION: What does this assert msg mean? It looks to me like a mutex locking failure but the msg is quite cryptic.**

### Sketch

```cpp
Crash happens inside IDF / RTOS Code (see Debug Message part)
```

### Debug Message

```plain
Here the full traceback 

assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Decoding stack results
0x400836ad: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4009328d: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 128
0x40098249: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 85
0x40093c16: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 820
0x400ef731: spiEndTransaction at C:/Users/Save_/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c line 1153
0x400e0f5e: SPIClass::endTransaction() at C:/Users/Save_/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp line 155
0x40183f61: SdSpiArduinoDriver::deactivate() at lib/SdFat/src/SpiDriver/SdSpiESP.cpp line 48
0x401838b5: SdSpiCard::spiStop() at lib/SdFat/src/SdCard/SdSpiCard.h line 307
0x40183a39: SdSpiCard::writeStop() at lib/SdFat/src/SdCard/SdSpiCard.cpp line 863
0x40183a5f: SdSpiCard::syncDevice() at lib/SdFat/src/SdCard/SdSpiCard.cpp line 672
0x40183af5: SdSpiCard::cardCommand(unsigned char, unsigned int) at lib/SdFat/src/SdCard/SdSpiCard.cpp line 381
0x40183e2c: SdSpiCard::writeStart(unsigned int) at lib/SdFat/src/SdCard/SdSpiCard.cpp line 819
0x40183eba: SdSpiCard::writeSectors(unsigned int, unsigned char const*, unsigned int) at lib/SdFat/src/SdCard/SdSpiCard.cpp line 784
0x4018125e: SdSpiCard::writeSector(unsigned int, unsigned char const*) at lib/SdFat/src/SdCard/SdSpiCard.h line 220
0x40183fcf: FsCache::sync() at lib/SdFat/src/common/FsCache.cpp line 57
0x4018401d: FsCache::get(unsigned int, unsigned char) at lib/SdFat/src/common/FsCache.cpp line 35
0x40182af6: FatFile::write(void const*, unsigned int) at lib/SdFat/src/FatLib/FatPartition.h line 266
0x400ec116: FsBaseFile::write(void const*, unsigned int) at lib/SdFat/src/FsLib/FsFile.h line 783
0x400ec7d8: jFileSdFat::write(unsigned char const*, unsigned int) at lib/SdFat/src/ExFatLib/../common/ArduinoFiles.h line 156
0x400df83b: jLogFile::writeBytes(unsigned char*, unsigned int) at src/Storage/jLogFile.cpp line 286
0x400dfcd4: jLogFile_RingBuffered::Writeout(bool) at src/Storage/jLogFile_RingBuffered.cpp line 212
0x400dfd05: jLogFile_RingBuffered::Close(bool) at src/Storage/jLogFile_RingBuffered.cpp line 55
0x400de04d: storage_close_eventfile() at src/StorageModule.cpp line 315
0x400dedea: StorageModule_End() at src/StorageModule.cpp line 775
0x400dadd8: Mode_HaltClass::enter_Mode() at src/Mode_Halt.cpp line 141
0x400dbd9b: OperationModeClass::RunCore0() at src/OperationMode.cpp line 101
0x400e0a2a: Main0loop() at src/main0.cpp line 124
0x400e0b06: _main0() at src/main0.cpp line 55
0x400e0877: Task1code(void*) at src/main.cpp line 136
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-06-26:
The error means that the mutex either has not been taken, or that it has been taken by another thread. Make sure that begin/end transmission/transaction happens and that it happens in the same thread. After you end transaction/transmission, then you can begin again in another thread.

savejeff, 2022-06-27:
Play thanks. that is what i was guessing this is. 
I have already looked through all my I2C and SPI Transaction code andere there is always a endTransmission after a start Transmission. 
I run two tasks. each pinned to one of the cores. 
Is is possible to have one hardware I2C (Wire) used by on one core and the other hardware I2C (Wire1) simultaneously used by the other core ? 
This this should be supported, then there might be a bug/race condition. I currently have switched over to software implementations of I2C to avoid crashes. I can tests it again with the hardware I2Cs to make sure.

me-no-dev, 2022-06-27:
yes, you can also run the same Wire in both tasks. As long as everything is wrapped in transactions, it should be OK. Once you begin in one task, the other will wait until the first one ends the transaction.

savejeff, 2022-06-28:
is this possible, officially supported: 

[Core0/Task0]: Wire.beginTransaction();
[Core1/Task1]: Wire1.beginTransaction();
[.....]
[Core1/Task1]: Wire1.endTransaction();
[Core0/Task0]: Wire.endTransaction();

me-no-dev, 2022-06-28:
yes

savejeff, 2022-06-28:
okay i'll keep an eye on that. I'll create a new bug report if i find something with substance

closed 2022-06-28 by savejeff

renanrms, 2023-08-14:
I'm getting this error, but not using I2C at moment. Could this be caused by using Serial.print()?

I'm working on a lib, and this is the header file that contains most of the includes and headers:
https://github.com/renanrms/FTRLab-lib/blob/main/include/FTRLab/Device.hpp

This creates two tasks: one for measure sensors and other to communication. But I can not see any common usage of resources that can cause this problem.

```
assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace: 0x40083781:0x3ffb46c0 0x4008d399:0x3ffb46e0 0x40092fc9:0x3ffb4700 0x4008de4e:0x3ffb4830 0x400d3f6e:0x3ffb4870 0x400d2dd1:0x3ffb4890 0x400d2ac6:0x3ffb4900 0x400d2641:0x3ffb4930 0x400d1ffe:0x3ffb49b0
```

I'm new with esp32 and I don't know how to decode this backtrace and get the call stack.

I would apreciate If anyone has a hint as to what might be causing the problem based on the libs I'm using or this description.