espressif/arduino-esp32 #6920: ESP32-C3  ESP.getCycleCount()

fanfanlatulipe26, 2022-06-28:
### Board

LILYGO®TTGO T-01C3 ESP32-C3 and ESP32 Dev Module

### Device Description

None

### Hardware Configuration

No

### Version

v2.0.3

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am facing some issues with espsoftwareserial on ESP32C3 and this lead me to look at  ESP.getCycleCount() which is used for accurate timing in the library and seems to give incorrect measure.
I made a small test and try to get the number of cycle during 1s:  should be more or less the clock frequency ??
On the ESP32C3 it looks like the results are not the same over a given period of time depending on the activity.
On ESP32 results seems more in line.

### Sketch

```cpp
// Test  ESP.getCpuFreqMHz())   ESP.getCycleCount();
// Results 
/* 
  ESP32C3:  LILYGO®TTGO T-01C3 ESP32-C3  compiled with ESP32C3 dev kit 160Mhz
14:01:39.948 -> Starting ...
14:01:39.948 -> ESP.getCpuFreqMHz(): 160
14:01:40.951 -> Nbr of cycle with while: 159852038
14:01:41.952 -> Nbr of cycle with delay: 1344803
14:01:42.953 -> Nbr of cycle with while: 159875882
14:01:43.956 -> Nbr of cycle with delay: 1339365
14:01:44.959 -> Nbr of cycle with while: 159875882
14:01:45.961 -> Nbr of cycle with delay: 1339365
Result ESP32 dev kit cpu freq 230mhz
14:08:41.546 -> ESP.getCpuFreqMHz(): 240
14:08:42.549 -> Nbr of cycles with while: 239768414
14:08:43.551 -> Nbr of cycles with delay: 239849010
14:08:44.551 -> Nbr of cycles with while: 239879752
14:08:45.553 -> Nbr of cycles with delay: 239849988

*/
void setup() {
  Serial.begin(115200);
  Serial.println("Starting ...");
  Serial.printf("ESP.getCpuFreqMHz(): %u\n", ESP.getCpuFreqMHz());
  uint32_t curCycle = ESP.getCycleCount();
}

void loop() {

  long T0 = millis();
  uint32_t startCycle = ESP.getCycleCount();
  while (millis() < T0 + 1000) {}
  uint32_t endCycle = ESP.getCycleCount();
  Serial.printf("Nbr of cycles with while: %u\n", endCycle - startCycle);
  startCycle = ESP.getCycleCount();
  delay(1000);
  endCycle = ESP.getCycleCount();
  Serial.printf("Nbr of cycles with delay: %u\n", endCycle - startCycle);
}
```

### Debug Message

```plain
ESP32C3:  LILYGO®TTGO T-01C3 ESP32-C3  compiled with ESP32C3 dev kit 160Mhz
14:01:39.948 -> Starting ...
14:01:39.948 -> ESP.getCpuFreqMHz(): 160
14:01:40.951 -> Nbr of cycle with while: 159852038
14:01:41.952 -> Nbr of cycle with delay: 1344803
14:01:42.953 -> Nbr of cycle with while: 159875882
14:01:43.956 -> Nbr of cycle with delay: 1339365
14:01:44.959 -> Nbr of cycle with while: 159875882
14:01:45.961 -> Nbr of cycle with delay: 1339365
Result ESP32 dev kit cpu freq 230mhz
14:08:41.546 -> ESP.getCpuFreqMHz(): 240
14:08:42.549 -> Nbr of cycles with while: 239768414
14:08:43.551 -> Nbr of cycles with delay: 239849010
14:08:44.551 -> Nbr of cycles with while: 239879752
14:08:45.553 -> Nbr of cycles with delay: 239849988
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

fanfanlatulipe26, 2022-09-20:
Any news ?
This issue prevent the use of espsoftwareserial on ESP32C3 as soon as a delay(xx) is use anywhere.
Example: As soon as a wifi server is active espsoftwareserial witl fail on ESP32-C3
Is there a way to prevent a slow down of the clock  ??

avillacis, 2022-10-03:
I was asked to provide tests on an ESP32-S2, so here they are, using the same sketch as shown above. There appears to be no problem on ESP32-S2. This was with Arduino-ESP32 2.0.5, default CPU frequency of 240 MHz:

```
15:53:26.508 -> ESP-ROM:esp32s2-rc4-20191025
15:53:26.508 -> Build:Oct 25 2019
15:53:26.508 -> rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
15:53:26.508 -> SPIWP:0xee
15:53:26.508 -> mode:DIO, clock div:1
15:53:26.508 -> load:0x3ffe6100,len:0x524
15:53:26.508 -> load:0x4004c000,len:0xa70
15:53:26.508 -> load:0x40050000,len:0x292c
15:53:26.508 -> entry 0x4004c18c
15:53:26.806 -> [   698][I][esp32-hal-psram.c:96] psramInit(): PSStarting ...
15:53:26.806 -> ESP.getCpuFreqMHz(): 240
15:53:27.768 -> Nbr of cycles with while: 239834208
15:53:28.796 -> Nbr of cycles with delay: 239784417
15:53:29.791 -> Nbr of cycles with while: 239966350
15:53:30.786 -> Nbr of cycles with delay: 239781373
15:53:31.814 -> Nbr of cycles with while: 239966275
15:53:32.776 -> Nbr of cycles with delay: 239782438
15:53:33.804 -> Nbr of cycles with while: 239966350
15:53:34.799 -> Nbr of cycles with delay: 239782192
15:53:35.794 -> Nbr of cycles with while: 239966380
15:53:36.822 -> Nbr of cycles with delay: 239783236
```

fanfanlatulipe26, 2022-11-11:
The function pulseIn may also be strongly impacted on ESP32C3 because it uses ESP.getCycleCount
We should give the user of Arduino environment a way to control power/clock management .

closed 2025-01-16 by Parsaabasi

VovanPitersky, 2025-11-21:
Все чипы Esp32-C3 - КРИВЫЕ !!! увы
в них кривые тайминги контроллера, поэтому использовать в них ESP.getCycleCount() не возможно
кривизна тайминга заключается в том что, что каждые 2 секунды тайминг ступенчато тормозится на 4 миллисекунды )))
сам я это выяснил совершенно случайно - когда использовал внешний задающий генератор тактов
перепроверил кучу производителей плат - результат одинаковый и не зависит от производителя платы
то есть КРИВОЙ именно чип ))) и это ни как не лечится ))))
просто ЗАБУДЬТЕ про Esp32-C3, обратите свой взор на другие виды чипов Esp32, там таких проблем - НЕТ )))
УДАЧИ!

All Esp32-C3 chips are CURVED!!! alas
they have curved timings of the controller, so it is not possible to use ESP.getCycleCount() in them. 
the curvature of the timing lies in the fact that every 2 seconds the timing is gradually slowed down by 4 milliseconds)))
I myself found out completely by accident - when I used an external master clock generator
I rechecked a bunch of board manufacturers - the result is the same and not it depends on the manufacturer of the board
that is, it is the chip that is CURVED))) and this is not treated in any way))))
just FORGET about the Esp32-C3, pay attention to other types of Esp32 chips, there are NO such problems there)))
good luck!