espressif/arduino-esp32 #6921: No real-time serial.read() possible

OekoSolveMG, 2022-06-28:
### Board

M5 Stack Core 1 / 2

### Device Description

[M5 Stack Core 1 / 2](https://docs.m5stack.com/en/core/core2) with the [LAN Module with W5500](https://docs.m5stack.com/en/base/lan_v12)

### Hardware Configuration

-

### Version

v2.0.2

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Low baudrate (9600) causes the `serial.read()` command to fail and not read bytes on the Modbus. Might there be some interfering code that runs in timed cycles with a very high priority that could creates issues for the `serial.read()`.

[image]

The failed read seen above happens every 6th read.

[image]

**Complete message:**

The third line is a pin that gets toggled to high before `serial.read()` and to low after  `seral.read()`.

[image]

### Sketch

```cpp
// Allocate initial receive buffer size: 1 block of BUFBLOCKSIZE bytes
  const uint16_t BUFBLOCKSIZE(512);
  uint8_t *buffer = new uint8_t[BUFBLOCKSIZE];
  ModbusMessage rv;

  // Index into buffer
  register uint16_t bufferPtr = 0;
  // Byte read
  register int b; 

  // State machine states, RTU mode
  enum STATES : uint8_t { WAIT_DATA = 0, IN_PACKET, DATA_READ, FINISHED };

  // State machine states, ASCII mode
  enum ASTATES : uint8_t { A_WAIT_DATA = 0, A_DATA, A_WAIT_LEAD_OUT, A_FINISHED };

  register uint8_t state;

  // Timeout tracker
  unsigned long TimeOut = millis();

  // RTU mode?
  if (!ASCIImode) {
    // Yes.
    state = WAIT_DATA;
    // interval tracker 
    lastMicros = micros();
  
    while (state != FINISHED) {
      switch (state) {
      // WAIT_DATA: await first data byte, but watch timeout
      case WAIT_DATA:
        // Blindly try to read a byte
        b = serial.read();
        // Did we get one?
        if (b >= 0) {
          // Yes. Note the time.
          lastMicros = micros();
          // Do we need to skip it, if it is zero?
          if (b > 0 || !skipLeadingZeroBytes) {
            // No, we can go process it regularly
            state = IN_PACKET;
          } 
        } else {
          // No, we had no byte. Just check the timeout period
          if (millis() - TimeOut >= timeout) {
            rv.push_back(TIMEOUT);
            state = FINISHED;
          }
          delay(1);
        }
        break;
      // IN_PACKET: read data until a gap of at least _interval time passed without another byte arriving
      case IN_PACKET:
        // Are we past the interval gap without another byte?
        if (micros() - lastMicros >= interval) {
          // Yes, terminate reading
          LOG_V("%ldus without data\n", micros() - lastMicros);
          state = DATA_READ;
        } else {
          // No, still in reading sequence
          // Did we get a byte?
          if (b >= 0) {
            // Yes, collect it
            buffer[bufferPtr++] = b;
            // Mark time of last byte
            lastMicros = micros();
            // Buffer full?
            if (bufferPtr >= BUFBLOCKSIZE) {
              // Yes. Something fishy here - bail out!
              rv.push_back(PACKET_LENGTH_ERROR);
              state = FINISHED;
              break;
            }
          }
          // Buffer has space left - try to read another byte
          b = serial.read();
        }
        break;
      // DATA_READ: successfully gathered some data. Prepare return object.
      case DATA_READ:
        // Did we get a sensible buffer length?
        HEXDUMP_D("Raw buffer received", buffer, bufferPtr);
        tmp_copy.assign(buffer, buffer + bufferPtr);

        if (bufferPtr >= 4)
        {
          // Yes. Check CRC
          if (!validCRC(buffer, bufferPtr)) {
            // Ooops. CRC is wrong.
            rv.push_back(CRC_ERROR);
            tmp_failed = true;
          } else {
            // CRC was fine, Now allocate response object without the CRC
            for (uint16_t i = 0; i < bufferPtr - 2; ++i) {
              rv.push_back(buffer[i]);
            }
          }
        } else {
          // No, packet was too short for anything usable. Return error
          rv.push_back(PACKET_LENGTH_ERROR);
          tmp_failed = true;
        }
        state = FINISHED;
        break;
      // FINISHED: we are done, clean up.
      case FINISHED:
        // CLear serial buffer in case something is left trailing
        // May happen with servers too slow!
        while (serial.available()) serial.read();
        break;
      }
    }
  }
  // Deallocate buffer
  delete[] buffer;

  return rv;
```

### Debug Message

```plain
[244292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [23], Content: [02 5E EA 06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 2B A9 ]
[244300][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[269243][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7B 00 04 01 3C 00 00 00 00 02 79 01 ]
[269248][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[269292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [23], Content: [F9 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 33 EA ]
[269298][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[294243][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7C 00 04 01 3C 00 00 00 00 02 79 01 ]
[294248][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[294293][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [22], Content: [01 EA 06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 49 01 ]
[294299][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[319242][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [18], Content: [01 03 3E 00 71 02 7C 00 04 01 3C 00 00 00 00 02 79 00 ]
[319247][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[319292][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [23], Content: [FF 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 8C 91 ]
[319297][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[339351][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [3], Content: [D3 12 FF ]
[339353][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E5) - (Packet length error).

[344233][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [7], Content: [01 03 3E 00 71 02 7C ]
[344235][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[344294][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [35], Content: [7A 00 00 02 B9 00 00 0B D6 27 28 02 79 4C 9F 00 00 00 08 00 50 0F 71 0C A8 00 01 00 01 
00 64 00 03 A2 09 ]
[344303][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[364269][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [42], Content: [01 03 3E 00 71 02 7B 00 04 01 3D 00 00 00 00 02 79 01 01 00 78 FF FF 00 00 02 72 00 01 
00 01 01 84 00 00 02 B9 00 00 0B D6 27 ]
[364281][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[389244][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [19], Content: [01 03 3E 00 71 02 7C 00 04 01 3C 00 00 00 00 02 79 01 00 ]
[389249][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).

[389295][E-MB][DUMP] (.pio/libdeps/m5stack-core-esp32/eModbus/src/RTUutils.cpp:497) - Size: [20], Content: [06 00 00 08 00 50 0F 71 0C A8 00 01 00 01 00 64 00 03 D2 F8 ]
[389300][MB][E] (lib/modbus/ModbusMaster.cpp:304) - Error: (E2) - (CRC check error).
```
```

### Other Steps to Reproduce

See issue on the eModbus library about the same problem. Conclusion was that it might be a problem with the Base library. (https://github.com/eModbus/eModbus/issues/198)

The same issue happens on both Core 1 and Core 2 as well as with a custom adapter for modbus or the WAN 5500 module.

I wanted to try it with Arduino 2.0.3 as well, but it seems to not yet be available for Platform IO, will try as soon as it is available tough.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-06-28:
Hello @OekoSolveMG, thanks for your precise issue report. Is there any  reason for you to run your code on Arduino v.2.0.2? I will suggest to retest your code on v2.0.3, there were some changes made on Serial.

OekoSolveMG, 2022-06-28:
Thanks a lot for the fast response as written above, I wanted to try it but it seems that it hasn't been included in platformio yet.

bertmelis, 2022-06-28:
> Thanks a lot for the fast response as written above, I wanted to try it but it seems that it hasn't been included in platformio yet. I'll definetly try it once it is available tough.

You're sure about that? I have 2.0.3 on VSCode/Platformio:
[image]

OekoSolveMG, 2022-06-28:
> You're sure about that? I have 2.0.3 on VSCode/Platformio:

Nvm seems I have the newest version as well, was confused because at least in the release note the corresponding espressif32 version is written to be `4.4.1` and not `4.4.0`.

```
Resolving m5stack-core-esp32 environment packages...        
Platform espressif32 @ 4.4.0 (required: espressif32 @ 4.4.0)
├── framework-arduinoespressif32 @ 3.20003.220626 (required: platformio/framework-arduinoespressif32 @ ~3.20003.0)
├── framework-espidf @ 3.40302.0 (required: platformio/framework-espidf @ ~3.40302.0)
```

[image]

Miq1, 2022-06-28:
So it looks like the issue is persisting despite the recent changes to the Serial handling.

SuGlider, 2022-06-28:
@OekoSolveMG 

Could you please try this solution:

``` cpp
// Necessary include for testing the fix
#include "driver/uart.h"

void setup() {
  // for example, start Serial - UART0
  Serial.begin(115200);

  // right after starting UART0, add this code:
  uart_intr_config_t uart_intr = {
      .intr_enable_mask = UART_INTR_RXFIFO_FULL | UART_INTR_RXFIFO_TOUT,
      .rxfifo_full_thresh = 112,
      .rx_timeout_thresh = 1,
      .txfifo_empty_intr_thresh = 10,
  };
  uart_intr_config((uart_port_t) 0, &uart_intr);  // Zero is the UART number for Arduino Serial
}
```

I think that adding this fix may make UART RX faster and "real time".
Please try it in the same testing environment used to this issue and let me know.

Thanks!

OekoSolveMG, 2022-06-28:
The pin seems to be toggling differently now. The read doesn't seem to start until the message has been sent completly. Will let it run over night and see if the issue doesn't occur anymore but the LA result looks different (Especially relevenat is the third line that toggles to HIGH before `serial.read` and toggle to low after `serial.read`.

**LA result with changes above:**
[image]

**Old LA result with no changes:**
[image]

Miq1, 2022-09-19:
Excellent, thanks! I already deployed the ``setRxFIFOFull(()`` method to get the ISR trigger for each single byte and it works well for me.

SuGlider, 2022-10-28:
I have tested the changes from PR #6930 that adds `HardwareSerial::setRxFIFOFull(uint8_t fifoBytes)` API.
It solves this issue for sure. It allows Arduino to read byte by byte from UART as it is received.

The testing was done receiving 125 bytes to UART RX from an external device at 9600 baud.

#### Default Arduino 2.0.5 Behavior (after just using `Serial.begin(9600)`):
It will wait to receive 120 bytes before Serial.read() gets any data. The FIFO full default value is 120, thus filling up the Arduino Queue will happen after UART FIFO reaches 120 bytes.
Arduino gets 120 bytes in 126 ms. Each byte has about 10 bits (8 + parity bit + stop bit). 
Therefore, it takes 1/9600 * 120 * 10 ~ 125ms for Arduino to read this first chunk of data...
And the remaining 5 bytes in 6 ms after previous event. About 1/900 * 6 * 10 = 6.25ms, in a second chunk of data.

This is **not the Real Time** desired behaviour.

####  Using setRxFIFOFull(1) to make each byte to be copied to the Arduino internal HardwareSerial buffer as soon as it gets to the UART:
It will get byte by byte the Serial.read() will get 1 byte every 1 ms at 9600 baud. 
The time is 1/9600 x 10 ~1ms, per byte transmited.

This is exactly the **Real Time** requested behaviour on this issue.

In order to make it work just use this code snippet:

``` cpp
  Serial.begin(9600);
  Serial.setRxFIFOFull(1);
```

closed 2022-10-28 by SuGlider