espressif/arduino-esp32 #6922: ESP32S3 - HID  USBHID  SendReport error and USB Replug

RichardsATcn, 2022-06-28:
### Board

ESP32-S3-WROOM-1  Moudle

### Device Description

ESP32-S3-WROOM-1

### Hardware Configuration

[image]
 only use USB ；

### Version

v2.0.3

### IDE Name

Arduino  1.8.19

### Operating System

Windows 10

### Flash frequency

QIO 80

### PSRAM enabled

no

### Upload speed

115200

### Description

[ESP32-S3-USB.zip](https://github.com/espressif/arduino-esp32/files/8999685/ESP32-S3-USB.zip)

[src.zip](https://github.com/espressif/arduino-esp32/files/8999715/src.zip)

USB device will appear error after using for a while.

If the CDC serial port is enabled, it will be easier to send at fixed time.

Devices will be re enumerated repeatedly

The error code segment is as follows

`  bool res = ready();
    if (!res)
    {
        log_e("not ready");
    }
    else
    {
        res = tud_hid_n_report(0, id, data, len);
        if (!res)
        {
            log_e("report %u failed", id);
        }
        else
        {
            xSemaphoreTake(tinyusb_hid_device_input_sem, 0);
            if (xSemaphoreTake(tinyusb_hid_device_input_sem, timeout_ms / portTICK_PERIOD_MS) != pdTRUE)
            {
                log_e("report %u wait failed", id);
                res = false;
            }
        }
    }
    xSemaphoreGive(tinyusb_hid_device_input_mutex);

//Added code snippet
    if (tinyusb_hid_device_input_sem)
    {
        xSemaphoreGive(tinyusb_hid_device_input_sem);
    }
    log_d("report %u wait over %s", id, res == true ? "true" : "false");
//Added code snippet

    return res;`

usb设备使用一会儿会出现错误。
如果cdc串口开启定时发送时候会更容易出现。
设备会反复的 重新枚举

### Sketch

```cpp
//Added code snippet
    if (tinyusb_hid_device_input_sem)
    {
        xSemaphoreGive(tinyusb_hid_device_input_sem);
    }
    log_d("report %u wait over %s", id, res == true ? "true" : "false");
//Added code snippet
```

### Debug Message

```plain
17:27:54.138 -> [ 34283][E][USBHID.cpp:410] SendReport(): report 3 wait failed
17:27:54.138 -> [ 34284][D][USBHID.cpp:420] SendReport(): report 3 wait over false
17:27:54.234 -> USB PLUGGED
17:27:54.330 -> [ 34475][E][USBHID.cpp:410] SendReport(): report 3 wait failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-07-27:
This will be triage by @SuGlider, in next 2 weeks we will specify next steps.

SuGlider, 2022-08-10:
@RichardsATcn - It seems that the code you posted is some modified version of "official" USBHID.cpp - Not sure how to help you. Please confirm.

Can you try using Arduino Core 2.0.4?
Could you please post a Arduino Sketch that reproduces the issue, using **only official Arduino Core**?

Thanks.

VojtechBartoska, 2022-08-24:
@RichardsATcn any updates?

hoantv, 2023-04-08:
same issue with esps3 wroom 1U
Just use gamepad example, 
[  7621][E][USBHID.cpp�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
[    93][D][esp32-hal-tinyusb.c:675] tinyusb_enable_interface(): Interface CDC enabled
[    94][D][esp32-hal-tinyusb.c:675] tinyusb_enable_interface(): Interface MSC enabled
[    98][D][esp32-hal-tinyusb.c:675] tinyusb_enable_interface(): Interface HID enabled
[   105][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[0] len: 68
[   160][D][FirmwareMSC.cpp:94] msc_update_get_required_disk_sectors(): APP size: 336928 (659 sectors)
[   160][D][FirmwareMSC.cpp:100] msc_update_get_required_disk_sectors(): OTA size: 1310720 (2560 sectors)
[   167][D][FirmwareMSC.cpp:113] msc_update_get_required_disk_sectors(): USING FAT12
[   174][D][FirmwareMSC.cpp:116] msc_update_get_required_disk_sectors(): FAT sector size: 512
[   183][D][FirmwareMSC.cpp:117] msc_update_get_required_disk_sectors(): FAT data sectors: 3235
[   191][D][FirmwareMSC.cpp:118] msc_update_get_required_disk_sectors(): FAT table sectors: 10
[   199][D][FirmwareMSC.cpp:119] msc_update_get_required_disk_sectors(): FAT total sectors: 3247 (1623KB)
[   209][D][esp32-hal-tinyusb.c:564] tinyusb_load_enabled_interfaces(): Load Done: if_num: 4, descr_len: 130, if_mask: 0x15
[   220][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 3 = 160 Mhz, APB: 80000000 Hz
[   236][E][USBHID.cpp:346] SendReport(): not ready
[   236][E][USBHID.cpp:346] SendReport(): not ready
[   237][E][USBHID.cpp:346] SendReport(): not ready
[   242][E][USBHID.cpp:346] SendReport(): not ready
[   246][E][USBHID.cpp:346] SendReport(): not ready
[   251][E][USBHID.cpp:346] SendReport(): not ready
[   256][E][USBHID.cpp:346] SendReport(): not ready
[   260][E][USBHID.cpp:346] SendReport(): not ready
[   265][E][USBHID.cpp:346] SendReport(): not ready

SuGlider, 2023-04-10:
@hoantv - I have tested it with Win11. The problem is that Win doesn't know what is START_BUTTON.

I tested it with an ESP32-S3 and when pressing the BOOT button on the board, the Joystick moves to the right bottom corner and returns back to the center.

This is the code that I have tested:

``` cpp
#if ARDUINO_USB_MODE
#warning This sketch should be used when USB is in OTG mode
void setup() {}
void loop() {}
#else
#include "USB.h"
#include "USBHIDGamepad.h"
USBHIDGamepad Gamepad;

const int buttonPin = 0;
int previousButtonState = HIGH;

void setup() {
  Serial.begin(115200);
  pinMode(buttonPin, INPUT_PULLUP);
  Gamepad.begin();
  USB.begin();
}

void loop() {
  int buttonState = digitalRead(buttonPin);
  if ((buttonState != previousButtonState) && (buttonState == LOW)) {
    Gamepad.leftStick(100, 100);
    delay(500);
    Gamepad.leftStick(0, 0);
    Serial.println("Button Pressed - gamePad should have sent event.");
  }
  previousButtonState = buttonState;
}
#endif /* ARDUINO_USB_MODE */
```

SuGlider, 2023-04-10:
@RichardsATcn - I used Arduino Core 2.0.7 and it works fine also when USB is unplugged and plugged back.

This issue is solved.

closed 2023-04-10 by SuGlider

SuGlider, 2023-04-10:
@hoantv - Check PR #8058