espressif/arduino-esp32 #6923: ledc fails since 2.0.2 for Arduino-IRremote

ArminJo, 2022-06-28:
### Board

ESP32DevModule

### Device Description

no

### Hardware Configuration

Ledc Output pin is D4

### Version

v2.0.3

### IDE Name

Arduino, Sloeber

### Operating System

%

### Flash frequency

%

### PSRAM enabled

no

### Upload speed

%

### Description

see https://github.com/Arduino-IRremote/Arduino-IRremote/issues/997

### Sketch

```cpp
SendDemo example of the Arduino-IRremote library
can be reduced to:
ledcSetup(0, 38000, 8);
ledcAttachPin(4, 0);
ledcWrite(0, 76);
delay(1);
ledcWrite(0, 0);
delay(1);
ledcWrite(0, 76);
delay(1);
ledcWrite(0, 0);
delay(1);
...
```

### Debug Message

```plain
%
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-06-28:
@P-R-O-C-H-Y PTAL.

ArminJo, 2022-06-28:
Sorry, the correct sketch is:
void setup() {
  Serial.begin(115200);
  ledcSetup(0, 38000, 8);
  ledcAttachPin(4, 0);
  **pinMode(4, OUTPUT);** // this is not required for ESP but for all AVR CPUs...
}

void loop() {
   ledcWrite(0, 76);
   delay(1);
   ledcWrite(0, 0);
   delay(1);
}

closed 2022-06-28 by ArminJo