espressif/arduino-esp32 #6925: DNS_PROBE_FINISHED_NXDOMAIN on Android.

zekageri, 2022-06-29:
### Board

esp-wrover-kit

### Device Description

ETH_LAN-8720

### Hardware Configuration

ETH_LAN-8720

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Android

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I get DNS_PROBE_FINISHED_NXDOMAIN when i try to access my esp32's webserver with local mdns like **test.local/**
I can reach it with it's IP adress and i can reach with local dns on windows and mac and iphones too.
Sometimes it is working on android too.

### Sketch

```cpp
void networkSystem::setupMDNS(){
    if( MDNS.begin( fileSys.config.hostname ) ) {
        MDNS.addService("http", "tcp", 80);
        MDNS.addServiceTxt("http", "tcp", "hsh_service", "HsH");
        MDNS.addServiceTxt("http", "tcp", "hsh_service_prop2", "HsH");
        #if NETWORK_DEBUG_MODE
            Serial.println("MDNS responder started");
        #endif
    }else{
        #if NETWORK_DEBUG_MODE
            Serial.println("MDNS responder failed");
        #endif
    }
}

void ethEvent(WiFiEvent_t event){
    Serial.printf("\nNS - WiFiEvent: %d\n", event);
    if( event == ARDUINO_EVENT_ETH_START ){
        ETH.setHostname(fileSys.config.hostname);
        networkSys.setupMDNS();
    }else if( event == ARDUINO_EVENT_ETH_GOT_IP ){
        networkSys.setupMDNS();
    }else if( event == ARDUINO_EVENT_ETH_DISCONNECTED ){
        #if NETWORK_DEBUG_MODE
            Serial.println("\nNS - Ethernet disconnected");
        #endif
    }
}
```

### Debug Message

```plain
DNS_PROBE_FINISHED_NXDOMAIN on every browser using android.
```

### Other Steps to Reproduce

nothing

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2022-06-29:
Android does not include mDNS support in their name resolver implementation by default. You can work around this limitation of the Android OS by swapping the DNS server in use to one that does implement mDNS resolving.

This is not a bug on the ESP32 side.

zekageri, 2022-06-29:
Android does support mDNS resolving and it was working, i used it for almost three week.
[https://www.androidpolice.com/android-mdns-local-hostname/](https://www.androidpolice.com/android-mdns-local-hostname/)

atanisoft, 2022-06-29:
@zekageri It's great that Android is finally catching up to the rest of the platforms that support mDNS natively. It looks like it will require Android 12+ which may not be available for most Android devices though.

This however is still not an ESP32 specific bug/limitation but Android specific issue.

closed 2022-06-29 by zekageri

zekageri, 2024-03-15:
Nothing. I did not experienced it again so far but i did not delete my cache either. I suspect that the bug is still there.