espressif/arduino-esp32 #6926: ESP32 BLE client connect to multiple servers

UmangSuthar101, 2022-06-30:
### Board

ESP32 DevKitc V4

### Device Description

- Development Kit: ESP32 DevKitc V4(attached picture)

- Module or chip used: ESP32-WROOM-32

- Compiler version: Extensa-esp32-elf-GCC

### Hardware Configuration

GPIO(34)  connected to the resistor n/w for getting analog value of the battery!!

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I upload the code for the BLE multi-client in this module.

After some time such logs appear in the console:
....
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 2)
....
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)

BLE disconnects happen too often and show me these types of logs.
[image]

### **Expected Behavior**

- Stable BLE connection 
- Can connect multiple BLE servers on this module and get the data 
   simultaneously

### Sketch

```cpp
bool connectToServer() {

  BLEClient*  pClient  = BLEDevice::createClient();
  pClient->setClientCallbacks(new MyClientCallback());
  pClient->connect(myDevice);
  Serial.println(" - Connected to Spo2");

  BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr) {
    pClient->disconnect();
    return false;
  }
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  if (pRemoteCharacteristic == nullptr) {
    pClient->disconnect();
    return false;
  }
  if (pRemoteCharacteristic->canNotify())
    pRemoteCharacteristic->registerForNotify(notifyCallback);
  return true;
}

//This upper code will write multiple times as per servers //numbers

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      // We have found a device, let us now see if it contains the service we are looking for.
      if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {
        BLEDevice::getScan()->stop();
        myDevice = new BLEAdvertisedDevice(advertisedDevice);
        doConnect = true;
        doScan = true;
      }
    }
};

void setup() {
  Serial.begin(115200);
  ble();
  ble1();
  ble2();
  ble3();
}

void ble() {
  BLEDevice::init("abc");

  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(1);
}

// multiple ble's loop functions
```

### Debug Message

```plain
....
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 2)
....
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)

Note: I already attached serial output for the perticular error.
```

### Other Steps to Reproduce

I go thruogh the espriff forum and also check github for HCI error but nothing get any solution.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-06-30:
@UmangSuthar101 
The issue header says you are using Arduino Core 1.0.6. It is based on an old IDF version for BLE.
Could you please update your Arduino Core to the latest version in the Arduino IDE.
Current Core version is 2.0.3.

Let me know is the issue still occurs.
Thanks.

UmangSuthar101, 2022-07-02:
@SuGlider , as you mentioned I update the esp-arduino to latest version 2.0.3 and again I got the following log check this:

Pulse: 84
SPO2: 99
Disconnect to Spo2 Device
Connect to Temp. Device
Connected to Temp.
We are now connected to the Temp. BLE Server.
Temp. : 98.26
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
Connect to  BP Device
Connected to BP
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Disconnect to  BP Device
We have failed to connect to the BP server; there is nothin more we will do.
Disconnect to Temp. Device

[image]

SuGlider, 2022-07-04:
@chegewara - Could you please help @UmangSuthar101 on this issue? Thanks!

UmangSuthar101, 2022-07-05:
Greetings of the day,

@SuGlider & @chegewara could you plz help me with this issue, its very difficult for me to get multiple ble server data to ESP-32 ble client !!!

chegewara, 2022-07-05:
This is low level error/warning log. I can only guessing it may be related to low memory, but this should be asked on esp-idf.

UmangSuthar101, 2022-07-05:
@chegewara I asked espriff team and they suggested me asking on esp-Arduino.

I also try this code on 8MB module but the same error show me, whenever I connect multiple BLE servers.

Is there any proper example that connects multiple BLE servers to ESP-32 BLE client?? Suggests me.

chegewara, 2022-07-05:
Again, `lld_pdu_*` are low level library messages, and we dont even have access to source code to ungerstand what it means.
If espressif is not willing to help you then maybe you should switch to nordic.

This is not problem with flash, so 4MB or 8MB makes no difference, but you should try to switch to NimBLE library.
Its easy, and ive read its working better, has better support and most important its using less RAM.

UmangSuthar101, 2022-07-05:
@chegewara ok, get it!! I'll check NimBLE.

But is there any example or reference link(with ArduinoBLE library) which already connects multiple servers to the ESP BLE client, if that one gives me success then I will go through it?

chegewara, 2022-07-05:
Im pretty sure you can find some, but nothing i am aware of.

SuGlider, 2022-07-05:
> Is there any proper example that connects multiple BLE servers to ESP-32 BLE client?? Suggests me.

There is an IDF example of GATTC connecting to 3 BLE Servers:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gattc_multi_connect

Tutorial and explanation:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gattc_multi_connect/tutorial/Gatt_Client_Multi_Connection_Example_Walkthrough.md

SuGlider, 2022-07-05:
@UmangSuthar101 
I believe that it's necessary to instantiate one BLEClient object per Server connection in BLE Arduino Layer in order to make it work the way you need it.

khalidnk83, 2022-07-30:
The best and more stable BLE stack is the "NimbleBLE"....
try using it from the link: https://github.com/h2zero/NimBLE-Arduino.git

SL06, 2022-10-27:
Hi, this program is working but need further improvement  on starting and stopping scan...
IMPORTANT: might not work with older version
   Arduino Version: 2.0.1-nightly-20221024
   ESP32 2.0.5
```
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * Copyright © 2022 Arduino SA
 * author unknown
 * updated by chegewara
 * modified for 3 connections by Sylvain Lareau
 *  usefull exemple https://www.youtube.com/watch?v=iCKIIMrphtg

 * IMPORTANT: might not work with older version
   Arduino Version: 2.0.1-nightly-20221024
   ESP32 2.0.5
The program is not final  and could easely be improve and simplified...
  
 */

#include "BLEDevice.h"

int LED_BUILTIN = 2;
bool toggle;

// The remote service we wish to connect to

//static BLEUUID0 serviceUUID("0000180d-0000-1000-8000-00805f9b34fb");  //0x180D
//BLEUUID become a type, like int or boolean
static BLEUUID serviceUUID0(BLEUUID((uint16_t)0x180D )); //Heart Rate
static BLEUUID charUUID0(BLEUUID((uint16_t)0x2A37));

static BLEUUID serviceUUID1(BLEUUID((uint16_t)0x1818 ));  // Cycling Power
static BLEUUID charUUID1(BLEUUID((uint16_t)0x2A63));

static BLEUUID serviceUUID2(BLEUUID((uint16_t)0xFEE0 ));  // SPO2
static BLEUUID charUUID2(BLEUUID((uint16_t)0xFEE1));

static boolean doConnect0 = false;
static boolean connected0 = false;

static boolean doConnect1 = false;
static boolean connected1 = false;

static boolean doConnect2 = false;
static boolean connected2 = false;

static boolean doScan0 = false;
static boolean doScan1 = false;
static boolean doScan2 = false;

static boolean notification0 = false;
static boolean notification1 = false;
static boolean notification2 = false;

static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback0(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    //Serial.print("Notify callback for characteristic ");
   // Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
   // Serial.print(" of data length ");
   // Serial.println(length);
   // Serial.print("data: ");
    //Serial.println((char*)pData);
[... 274 lines cut ...]
    if (notification1 == false) {
      Serial.println(F("Turning Notification On for sensor 1"));
      const uint8_t onPacket[] = {0x1, 0x0}; // not shure about these number but it works
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)onPacket, 2, true);
      notification1 = true;
    } }
    else if(doScan1){
    BLEDevice::getScan()->start(0);  // this is just example to start scan after disconnect, most likely there is better way to do it in arduino
   
  }

  if (connected2) {
    if (notification2 == false) {
      Serial.println(F("Turning Notification On for sensor 2"));
      const uint8_t onPacket[] = {0x1, 0x0}; // not shure about these number but it works
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)onPacket, 2, true);
      notification2 = true;
    } }
    else if(doScan2){
      BLEDevice::getScan()->start(5, false);  // this is just example to start scan after disconnect, most likely there is better way to do it in arduino
    }
  
  delay(1000); // Delay a second between loops.
} // End of loop
```

petrrpancon, 2023-04-03:
> There is an IDF example of GATTC connecting to 3 BLE Servers: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gattc_multi_connect
>
> Tutorial and explanation: [...]

I think [UmangSuthar101](https://github.com/UmangSuthar101) was interested in using Arduino API not IDF. I have the same question and have not found any good example so far.

petrrpancon, 2023-04-03:
[SL06](https://github.com/SL06) BLEDevice::createClient(); overwrites a pointer in BLEdevice so making second client will likely break some callback events to 1st one going through that pointer and create memory leaks. The BLEDevice does not support multiple instances either. There is no reference to the client instance in call backs so I do not see how the connection is managed. The lack of any example lets me to conclusion that was not written to support multiple server connections properly.

Melmac2, 2023-04-30:
petrrpancon - I would be interested in the code when finished.  I'm trying to connect a single ESP32 client to multiple identical BLE motion sensors (with different addresses) at the same time and read their notification data; Arduino IDE.

davidangelo64, 2023-10-23:
@petrrpancon could share the code please, I'm trying to connect more than two clients and get their rssi continously

petrrpancon, 2023-10-23:
Hi, the code is part of a project and I can not easily extract it, but let me show what i did.
getScan is singleton so it will give you the same instance every time. I would run it only from 1 thread anyway. 
I call this once at start:
{
    BLEDevice::init("");
    BLEScan *scn = BLEDevice::getScan();
    scn->setAdvertisedDeviceCallbacks(&scan_cb);
    scn->setActiveScan(false);
}

then I have 2 clients as example:
ClientExt1 client1;
ClientExt2 client2;

then call this in loop task
for(;;)
{
      if (should_scan_for_connection(client)) // what ever condition to start the search
      {
            BLEScan *scn = BLEDevice::getScan();
            scn->start(2, false);
            if (check_for_found_device(client1)) // BLEAdvertisedDeviceCallbacks was called
            {  
                 here you can use 
                client1.connect(BLEAddress address, esp_ble_addr_type_t type) 
                with data from the event advertisedDevice.getAddress()
            }      
            if (check_for_found_device(client2)) // BLEAdvertisedDeviceCallbacks was called
            {  
                 same here for the other client
            }      
      }
     else
     {
          block the thread until something changes
     }
}

scan_cb is the instance class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks

for event when device is found and you need to check if it is something compatible in this method:
void MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice advertisedDevice)
{
    check uuid or some other stuff matching your device here
    if you find the device you can stop the scan here immediately and store the address.
}
 
the advertisedDevice contains address to make connection advertisedDevice.getAddress()
so you can use that to connect to client.

you can create multiple instance of BLEClient, even extend this class with you own data. It woks as global instance or on heap with "new BLEClient". However managing dynamic data is harder to do right. Just do not use createClient. If you make more instances of the same ClientExt1 you check the client pointer inside any share callback so it still works for multiples.

class ClientExt1 : public BLEClient
{
    // data specific to your client
}

After the connection i run each client in its thread. It seem to work fine but don't expect a production quality from the arduino code. especially with threads. Esp32 had limit of 4 connections so be aware of that too.

closed 2025-01-16 by Parsaabasi