espressif/arduino-esp32 #6928: ESP32-C3 SHA-256 comparison failed on arduino 2.0.3

11wanqin, 2022-06-30:
### Board

ESP32-C3

### Device Description

[image]
[image]

cannot use esp32_c3 usb_serial to upload program,but can use idf to upload program.

### Hardware Configuration

connected  nothing.

### Version

v2.0.3

### IDE Name

arduino ide ,platformio

### Operating System

windows10

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

[image]
[image]

cannot use esp32_c3 usb_serial to upload program,but can use idf to upload program.

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("111");
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381782
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
Attempting to boot anyway...
entry 0x403ce000
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-06-30:
this is OK. the error will go away when we switch to esptool v4. It does not prevent you from running

closed 2022-06-30 by me-no-dev

lbernstone, 2022-06-30:
Use flash mode qio for esp32-c3

11wanqin, 2022-07-02:
> this is OK. the error will go away when we switch to esptool v4. It does not prevent you from running

I try to use esptool v4.2,but the soc cannot running too.
[image]
[image]
I have no idea now.

lbernstone, 2022-07-02:
Are you able to flash an example (https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/examples/SPIFFS_Test/SPIFFS_Test.ino)?  Do you have an ERROR? We cannot fix problems that we cannot see.

11wanqin, 2022-07-03:
> 

[image]
Error still exists，when i flash this example.

11wanqin, 2022-07-03:
This device use the usb port to download,.it didn't have a usb_ttl ic.