espressif/arduino-esp32 #6932: ESP32-C3 bootloop trying MQTT connection

ridoluc, 2022-07-02:
### Board

ESP32-C3

### Device Description

The board is Adafruit QT Py ESP32-C3 WiFi Dev Board.
Nothing attached to it. 
Powered through USB C. Tested with alternative power sources.

### Hardware Configuration

Nothing attached to the board.

### Version

latest master (checkout manually)

### IDE Name

PlatformIO, Arduino IDE

### Operating System

macOS

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

 The board reset itself after a few lines of execution and never executes the main loop for a second time. I'm just trying the connection to AWS server through MQTT. I tried to remove all unnecessary code and now it's just made of the connection setup and a loop that writes a message on the serial port every 10 seconds.

### Sketch

```cpp
#include <Arduino.h>
#include "WiFi.h"
#include <WiFiClientSecure.h>
#include <MQTTClient.h>
#include <ArduinoJson.h>
#include <esp_wifi.h>

// The MQTT topics that this device should publish/subscribe
#define AWS_IOT_PUBLISH_TOPIC   "esp32/pub"
#define AWS_IOT_SUBSCRIBE_TOPIC "esp32/sub"

typedef enum {
   LOG,
   DATA
} MsgType;

const char * const messagesTypes[] = {"log","data"};

class IoTConnection
{
private:
   /* data */
   WiFiClientSecure net;
   MQTTClient client;

public:
   IoTConnection(/* args */);
   
   bool startConnection();
   void closeConnection();
   
   bool publishMessage(MsgType msg, const char *data);
   bool publishLog(const char *logEntry);
   bool publishData(const char *data);

   void transmit(){client.loop();};
   bool connected(){return client.connected();};

   ~IoTConnection();
};

IoTConnection::IoTConnection(/* args */)
{
   net = WiFiClientSecure();
   client = MQTTClient();
}

bool IoTConnection::startConnection()
{

   WiFi.mode(WIFI_STA);
   esp_wifi_set_ps (WIFI_PS_NONE);
   WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

   Serial.println("Connecting to Wi-Fi");

   const int retryIntervalMs = 500;
   const int timeoutSeconds = 5*1000;
   int timeoutCounter = timeoutSeconds/retryIntervalMs;
   while (WiFi.status() != WL_CONNECTED && timeoutCounter > 0)
   {
      delay(retryIntervalMs);
      Serial.print(".");
      if (timeoutCounter <=0)
      {
         timeoutCounter = timeoutSeconds/retryIntervalMs;
         WiFi.reconnect();
      }
      timeoutCounter--;
   }
   Serial.println("\nwifi connected");

   // Configure WiFiClientSecure to use the AWS IoT device credentials
   net.setCACert(AWS_CERT_CA);
   net.setCertificate(AWS_CERT_CRT);
   net.setPrivateKey(AWS_CERT_PRIVATE);

   // Connect to the MQTT broker on the AWS endpoint we defined earlier
   client.begin(AWS_IOT_ENDPOINT, 8883, net);

   Serial.print("Connecting to AWS IOT\n");

   while (!client.connect(THINGNAME)) {
      Serial.print(".");
      delay(100);
   }

   if(!client.connected()){
      Serial.println("AWS IoT Timeout!");
      return false;
   }

   // Retain the session

   // client.setCleanSession(false);

   // Subscribe to a topic with Quality of Service 1 (QoS)
   client.subscribe(AWS_IOT_SUBSCRIBE_TOPIC);

   Serial.println("AWS IoT Connected!");

   return true;
}

void IoTConnection::closeConnection()
{
   client.disconnect();
   
   net.stop();

   WiFi.disconnect(true);

}

IoTConnection::~IoTConnection()
{
   closeConnection();
}

bool IoTConnection::publishMessage(MsgType msg, const char *data)
{
   StaticJsonDocument<200> doc;
   doc["MsgType"] = messagesTypes[msg];
   doc["data"] = data;
   char jsonBuffer[256];
   serializeJson(doc, jsonBuffer); // print to client

   return client.publish(AWS_IOT_PUBLISH_TOPIC, jsonBuffer);
}

bool IoTConnection::publishLog(const char *logEntry)
{
   return publishMessage(LOG,logEntry);
}

bool IoTConnection::publishData(const char *data)
{
   return publishMessage(DATA,data);

}

IoTConnection iot_conn = IoTConnection();

unsigned long sys_time;

void setup() {
  sys_time = millis();
  Serial.begin(11520);
  iot_conn.startConnection();
  if(iot_conn.connected()) Serial.print("Connection Established\n");
  iot_conn.publishLog("ESP32 Connected");

}

void loop() {
    if((millis() - sys_time )> 10000){
    // iot_conn.publishLog("hello");
    Serial.println("got here");
    sys_time = millis();
  }
  delay(100);
}
```

### Debug Message

```plain
Connecting to Wi-Fi
......
wifi connected
Connecting to AWS IOT
AWS IoT Connected!
Connection Established
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381846
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 81622325b5b3d82bf48506f210cd42cc4a45185255565a883ae891633570c8f5
Expected: bef5af017887fc9f8b85cdc8d157e714dd15305a39a5b8a4fc8312b760075a67
Attempting to boot anyway...
entry 0x403ce000
Connecting to Wi-Fi
......
wifi connected
Connecting to AWS IOT
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381846
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 81622325b5b3d82bf48506f210cd42cc4a45185255565a883ae891633570c8f5
Expected: bef5af017887fc9f8b85cdc8d157e714dd15305a39a5b8a4fc8312b760075a67
Attempting to boot anyway...
entry 0x403ce000
Connecting to Wi-Fi
..
```

### Other Steps to Reproduce

I tried
- different flash modes, frequencies
- different partition tables
- Arduino IDE

At most I can get worse results

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ridoluc, 2022-07-26:
I don't have access to the hardware right now and for the next month or so. Sorry.

any updates?

VojtechBartoska, 2022-08-23:
@ridoluc any news?

VojtechBartoska, 2023-01-18:
Hello, I'm closing this issue as expired. If needed, you can reopen it.

closed 2023-01-18 by VojtechBartoska