espressif/arduino-esp32 #6935: Missing build information for the binaries in this project

nomis, 2022-07-03:
### Board

N/A

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v2.0.3

### IDE Name

N/A

### Operating System

N/A

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

It looks like https://github.com/espressif/esp32-arduino-lib-builder is used to build the contents of `tools/sdk/` but there's no information on which commit in that repository was used.

There's a reference to a truncated commit `heads/idf-release/v4.4-0-gd3340` in the `esp_app_desc_t` but that doesn't appear to be published anywhere.

The build scripts in https://github.com/espressif/esp32-arduino-lib-builder always use the latest branch of dependencies. There's no information on which commits of these were used to build the 2.0.3 release.

This makes it very difficult to make one change to the `sdkconfig` and rebuild without making other changes (assuming the code is even still compatible across all of the dependencies).

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-07-03:
Every time me-no-dev commits a new version of IDF (eg https://github.com/espressif/arduino-esp32/pull/6594), he indicates which versions of the components were used.  AFAIK the daily CI stuff uses whatever is master in each of those sources.  For release versions, look for the last IDF commit in the release notes, which will have the list of the necessary components.

nomis, 2022-07-03:
It looks like `components/arduino_tinyusb/tinyusb` is the only component not in that list.

I still have to guess which version of [esp32-arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) was used.

me-no-dev, 2022-08-22:
Fixed here: https://github.com/espressif/esp32-arduino-lib-builder/commit/dbcc142bb4605a0fac984f2409374783f1948dd2

closed 2022-08-22 by me-no-dev