espressif/arduino-esp32 #6937: Unable to compile TWAI (CAN) with arduino IDE ESP32-C3

TurboMarian, 2022-07-03:
### Board

ESP32-C3

### Device Description

Adafruit QT Py ESP32-C3 WiFi Dev Board

### Hardware Configuration

NO

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHZ

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm unable to compile my sketch on that board (with any other board like S2 or S3 compiling just fine. 
I have error:

.........\Arduino\libraries\ESP32-Arduino-CAN-master\src\CAN.c:38:10: fatal error: soc/dport_reg.h: No such file or directory
 #include "soc/dport_reg.h"

with ESP32-C3 board selected

Checked with ESP32CAN and CAN librarys. On both I have same error.
 

### Sketch

```cpp
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <Adafruit_NeoPixel.h>

#include <ESP32CAN.h>
#define CANRX GPIO_NUM_3 // Define CAN RX pin
#define CANTX GPIO_NUM_4 // Define CAN TX pin
#define CANSPEED 1000E3  //define can speed

#define BUTTON  (1)
#define PIN       2
#define NUMPIXELS 3
```

### Debug Message

```plain
..........
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-04:
@TurboMarian - 
I think that ESP32CAN library was designed for ESP32 SoC and only works with it.
It is 5 years old. There was no S2/S3/C3 at that time. 
I don't believe that currently any ESP32 CAN Arduino Library has support for C3/S2/S3.

An alternative is the TWAI IDF library.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/twai.html
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/twai

VojtechBartoska, 2022-08-24:
@TurboMarian is this still valid?

VojtechBartoska, 2022-10-10:
I'm closing this issue as expired due to no answer. If needed, you can reopen it. Thanks for understanding.

closed 2022-10-10 by VojtechBartoska

Sheimy, 2022-10-24:
i try to use this one which is for s3 but esp always restart 
https://github.com/pawelka/esp32_can

Sheimy, 2022-10-25:
This is how to use TWAI with ESP32 S3
https://github.com/Sheimy/TWAI_S3_LISEN

gbkwiatt, 2022-12-18:
> i try to use this one which is for s3 but esp always restart https://github.com/pawelka/esp32_can

Have you managed to get it working ? I am in the same place now

Sheimy, 2022-12-18:
> > i try to use this one which is for s3 but esp always restart https://github.com/pawelka/esp32_can
> 
> Have you managed to get it working ? I am in the same place now

https://github.com/Sheimy/TWAI_S3_LISEN

TurboMarian, 2022-12-18:
Hi. No I didn't. I used mcp2515 to my project instead.

gbkwiatt, 2022-12-18:
> > > i try to use this one which is for s3 but esp always restart https://github.com/pawelka/esp32_can
> > 
> > 
> > Have you managed to get it working ? I am in the same place now
> 
> https://github.com/Sheimy/TWAI_S3_LISEN

This does indeed work. Is there an example with sending message too ?

Borkuz, 2023-04-01:
> > > > i try to use this one which is for s3 but esp always restart https://github.com/pawelka/esp32_can
> > > 
> > > 
> > > Have you managed to get it working ? I am in the same place now
> > 
> > 
> > https://github.com/Sheimy/TWAI_S3_LISEN
> 
> This does indeed work. Is there an example with sending message too ? Edit: Actually sending looks pretty straight forward.

Care to share some examples 😇

gbkwiatt, 2023-04-01:
> Care to share some examples 😇

Documentation is pretty good for it. but writing message looks like this 
```cpp
#include "driver/twai.h"

...

//Configure message to transmit
twai_message_t message;
message.identifier = 0xAAAA;
message.extd = 1;
message.data_length_code = 4;
for (int i = 0; i < 4; i++) {
    message.data[i] = 0;
}

//Queue message for transmission
if (twai_transmit(&message, pdMS_TO_TICKS(1000)) == ESP_OK) {
    printf("Message queued for transmission\n");
} else {
    printf("Failed to queue message for transmission\n");
}
```

this is from the docs 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html

zalexzperez, 2023-12-15:
@gbkwiatt, I'm currently working on a CAN Bus project.
I've successfully used an old ESP32 with Arduino CAN library from Sandeep Mistry.
With that library, when any CAN message is received, a callback function automatically interrupts program flow and we are able to determine which message was received with a switch case, like this:

```
void onReceive(int packetSize) {
  CAN.readBytes(CanB, 8); // CanB is the name of the Buffer

  switch (CAN.packetId()) {

    case ID1: 
    break;
    case ID2:
    break;
    default: 
    break;
  }
```
With IF statements we determine which CAN message was received and change flag variables accordingly, that we later use in the loop function. 

However, I'm now designing a PCB with an ESP32-S3 and realized the old library doesn't support the new ESP32 chips.

With the TWAI driver implementation, I see the program checks for new received messages only at the end of the void loop. This means that if the void loop code is long enough, some frames could be lost (?). 

So, could a callback function be implemented to replicate the old library behavior? Is it even necessary? 
Thanks for your time reading this.