espressif/arduino-esp32 #6946: delay() with small values (e.g. delay(1)) does not delay if system HZ (CONFIG_FREERTOS_HZ) <1000

egnor, 2022-07-05:
### Board

OLIMEX ESP32-POE (but will happen anywhere I think)

### Device Description

Stock OLIMEX ESP32-POE, but I believe this will happen on any board

### Hardware Configuration

Nothing attached

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

Ubuntu 22.04

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

The Arduino delay() function is implemented as follows:
https://github.com/espressif/arduino-esp32/blob/fcd4799c6de6eb5a5a8eba94818adf770238ecc0/cores/esp32/esp32-hal-misc.c#L178

If portTICK_PERIOD_MS > 1 (i.e. CONFIG_FREERTOS_HZ < 1000), then small values will round down to 0, and no delay will happen. I'm not super familiar with the delay facilities in FreeRTOS / ESP-IDF but there has to be some way to do a shorter delay (busy-wait, if nothing else)?

See also: https://github.com/platformio/platform-espressif32/issues/846

### Sketch

```cpp
// Run me with CONFIG_FREERTOS_HZ=100 to see the problem

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.printf("BEFORE msec=%lu\n", millis());
  for (int i = 0; i < 20; ++i) delay(5);
  Serial.printf("AFTER msec=%lu\n\n", millis());
}
```

### Debug Message

```plain
(none)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-07-05:
Arduino comes configured with 1000HZ. Why is this an issue? You can switch your project to 1000HZ as well and delay properly.

egnor, 2022-07-06:
When using [Arduino as an ESP32 component](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html), it becomes an issue, because CONFIG_FREERTOS_HZ is not necessarily 1000. In fact, it defaults to 100.

This caveat [is mentioned in the doc](https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html#freertos-tick-rate-hz), but it seemed preferable for the API to generally work? It might be better to fail at compile time by default if HZ<1000? Or somehow emit a warning somewhere?

lbernstone, 2022-07-06:
delayMicroseconds() is a blocking delay.  This is not a defect, if you ask for a delay(0), you get it...

egnor, 2022-07-06:
If there's a semantic difference between `delayMicroseconds()` and `delay()`, it would be nice to document that! There's certainly no such difference in the general Arduino interface expectations. They're both "blocking".

(I'd been hoping for a `delay(5)`, not a `delay(0)`...!)

egnor, 2022-07-06:
Erroring out (ideally at compile time!) if HZ != 1000 would certainly seem reasonable to me. That would have forced the bug I filed against platformio (example arduino-in-ESPIDF projects should set HZ to 1000) to get fixed.

Would you take that PR?

me-no-dev, 2022-07-07:
> Would you take that PR?

Definitely!

egnor, 2022-07-07:
https://github.com/espressif/arduino-esp32/pull/6955

VojtechBartoska, 2022-07-11:
Thanks for the PR @egnor!

egnor, 2022-07-15:
> Thanks for the PR @egnor!

Of course! It seems a bit stuck now, I have an open question in the PR on how to address the test failure? (I can sit patiently if needed, not sure how the review workflow works!)

closed 2025-01-16 by Parsaabasi