espressif/arduino-esp32 #6947: WiFi (specifically WifiGeneric) handling of events is subject to race conditions

egnor, 2022-07-06:
### Board

any board (with networking)

### Device Description

any device (with networking)

### Hardware Configuration

no special configuration

### Version

latest development Release Candidate (RC-X)

### IDE Name

any IDE

### Operating System

any OS

### Flash frequency

any frequency

### PSRAM enabled

yes

### Upload speed

any speed

### Description

It's hard to write a specific repro case for this since it depends on race conditions.

(Also, my understanding may be off, please correct!)

The `WiFi` instance allows user code to register for events with `WiFi.onEvent()`. This isn't documented, but it is present in most example code, described in various tutorials and copied into many/most network-using sketches.

This is implemented the `WiFiGeneric` class (one of the base classes for `WiFiClass`, of which `WiFi` is an instance) using an independently running FreeRTOS task (pinned to core 0 by default):
https://github.com/espressif/arduino-esp32/blob/fcd4799c6de6eb5a5a8eba94818adf770238ecc0/libraries/WiFi/src/WiFiGeneric.cpp#L299

That free-running task reads from a queue and invokes `WiFiGenericClass::_eventCallback()`:
https://github.com/espressif/arduino-esp32/blob/fcd4799c6de6eb5a5a8eba94818adf770238ecc0/libraries/WiFi/src/WiFiGeneric.cpp#L924

After translating the event, `_eventCallback()` loops through all registered callbacks and invokes them serially:
https://github.com/espressif/arduino-esp32/blob/fcd4799c6de6eb5a5a8eba94818adf770238ecc0/libraries/WiFi/src/WiFiGeneric.cpp#L1049

Nowhere in this process is locking or other synchronization done, that I can figure out. The implications seem like:

1. If code adds/removes callbacks at the same time as callbacks are being invoked, arbitrarily bad undefined behavior can result due to conflict over `cbEventList`. Most code will add callbacks at the start and leave them there forever, but if that's the expectation it seems worth documenting?
2. User callbacks are invoked on a separate task/thread (the "arduino event thread") from the user's main loop. That means the user should be careful about synchronization between those callbacks and the user's main code. However, this isn't documented, and example code (e.g. [WiFiClientEvents.ino](https://github.com/espressif/arduino-esp32/blob/371f382db7dd36c470bb2669b222adf0a497600d/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino), [WiFiUDPClient.ino](https://github.com/espressif/arduino-esp32/blob/371f382db7dd36c470bb2669b222adf0a497600d/libraries/WiFi/examples/WiFiUDPClient/WiFiUDPClient.ino), [ETH_LAN8720](https://github.com/espressif/arduino-esp32/blob/4b9dc6144706983f2d1fc2d810800b792939322e/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino), etc) make lots of unsynchronized calls and global variable changes in ways that seem unsafe.

(Hopefully you'll tell me why this is all wrong and secretly it's all serialized so the callbacks only happen between calls to `loop()` or something like that...?)

### Sketch

```cpp
(many of the example sketches, see above)
```

### Debug Message

```plain
(no specific debug logs)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2022-07-07:
> (Hopefully you'll tell me why this is all wrong and secretly it's all serialized so the callbacks only happen between calls to `loop()` or something like that...?)

Nope, there are for sure race condition bugs, both on ESP32 and ESP8266.
But it is really hard to pinpoint where they are.

mrengineer7777, 2023-04-13:
Your analysis of the library seems sound.
You are indeed correct that removing a callback while _eventCallback() is iterating through the loop could cause a crash, unless `cbEventList.size()` recalculates for every iteration.  Probably does recalculate - could be tested.
`_eventCallback` does indeed run on a separate thread.

The developers are quite busy working on the 3.0.0 release.

Please submit a PR to improve the documentation.  These are worth telling others about it.

egnor, 2023-04-16:
https://github.com/espressif/arduino-esp32/pull/8081 submitted.

closed 2025-01-16 by Parsaabasi

vortigont, 2025-01-16:
BTW, safety locking was added in #10376 
you need to define `NETWORK_EVENTS_MUTEX` build flag if you are concerned for thread safety.