espressif/arduino-esp32 #6954: undefined reference to `esp_ble_gap_start_advertising'

Kampi, 2022-07-07:
### Board

esp32-c3-devkitm-1

### Device Description

No

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I try to compile a project for the ESP32-C3 by using esp-idf 4.4.1 in PlatformIO and the Arduino Core as a component. The linking failes, because of this error:

```
c:/users/konta/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio\build\debug\esp-idf\arduino-esp32\libarduino-esp32.a(BLEAdvertising.o): in function `BLEAdvertising::start()':
C:\Users\konta\Desktop\firmware/components/arduino-esp32/libraries/BLE/src/BLEAdvertising.cpp:247: undefined reference to `esp_ble_gap_start_advertising'
c:/users/konta/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\konta\Desktop\firmware/components/arduino-esp32/libraries/BLE/src/BLEAdvertising.cpp:214: undefined reference to `esp_ble_gap_config_adv_data'
c:/users/konta/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:\Users\konta\Desktop\firmware/components/arduino-esp32/libraries/BLE/src/BLEAdvertising.cpp:225: undefined reference to `esp_ble_gap_config_adv_data'
```

Compiling for the ESP32 Dev-board works without an issue.fails

### Sketch

```cpp
Nothing
```

### Debug Message

```plain
Nothing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-08:
@Kampi

I tested the `BLE_server.ino` example that uses this same API and I compiled it using the Arduino IDE with no errors.

I think it may be a problem of PlatformIO configuration from your side. 
Please test your Project using the Arduino IDE if necessary.

If the problem persists, maybe you can open an issue within PlatformIO github.

Kampi, 2022-07-08:
`CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y` was not set in `sdkconfig`.

closed 2022-07-08 by Kampi

SuGlider, 2022-07-08:
@valeros - Is this something that should be configured in the PIO side?
Please let me know.

valeros, 2022-07-11:
Hi @SuGlider, everything is fine. It was a mixed IDF/Arduino project that required a special option in the sdkconfig.

ZhongWwwHhh, 2024-09-10:
I also face this problem today. It only needs to tick `CONFIG_BT_BLE_42_FEATURES_SUPPORTED=y` in menuconfig  

[image]

But it is not tick by default. And some document like this [https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf](https://github.com/nkolban/esp32-snippets/blob/master/Documentation/BLE%20C%2B%2B%20Guide.pdf) not mention this.  

Could we add this to this document? [https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html)