espressif/arduino-esp32 #6957: Syntax error in gen_esp32part.py of v2.0.4

cxkkxrr, 2022-07-08:
### Board

ESP32 Dev Module

### Device Description

No

### Hardware Configuration

No

### Version

latest master (checkout manually)

### IDE Name

Arduino Ide

### Operating System

macOS

### Flash frequency

4mb

### PSRAM enabled

no

### Upload speed

115200

### Description

gen_esp32part.py   SyntaxError: invalid syntax
[line 254]  `def verify_size_fits(self, flash_size_bytes: int) -> None:`

It's normal for me to modify it like this
` def verify_size_fits(self, flash_size_bytes):`

### Sketch

```cpp
Nothing
```

### Debug Message

```plain
File "/Users/xxx/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/tools/gen_esp32part.py", line 254
    def verify_size_fits(self, flash_size_bytes: int) -> None:
                                               ^
SyntaxError: invalid syntax
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-08:
@me-no-dev - It seems related to MacOS + update Core Version from GitHub.
Have you seen such issue in your machine?

cxkkxrr, 2022-07-12:
> @cxkkxrr Can you call `python --version` to check what version of python are you running?

@P-R-O-C-H-Y  Python 3.10.5

SuGlider, 2022-07-12:
>gen_esp32part.py   SyntaxError: invalid syntax
>[line 254]  `def verify_size_fits(self, flash_size_bytes: int) -> None:`
>
>It's normal for me to modify it like this
>` def verify_size_fits(self, flash_size_bytes):`

This should run in Python 3.5+
This feature is known as `Type Hints`, where it is possible to tell Python argument types and return type in the function declaration.
Not sure why your Python 3.10 is complaining about the syntax.

Using Windows 11 with Python 3.9.13 and Arduino Core 2.0.4 this script runs fine.

SuGlider, 2022-07-12:
@cxkkxrr 
Could you please post the complete output of the compilation using your MacOS and Arduino IDE with option to 
`Show verbose output during: ( )compilation ( )upload` set for `compilation` in the IDE Menu->File->Preferences?

cxkkxrr, 2022-07-13:
> @cxkkxrr Could you please post the complete output of the compilation using your MacOS and Arduino IDE with option to `Show verbose output during: ( )compilation ( )upload` set for `compilation` in the IDE Menu->File->Preferences?

Arduino:1.8.19 (Mac OS X), 开发板："ESP32C3 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 160MHz (WiFi), DIO, 80MHz, 4MB (32Mb), 921600, None"

/private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/hardware -hardware /Users/xxx/Library/Arduino15/packages -tools /private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/xxx/Library/Arduino15/packages -built-in-libraries /private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/libraries -libraries /Users/xxx/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32c3:CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=dio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10819 -build-path /var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/arduino_build_453998 -warnings=none -build-cache /var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/arduino_cache_565538 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.esptool_py.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/esptool_py/3.3.0 -prefs=runtime.tools.esptool_py-3.3.0.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/esptool_py/3.3.0 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mklittlefs.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.mkspiffs.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/untitled1620355145.tmp/sketch_jul13b/sketch_jul13b.ino
/private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/hardware -hardware /Users/xxx/Library/Arduino15/packages -tools /private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/xxx/Library/Arduino15/packages -built-in-libraries /private/var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/AppTranslocation/C4D639A7-F4FD-45AD-A629-5B6764D161FA/d/Arduino.app/Contents/Java/libraries -libraries /Users/xxx/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32c3:CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=dio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10819 -build-path /var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/arduino_build_453998 -warnings=none -build-cache /var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/arduino_cache_565538 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch3 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch3.path=/Users/xxx/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gc [... cut ...]
python /Users/xxx/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/tools/gen_esp32part.py -q /var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/arduino_build_453998/partitions.csv /var/folders/5g/lsvzl35j6c9c4xd3vtnth_zw0000gp/T/arduino_build_453998/sketch_jul13b.ino.partitions.bin
  File "/Users/xxx/Library/Arduino15/packages/esp32/hardware/esp32/2.0.4/tools/gen_esp32part.py", line 254
    def verify_size_fits(self, flash_size_bytes: int) -> None:
                                               ^
SyntaxError: invalid syntax
exit status 1
为开发板 ESP32C3 Dev Module 编译时出错。

cxkkxrr, 2022-07-13:
[image]

[image]

P-R-O-C-H-Y, 2022-07-13:
As @SuGlider mentioned, the syntax should work from Python 3.5+
As you can see, if I run the script from 2.7.16 I got the same error like you.

```
prochy@Jan-MacBook-Pro tools % python3 --version      
                        
Python 3.10.5

prochy@Jan-MacBook-Pro tools % python3 gen_esp32part.py partitions/default.bin

Parsing binary partition input...
Verifying table...
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,20K,
otadata,data,ota,0xe000,8K,
app0,app,ota_0,0x10000,1280K,
app1,app,ota_1,0x150000,1280K,
eeprom,data,153,0x290000,4K,
spiffs,data,spiffs,0x291000,1468K,

prochy@Jan-MacBook-Pro tools % python --version
                               
Python 2.7.16

prochy@Jan-MacBook-Pro tools % python gen_esp32part.py partitions/default.bin 

  File "gen_esp32part.py", line 254
    def verify_size_fits(self, flash_size_bytes: int) -> None:
                                               ^
SyntaxError: invalid syntax
```

Can you try to run the Python script directly from terminal? Maybe its Arduino IDE related.
To test it navigate to esp32/tools folder and run both to check if it works in both Python versions and post the results please.
`python gen_esp32part.py partitions/default.bin`
`python3 gen_esp32part.py partitions/default.bin`

P-R-O-C-H-Y, 2022-07-25:
@cxkkxrr Hello, did you take a look? Any news?

P-R-O-C-H-Y, 2022-08-01:
Closing the issue due to no answer. Feel free to reopen if is the issue still actual.

closed 2022-08-01 by P-R-O-C-H-Y