espressif/arduino-esp32 #6961: CORRUPT HEAP assert failed: multi_heap_free when connecting to BLE Server

savejeff, 2022-07-09:
### Board

ESP32-S3

### Device Description

DevKitC-1, soldered and Breadboard using multiple Sensors over I2C and SPI, running on both cors with Multiple Tasks

### Hardware Configuration

SARA R422M8N over Serial Port
SD Card over SPI
IMU over I2C
etc ...

### Version

v2.0.3

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

240Mhu

### PSRAM enabled

yes

### Upload speed

115200

### Description

Since using the ESP32 S3 with Both cores with the Core version 2.0.0 i get heap corrupted errors on BLE connect

The Code runs fine until i connect through my smartphone to the BLE Server running on the ESP32-S3 

this is what I'm getting
[image]

[image]

[image]

### Sketch

```cpp
.
```

### Debug Message

```plain
CORRUPT HEAP: Bad tail at 0x3fcb728c. Expected 0xbaad5678 got 0xbaad5600

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-10:
@savejeff - 
Couldn you please provide the smallest possible Arduino sketch that reproduces this issue?
That's the best way to help the community to help you.

This error maybe due to low Stack size of your Tasks. 
It is hard to tell at this moment because there is not enough information.

savejeff, 2022-07-11:
I know that it is hard to analyse this Problem. 
Currently, the Crash only occurs very infrequently and only when I'm in the field where the Mobile reception is bad (I'm using Cellular Connection over Serial). 
The Project is pretty big and it only crashes when most of the code is included. 
Thus I don't see a way to extract a minimum sketch that reproduces the problem.

I have checked the total heap usage (around 50%)
I have given all tasks a sufficient amount of memory I think. Is there a way to check the stack usage on each task? 

might this be a stack size problem in the Bluetooth task? Is there a way to check this?

Is there a way to dump the state of all takes at the time of crash?

savejeff, 2022-07-12:
Hi SuGlider,

Thx for the links to analyse the stack. 

about the first link: i already tried vTaskList but i get a compilation error even when setting configUSE_TRACE_FACILITY=1 and configUSE_STATS_FORMATTING_FUNCTIONS =1. i get `undefined reference to `vTaskList'` 

i have set both flags via buildflags. Might it be, that vTaskList is not included in the precompiled part ?

Ouss4, 2022-07-13:
To answer some of your questions:
 - Multi-heap is just a name used in IDF for the allocator since it manages multiple heap regions.
 - Heap poisoning is used to insert some fixed "canary" data before and after each allocated block, these data can be checked to make sure that the blocks were not overwritten.  You can read more here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#configuration

When using BLE, there is a memory region that's reserved for its operation, this region should not be part of the heap, otherwise the heap will be corrupted when the BLE uses its memory.
For now I can't see anything that could cause a corruption but looking at the errors you have, the canary values are supposed to be `0xbaad5678` and `0xABBA1234` but are `0xbaad5600` and `0x00001234`.  

Also in one of your screens there seems to be an allocation at `0x3fffd2bc` this looks incorrect as this address space is reserved.

savejeff, 2022-07-14:
Thx @Ouss4 

Thanks for the clarification and insight. 
My current guess is like earlier suggested. One of the Tasks runs out of Stack. Ill try to monitor the stack using the functions provided by SuGlider. 
i also found the code runs more stable if i do more delays/yielding on the cellular task. That task might have taken up to much computational time or something like that. 

What i can rule out with high certainty is that is something like writing out of bound on arrays in my code. i also use malloc very carefully.

VojtechBartoska, 2022-08-24:
@savejeff do you still need help?

closed 2022-08-24 by savejeff