espressif/arduino-esp32 #6965: erase_flash option in arduino ide

TheSeriousProgrammer, 2022-07-11:
### Related area

Clearing flash in esp32

### Hardware specification

ESP32-cam

### Is your feature request related to a problem?

My esp32 cam was never connecting to the internet, unless I cleared the flash to erase_flash option in esptool.py . However I had to search for the code system wide, install the dependencies and then run it

### Describe the solution you'd like

Since the esp_tool.py works out of the box in arduino IDE for flashing purposes , it would be good to see an option to erase flash directly integrated in the IDE

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-07-11:
Select Tools->Programmer->esptool
Burn bootloader.
Voila

TheSeriousProgrammer, 2022-07-11:
I am using a ESP-32 cam, to program the same I use a FTDI Chip because of the same it displays that No Programmers are available for the board. How can I fix the same ?

closed 2025-01-16 by Parsaabasi