espressif/arduino-esp32 #6966: TwoWire(1) broken in 2.0.1 and later?

JimDrewGH, 2022-07-11:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

EPS32 WROOM module with custom board.

### Hardware Configuration

There are three PCAL6416APW ICs and a ADC1015 IC.  A total of four I2C addresses, two on each I2C bus.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10 PRO

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

962100

### Description

Using any version prior to v2.0.1 of the ESP32 framework I can successfully communicate via dual I2C buses.  Each bus has 2 devices on it.

With v2.0.1 and later it seems that TwoWire(1) no longer works.  The exact same code works perfectly with v2.0.0 or earlier without any changes.  You can switch versions back and forth and see success and failure following the changes.  v1.0.6/v2.0.0 work correctly, and v2.0.1/v2.0.2/v2.0.3/v2.0.4 do not work correctly.

Below you will find the simple I2C scanner code taken from the examples pull-down menu under 'Wire'.  I just added a few lines of code to setup the pins and add the TwoWire functions.  With v2.0.1 and later and attempting to use TwoWire(1), the response is Error 5 (timeout).  This means that the I2C bus is doing "something", and it takes quite some time to go through each of the possible addresses.  Using TwoWire(0) always works with any version.  The pins do not matter.  I can use either pair with the same results.

How I found this problem is that I normally use both TwoWires, ie:

TwoWire I2C1 = TwoWire(0);
TwoWire I2C2 = TwoWire(1);

Along with:

  I2C1.begin(OurSDA1, OurSCL1, 400000);
  I2C2.begin(OurSDA2, OurSCL2, 400000);

I noticed that v2.0.1 and later has a fit about the frequency and needs that forced as a uint32_t.

This setup for I2C1 and I2C2 gives me both buses working perfectly, that is until I just upgraded to v2.0.4.  I went back and tried every version up to v1.0.5 and noticed that v2.0.1 is where it became broken.  I know there was a change with the I2C and I even tried the TwoWire setup as TwoWire I2C1(0) and TwoWire I2C1(1), which does compile and work,, except TwoWire I2C1(1) does not work (same results as the other method of setting up I2C1).

Any ideas?  Thanks!

### Sketch

```cpp
#include "Wire.h"

#define OurSDA1 21                            // (B) GPIO21 - SDA1 line
#define OurSCL1 22                            // (O) GPIO22 - SCL1 line
#define OurSDA2 26                            // (B) GPIO26 - SDA2 line
#define OurSCL2 27                            // (O) GPIO27 - SCL2 line
const uint32_t Freq = 400000;

//TwoWire I2C1 = TwoWire(0);                    // this works!
TwoWire I2C1 = TwoWire(1);                    // this does not work!

void setup()
{
  Serial.begin(115200);
  delay(10);

  I2C1.begin(OurSDA1, OurSCL1, Freq);         // enable I2C1 for 1st I2C bus
//  I2C1.begin(OurSDA2, OurSCL2, Freq);         // enable I2C1 for 2nd I2C bus
}

void loop() {
  byte error, address;
  int nDevices = 0;

  delay(1000);

  Serial.println("Scanning for I2C devices ...");
  for(address = 0x01; address < 0x7f; address++){
    I2C1.beginTransmission(address);
    error = I2C1.endTransmission();
    if (error == 0){
      Serial.printf("I2C device found at address 0x%02X\n", address);
      nDevices++;
    } else if(error != 2){
      Serial.printf("Error %d at address 0x%02X\n", error, address);
    }
  }
  if (nDevices == 0){
    Serial.println("No I2C devices found");
  }
}
```

### Debug Message

```plain
This is what you get with TwoWire(1) being used with v2.0.1 or later:

Error 5 at address 0x01
Error 5 at address 0x02
Error 5 at address 0x03
Error 5 at address 0x04
Error 5 at address 0x05
Error 5 at address 0x06
Error 5 at address 0x07
Error 5 at address 0x08

This is what I get
```

### Other Steps to Reproduce

Just compile the above sketch with v2.0.1 or later and comment/uncomment the two line setting the TwoWire number:

```
//TwoWire I2C1 = TwoWire(0);        // this one works with all versions
TwoWire I2C1 = TwoWire(1);          // this one doesn't work with v2.0.1 or later
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-07-11:
The [overloads](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L79-L80) have changed a bit in v2.0.1 with the introduction of I2C Slave.  If you ensure that your values are unambiguous by specifically casting them ((int)sda, (int)scl, (uint32_t)frequency) or declaring them explicitly, then it will work properly.

JimDrewGH, 2022-07-12:
So, after much experimenting I did find a simple work around for this problem for my particular setup, but I am not confident that this is going to work with various libraries that rely on the TwoWire configuration (there are a slew of examples based on this setup) - and it use to work, so it probably should still work.

In my particular case I just commented out these lines as shown:

```
//TwoWire I2C1 = TwoWire(0);
//TwoWire I2C2 = TwoWire(1);
```

and substituted them with these lines:

```
#define I2C1 Wire
#define I2C2 Wire1
```

This is simple enough, but again I am not sure if this will universally work with all libraries.

JimDrewGH, 2022-07-15:
You can not just substitute Wire and Wire1 for all occurrences of I2C1 and I2C2 because it won't compile.

Also, I can confirm that there are several libraries (adafruit types) that expect to have the TwoWire working and won't work by using the simple #define like I can get away with in my code because I am not using a library.

I look forward to seeing what you find.  Please let me know if I can help test something.  I have a lot of apps that use the TwoWire function.

P-R-O-C-H-Y, 2022-07-15:
@JimDrewGH Can you try just renaming the `I2C1` and `I2C2` to 
```
TwoWire WireI2C0 = TwoWire(0);       
TwoWire WireI2C1 = TwoWire(1);
```
For me both are working now. There has to be some conflict when using `I2C0` and `I2C1`.

closed 2022-07-15 by P-R-O-C-H-Y

JimDrewGH, 2022-07-17:
Interesting!  You will find that many examples in the wild will no longer work then because they use the same I2C1 and I2C2 names (which is how I got the idea to use these names).  I will change the names and try again.  :)

Thanks!