espressif/arduino-esp32 #6967: Wire.h buffer able to change buffer from .ino file

dharmik768, 2022-07-11:
### Board

ESP32 DEV Module

### Device Description

Using generic ESP32 dev board

### Hardware Configuration

non

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I am using a [vl53l3cx ](https://www.st.com/resource/en/datasheet/vl53l3cx.pdf)TOF sensor with an ESP32 board. 

I am using this [library and its example](https://github.com/pololu/vl53l1x-st-api-arduino) for testing and it works perfectly but I need to change inside the ESP32 package to make it work.  To be specific inside the Wire.h file of the ESP32 package. 

Initially, the example was not working directly but I read somewhere in the form that it needs to increase the I2C buffer size as the received packet size from the sensor is big therefore default size will not work.

> #define I2C_BUFFER_LENGTH 128

First I tested it by changing the I2C_BUFFER_LENGTH  to 256 directly in the Wire.h file and the example code is working perfectly.  This test was done in the old ESP32 package release V1.0.6

But In the latest releases that I can see after V2.0.3 in Wire.h file it's defined like this

> #ifndef I2C_BUFFER_LENGTH
>     #define I2C_BUFFER_LENGTH 128
> #endif

So I thought I can increase the size of the buffer externally directly from the Arduino .ino file itself like this before including the Wire.h

> #define I2C_BUFFER_LENGTH 256
> #include<Wire.h> 

But on uploading in this way the sensor data is not read properly like previously, So I checked by printing the I2C_BUFFER_LENGTH  in Arduino code as well as inside the Wire.begin(); library function.  
And shockingly it printed 256 in the Arduino code and when begin to function is called it printed 128. 
So it seems like it not making any change to the I2C_BUFFER_LENGTH  even it's defined before calling the library. 

Is there any way to change the I2C_BUFFER_LENGTH  without making any modifications to the ESP32 package itself? 

As I think the Arduino compile is somehow compiling all ESP32 packagess before even we include them. As I have done the same test with the blank sketch with only Wire.h library included and I2C_BUFFER_LENGTH  same results were seen.

Thank you,
Dharmik Patel

### Sketch

```cpp
#define I2C_BUFFER_LENGTH 256
#include<Wire.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  delay(2000);
  Serial.println(I2C_BUFFER_LENGTH);  // here it will print 256 that what we need 
  Wire.begin();  // but in actually its not changed here it will print 128
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
None
```

### Other Steps to Reproduce

To debug properly add this line inside your Wire.cpp file function 
> bool TwoWire::begin(int sdaPin, int sclPin, uint32_t frequency)

add this inside that 
> Serial.println(I2C_BUFFER_LENGTH);

Finally, it will look like this

> // Master Begin
> bool TwoWire::begin(int sdaPin, int sclPin, uint32_t frequency)
> {
> 	Serial.println(I2C_BUFFER_LENGTH);

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-13:
@dharmik768 
I have tested it with Arduino Core 2.0.3 and the output is 256 / 128.
But when I test it with Arduino Core 1.0.6 the output is 128 / 128.

If it works for 1.0.6, the buffer size used in Arduino is always 128. 

There is no way to change it to any other size just using a simple `#define I2C_BUFFER_LENGTH size`
This `#define` only works inside the Sketch. 
The library is compiled separately in another step and doesn't use any Arduino Sketch `#define` value.

Which makes me think that the size of the buffer may not be the problem for the [vl53l3cx ](https://www.st.com/resource/en/datasheet/vl53l3cx.pdf)TOF sensor

IF you want to test with a different Buffer Size, please try it by changing the value directly in Wire.h file.
at https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L38

dharmik768, 2022-07-14:
I have already tested successfully by changing the buffer length directly in the Wire.h and its works perfectly.

I thought it would be more simple for the end user if they do not need to go and change buffer sizes in their ESP32 packages. Anyway, if there is no way around the possible solution I was going to try out to make a copy of the Wire.h file and change it as per my need and add it locally with my custom library package such that the code used my Wire.h instead of the default one.

SuGlider, 2022-07-16:
@dharmik768 - Interesting... I just found this saying that it is necessary to change the buffer size directly in the Wire header file:
https://support.arduino.cc/hc/en-us/articles/4406686928786-Modify-the-buffer-size-of-the-Wire-library

dharmik768, 2022-07-18:
Yes it is required in some cases as I mentioned the TOF sensor without its not working may be a TOF library side issue but the majority of the library part API is suggested by ST itself. 

So it would be very handy if we can directly manage the buffer size in the code itself when needed without modifying the package files.

dharmik768, 2022-07-19:
> I have already tested successfully by changing the buffer length directly in the Wire.h and its works perfectly.
>
> I thought it would be more simple for the end user if they do not need to go and [...]

dharmik768, 2022-07-19:
Closed by mistake,
Today I tried to add a copy of Wire.h library with modified buffer size to my custom library, but the Arduino IDE somehow still first compiles the package Wire.h file, and then it shows a compilation error as it founds multiple functions so this way is also tricky to accomplish.

SuGlider, 2022-07-19:
I'm currently working on a `Wire::setBufferSize()`.

I will submit soon a PR.

dharmik768, 2022-07-19:
Let me know if you need any helping hand in that, I can test it out.

SuGlider, 2022-07-20:
@dharmik768 - Please test #7016 and let me know. Thanks!

closed 2022-07-28 by SuGlider

SuGlider, 2022-07-28:
PR https://github.com/espressif/arduino-esp32/pull/7016 merged.
@dharmik768  - you can try it using Arduino Core cloned from this repository.

dharmik768, 2022-08-07:
Hello @SuGlider,

Sorry for the late response today I got time to test the new modified wire library these are the steps I follow.
I extracted the new modified wire library from the master branch and added that to my ESP32 library package path and removed the old one. 

Now I uploaded the test code for the sensor that I have without changing the buffer size, and a strange thing happened without changing the buffer also the code is working perfectly and I am able to read sensor data. 

So apart from the new buffer size change function in the library have you changed anything in the older wire library same code is not working without changing the buffer size. 

Anyways thank you for adding that function.

dharmik768, 2022-08-07:
Do you think this is happening because in my library.h file I have added this line
#define I2C_BUFFER_LENGTH 256 

Will this make changes in the buffer directly without using that function as previously it was not changing, let me comment it out and check it once.

dharmik768, 2022-08-07:
I commented and tested it is still working.

SuGlider, 2022-08-08:
> Do you think this is happening because in my library.h file I have added this line #define I2C_BUFFER_LENGTH 256
> 
> Will this make changes in the buffer directly without using that function as previously it was not changing, let me comment it out and check it once.

Yes, this is used as default size when none is set using `setBufferSize()` when `begin()` is executed.
Default size is 128 https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h#L43
It is used here https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.cpp#L41

Make sure you don't have more than one Wire Library in use...

peteDDD, 2024-08-28:
As of today... August 28, 2024,
`
Wire.setBufferSize(255);
`
Does not change the buffer size.  

Whereas editing the Wire.h as follows does work.
```
#ifndef I2C_BUFFER_LENGTH
    #define I2C_BUFFER_LENGTH 255//128  // Default size, if none is set using Wire::setBuffersize(size_t)
#endif
```
Here is test code... main.cpp
`
#include <Arduino.h>
#include <Wire.h>

#define mySDA 6
#define mySCL 5
#define I2C_FREQUENCY 400000

void setup(void)
{
  Serial.begin(115200);
  Wire.begin(mySDA, mySCL, I2C_FREQUENCY);
  Wire.setBufferSize(255);
}

void loop(void)
{}

`

me-no-dev, 2024-08-29:
@peteDDD call `Wire.setBufferSize(255);` BEFORE `Wire.begin(mySDA, mySCL, I2C_FREQUENCY);`

PR #7016 (2022-07-28): Adds Wire::setBufferSize function
## Description of Change
Adds a new Wire (I2C) functionality to set the RX/TX buffer size and change it any time in the sketch.
Some I2C devices and sensors may require a bigger buffer for its transactions.

## Tests scenarios

``` cpp
#include "Wire.h"

void setup() {
  Wire::setBufferSize(512);
  Wire::begin();  // as I2C Master
}

void loop() {
}

```

## Related links
Fixes #6967 
Fixes #7042