espressif/arduino-esp32 #6972: Can't GPIO34 use digitalPinToInterrupt(esp32-wroom)?

hzhh110, 2022-07-12:
### Related area

esp32-wroom

### Hardware specification

esp32-wroom

### Is your feature request related to a problem?

Can't GPIO34 use digitalPinToInterrupt(esp32-wroom)?

### Describe the solution you'd like

Can't GPIO34 use digitalPinToInterrupt(esp32-wroom)?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-07-12:
Try with this example.
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/GPIO/GPIOInterrupt/GPIOInterrupt.ino

VojtechBartoska, 2022-07-26:
@P-R-O-C-H-Y Can you please help with answering this question? Thank you.

P-R-O-C-H-Y, 2022-11-15:
@hzhh110 Do you still have issues? The example **lbernstone** sent should help you with that. I am closing this as there is no issue. If you still need any help, feel free to reopen this.