espressif/arduino-esp32 #6976: ESP32C3 can not wake from Light-sleep by UART

soulyiran, 2022-07-12:
### Board

ESP32-C3-4M

### Device Description

Board: ESP32-C3-4M on custom board
IDE: Arduino
Version: Arduino ESP32 Version 2.0.4
Computer OS: Mac M1

### Hardware Configuration

only uart

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 12.4

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

ESP32C3 can not wake from Light-sleep by UART.

### Sketch

```cpp
#include <Arduino.h>
#include <driver/uart.h>

void setup() {
    // wake from sleep every 20s
    esp_sleep_enable_timer_wakeup(20*1000000); 

    // Configure UART0
    Serial.begin(115200);
    while (!Serial) { delay(500); }
    uart_set_wakeup_threshold(UART_NUM_0, 3);   // 3 edges on U0RXD to wakeup
    esp_sleep_enable_uart_wakeup(UART_NUM_0);   // Enable UART 0 as wakeup source

    Serial.printf("ESP32 light sleep with UART wake\r\n");
    Serial.printf("SDK: %s\r\n", ESP.getSdkVersion());
    Serial.flush();

}

void loop() {
    Serial.println("going to sleep");
    Serial.flush();
    esp_light_sleep_start();
    esp_sleep_wakeup_cause_t wakeup_cause;
    wakeup_cause = esp_sleep_get_wakeup_cause();
    
    if (wakeup_cause == 8){
      Serial.println("wakeup by uart");
      delay(5000);
      while(Serial.available()) Serial.write(Serial.read());
      Serial.flush();
    }
    if (wakeup_cause == 4){
      Serial.println("wakeup by timmer");
      Serial.flush();
    }
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
ESP32 light sleep with UART wake
SDK: v4.4.1-472-gc9140caf8c
going to sleep
wakeup by timmer
...
```

### Other Steps to Reproduce

I also try this code on ESP32, and it can be successfully wake by UART0.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-13:
Issue confirmed. The sketch works fine with ESP32, ESP32-S2 and ESP32-S3, but the ESP32-C3.

SuGlider, 2022-07-13:
It is not actually an Arduino issue, but instead it may be an IDF issue.
@igrr - Do you have any information about it?
I remember about a similar issue with ESP32 (https://github.com/espressif/arduino-esp32/issues/5107) related to IO_MUX.
Not sure if this is a similar problem.

SuGlider, 2022-07-13:
I tested it with all Arduino Core versions and none of them works.
- SDK: v4.4-dev-2313-gc69f0ec32 - **Core 2.0.0 - FAIL**
- SDK: v4.4-dev-3569-g6a7d83af19-dirty - **Core 2.0.1 - FAIL**
- SDK: v4.4-beta1-189-ga79dc75f0a - **Core 2.0.2 - FAIL**
- SDK: v4.4.1-1-gb8050b365e - **Core 2.0.3 - FAIL**
- SDK: v4.4.1-472-gc9140caf8c - **Core 2.0.4 - FAIL**

SuGlider, 2022-07-14:
@soulyiran 

#### SOLUTION: add `gpio_sleep_set_direction(GPIO_NUM_20, GPIO_MODE_INPUT);`  to the code:
``` cpp
void setup() {
    // wake from sleep every 20s
    esp_sleep_enable_timer_wakeup(20*1000000); 

    // Configure UART0
    Serial.begin(115200);
    while (!Serial) { delay(500); }

    // GPIO 20 is the default UART_0 RX pin
    gpio_sleep_set_direction(GPIO_NUM_20, GPIO_MODE_INPUT);
    gpio_sleep_set_pull_mode(GPIO_NUM_20, GPIO_PULLUP_ONLY);

    uart_set_wakeup_threshold(UART_NUM_0, 3);   // 3 edges on U0RXD to wakeup
    esp_sleep_enable_uart_wakeup(UART_NUM_0);   // Enable UART 0 as wakeup source

    Serial.printf("ESP32 light sleep with UART wake\r\n");
    Serial.printf("SDK: %s\r\n", ESP.getSdkVersion());
    Serial.flush();

}
```

Please confirm that it solves the issue. Thanks.

soulyiran, 2022-07-15:
> @soulyiran
> 
> #### SOLUTION: add `gpio_sleep_set_direction(GPIO_NUM_20, GPIO_MODE_INPUT);` to the code:
> ```c++
> void setup() {
>     // wake from sleep every 20s
>     esp_sleep_enable_timer_wakeup(20*1000000); 
> 
>     // Configure UART0
>     Serial.begin(115200);
>     while (!Serial) { delay(500); }
> 
>     // GPIO 20 is the default UART_0 RX pin
>     gpio_sleep_set_direction(GPIO_NUM_20, GPIO_MODE_INPUT);
>     gpio_sleep_set_pull_mode(GPIO_NUM_20, GPIO_PULLUP_ONLY);
> 
>     uart_set_wakeup_threshold(UART_NUM_0, 3);   // 3 edges on U0RXD to wakeup
>     esp_sleep_enable_uart_wakeup(UART_NUM_0);   // Enable UART 0 as wakeup source
> 
>     Serial.printf("ESP32 light sleep with UART wake\r\n");
>     Serial.printf("SDK: %s\r\n", ESP.getSdkVersion());
>     Serial.flush();
> 
> }
> ```
> 
> Please confirm that it solves the issue. Thanks.

this works for me too, thank you very much!

closed 2022-07-15 by soulyiran