espressif/arduino-esp32 #6981: Sketch using WiFi.softAPConfig() with 3 parameters no longer works under 2.0.4

avillacis, 2022-07-12:
### Board

ESP32, ESP32-S2

### Device Description

Any board using WiFi

### Hardware Configuration

No need for additional hardware

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Linux, Fedora 36 x86_64

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Beginning with Arduino-ESP32 2.0.4, depending on which headers are included and in which order, the declaration of WiFi.softAPConfig() might get an incorrect default parameter for the lease start, which will break sketches that previously worked under 2.0.3.

In 2.0.4, the WiFi class has added a new parameter to the declaration of softAPConfig, like this:
```diff
diff -ur esp32-alternatives/2.0.3/hardware/esp32/2.0.3/libraries/WiFi/src/WiFiAP.h esp32-alternatives/2.0.4/hardware/esp32/2.0.4/libraries/WiFi/src/WiFiAP.h
--- esp32-alternatives/2.0.3/hardware/esp32/2.0.3/libraries/WiFi/src/WiFiAP.h   2022-05-04 07:11:09.000000000 -0500
+++ esp32-alternatives/2.0.4/hardware/esp32/2.0.4/libraries/WiFi/src/WiFiAP.h   2022-07-06 04:47:49.000000000 -0500
@@ -38,7 +38,7 @@
 public:
 
     bool softAP(const char* ssid, const char* passphrase = NULL, int channel = 1, int ssid_hidden = 0, int max_connection = 4, bool ftm_responder = false);
-    bool softAPConfig(IPAddress local_ip, IPAddress gateway, IPAddress subnet);
+    bool softAPConfig(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dhcp_lease_start = INADDR_NONE);
     bool softAPdisconnect(bool wifioff = false);
 
     uint8_t softAPgetStationNum();
```

The problem is that, as discussed in https://github.com/espressif/arduino-esp32/issues/6760, there are conflicting definitions for INADDR_NONE. The LwIP libraries define INADDR_NONE as (uint32_t)0xFFFFFFFF , and the ESP32 Arduino WiFi library defines the same symbol as IPAddress(0,0,0,0), that is, all zeroes. The WiFi libraries treat the zero parameter as "use default lease range". However, if the LwIP libraries are included first, the 3-parameter call (with default value for dhcp_lease_start) will pick up instead 0xFFFFFFFF as  the lease start parameter, since a conversion from uint32_t to IPAddress happens to be defined and valid. This will break the lease range and therefore the configuration of scripts that previously worked.

My suggestion to fix this is to remove the default parameter of INADDR_NONE, and instead provide overloads for both 3-parameter and 4-parameter calls to WiFi.softAPConfig(), which should be internally routed as required. Alternatively, use a literal zero instead of INADDR_NONE as a default parameter so there cannot be any confusion on what is intended.

Please do not mark this report as a dupe or a wontfix. This is a real bug that can be easily fixed by not using INADDR_NONE as a default parameter in the class declaration.

### Sketch

```cpp
#include <Arduino.h>
/* Projects might not have full control of which libraries are included, or in which order */
extern "C" {
#include "lwip/err.h"
#include "lwip/sockets.h"
}
#include <WiFi.h>

void wifi_event_cb(WiFiEvent_t event, WiFiEventInfo_t)
{
    log_i("[WiFi-event] event: %d", event);

    switch (event) {
    case ARDUINO_EVENT_WIFI_AP_START:
        {
            IPAddress apIp(192, 168, 4, 1);
            IPAddress apNetmask(255, 255, 255, 0);

            if (!WiFi.softAPConfig(apIp, apIp, apNetmask)) {
                log_e("Failed to assign SoftAP interface");
            } else {
                log_i("softAP interface started correctly");
            }
        }
        break;
    }
}

void setup()
{
    Serial.begin(115200);

    WiFi.onEvent(wifi_event_cb);

    WiFi.mode(WIFI_AP_STA);
    WiFi.softAP("ESP32-TEST", NULL, 1, 0);
    
}

void loop()
{
    delay(1000);
}
```

### Debug Message

```plain
16:25:48.876 -> ets Jun  8 2016 00:22:57
16:25:48.876 -> 
16:25:48.876 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:25:48.876 -> configsip: 0, SPIWP:0xee
16:25:48.876 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:25:48.876 -> mode:DIO, clock div:1
16:25:48.876 -> load:0x3fff0030,len:1344
16:25:48.876 -> load:0x40078000,len:13864
16:25:48.876 -> load:0x40080400,len:3608
16:25:48.876 -> entry 0x400805f0
16:25:49.108 -> E (173) psram: PSRAM ID read error: 0xffffffff
16:25:49.141 -> [     5][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
16:25:49.208 -> [    86][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
16:25:49.208 -> [    87][I][test-softAP.ino:10] wifi_event_cb(): [WiFi-event] event: 0
16:25:49.307 -> [   180][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
16:25:49.307 -> [   185][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
16:25:49.307 -> [   185][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
16:25:49.307 -> [   186][I][test-softAP.ino:10] wifi_event_cb(): [WiFi-event] event: 2
16:25:49.307 -> [   193][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
16:25:49.341 -> [   198][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
16:25:49.341 -> [   204][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
16:25:49.341 -> [   211][I][test-softAP.ino:10] wifi_event_cb(): [WiFi-event] event: 10
16:25:49.341 -> [   223][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
16:25:49.374 -> [   235][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 255.255.255.255 | Netmask: 255.255.255.0
16:25:49.374 -> [   248][E][WiFiGeneric.cpp:163] set_esp_interface_ip(): The AP IP address (192.168.4.1) and the DHCP start address (255.255.255.255) must be in the same subnet
16:25:49.374 -> [   263][E][test-softAP.ino:19] wifi_event_cb(): Failed to assign SoftAP interface
16:25:49.407 -> [   270][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
16:25:49.407 -> [   277][I][test-softAP.ino:10] wifi_event_cb(): [WiFi-event] event: 11
16:25:49.407 -> [   283][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
16:25:49.407 -> [   290][I][test-softAP.ino:10] wifi_event_cb(): [WiFi-event] event: 10
16:25:49.407 -> [   296][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 192.168.4.1, MASK: 255.255.255.0, GW: 192.168.4.1
16:25:49.440 -> [   308][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 192.168.4.1 | Gateway: 192.168.4.1 | DHCP Start: 255.255.255.255 | Netmask: 255.255.255.0
16:25:49.440 -> [   322][E][WiFiGeneric.cpp:163] set_esp_interface_ip(): The AP IP address (192.168.4.1) and the DHCP start address (255.255.255.255) must be in the same subnet
16:25:49.473 -> [   336][E][test-softAP.ino:19] wifi_event_cb(): Failed to assign SoftAP interface
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-07-29:
@avillacis - I see the issue is about the order of includes.

So, my question would be: Why do you need such includes in your sketch?
``` cpp
extern "C" {
#include "lwip/err.h"
#include "lwip/sockets.h"
}
```

Another way to solve the issue is changing the siganture of `softAPConfig` to just this, which is the same:
`bool softAPConfig(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dhcp_lease_start = (unint32_t) 0);`

tsctrl, 2022-07-31:
This causing issue to load/call webserver for able to connect device and unable to assign ip to connect to ap for some new device in 2.0.4 @SuGlider. what i mean this issue above is critical and causing failure on the webserver and ap. suggesting this fix included in 2.0.4 as well.

PR #7059 (2022-08-01): Fixes LwIP INADDR_NONE conflict with softAPConfig()
## Description of Change
The header for softAPConfig() uses INADDR_NONE symbol that may conflict with LWIP defenition for the same symbol depending on the order to files included in a sketch, as describer in #6981 

## Tests scenarios
ESP32

## Related links
Fixes #6981
closed 2022-08-01 by SuGlider