espressif/arduino-esp32 #6982: Difficulty getting ESP32-S2 Timer interrupt to run. (IDFGH-4773)

sryan5, 2022-07-13:


VojtechBartoska, 2022-07-14:
Can you please help with triage @P-R-O-C-H-Y?

P-R-O-C-H-Y, 2022-08-03:
Hi @Dazza0, take a look on the [RepeatTimer example](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Timer/RepeatTimer/RepeatTimer.ino). There should be no problem with the timer interrupts. If you need any help, let me know.

VojtechBartoska, 2022-08-23:
@Dazza0 did you solve your problem?

Dazza0, 2022-08-23:
> I've been trying to get the timer interrupt for the esp32 s2 working with the repeated timer example but it seems like that the esp32S2 is not going into the interrupts. This code uses native arduino library functionsn for the HW_timer and compiles just fine. However there is no hardware response. 

Running your code snippet that uses the hw_timer_t, I can't reproduce this issue. The snippet I ran (and log is attached below)

<details>
  <summary>Sketch</summary>

```cpp
/*
 Repeat timer example
 This example shows how to use hardware timer in ESP32. The timer calls onTimer
 function every second. The timer can be stopped with button attached to PIN 0
 (IO0).
 This example code is in the public domain.
 */

// Stop button is attached to PIN 0 (IO0)
#define BTN_STOP_ALARM    0

hw_timer_t * timer = NULL;
volatile SemaphoreHandle_t timerSemaphore;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile uint32_t isrCounter = 0;
volatile uint32_t lastIsrAt = 0;

void ARDUINO_ISR_ATTR onTimer(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  isrCounter++;
  lastIsrAt = millis();
  portEXIT_CRITICAL_ISR(&timerMux);
  // Give a semaphore that we can check in the loop
  xSemaphoreGiveFromISR(timerSemaphore, NULL);
  // It is safe to use digitalRead/Write here if you want to toggle an output
}

void setup() {
  Serial.begin(115200);

  // Set BTN_STOP_ALARM to input mode
  pinMode(BTN_STOP_ALARM, INPUT);

  // Create semaphore to inform us when the timer has fired
  timerSemaphore = xSemaphoreCreateBinary();

  // Use 1st timer of 4 (counted from zero).
  // Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more
  // info).
  timer = timerBegin(0, 80, true);

  // Attach onTimer function to our timer.
  timerAttachInterrupt(timer, &onTimer, true);

  // Set alarm to call onTimer function every second (value in microseconds).
  // Repeat the alarm (third parameter)
  timerAlarmWrite(timer, 1000000, true);

  // Start an alarm
  timerAlarmEnable(timer);
}

void loop()
{
  xSemaphoreTake(timerSemaphore, portMAX_DELAY);
  printf("Taken\n");
}

```
</details>

<details>
  <summary>Log</summary>

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x510
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28bc
entry 0x4004c18c
Taken
Taken
Taken
```
</details>

> I've tried using the ESP-IDF API with the timer_config_t struct and timer_init() function but they seemed to be undefined. Do i have to include some header files? I didn't run into this problem when I was messing with the GPIO/WDT/Sleep APIs.

To use the IDF timer API, you need to `#include "driver/timer.h"`. As a general rule, if you want to include an IDF header, check the header's location in IDF, the add an include path related to its nearest parent `include` directory. For example:

- The the location of `i2c.h` in IDF is `esp-idf/components/driver/include/driver/i2c.h`
- Thus the include directive relative to the nearest parent `include` director will be `#include "driver/i2c.h"`

VojtechBartoska, 2022-08-23:
@Dazza0 I see, don't worry. Tagging the author @sryan5, is this still valid?

closed 2022-08-31 by P-R-O-C-H-Y